Koenkk/zigbee2mqtt #11855: [New device support]:

lukejagger, 2022-03-15:
### Link

https://www.zemismart.com/products/g95

### Database entry

"friendly_name":"0xb4e3f9fffeff9e39","hardwareVersion":1,"ieeeAddr":"0xb4e3f9fffeff9e39","lastSeen":1647345264995,"manufacturerID":4098,"manufacturerName":"_TZ3210_nehayyhx","model":"TS0501B","modelID":"TS0501B","networkAddress":18576,"powerSource":"Mains (single phase)","type":"Router","vendor":"-"},{"dateCode":"","description":"Light controller","friendly_name":"Zemismart filament retro","hardwareVersion":1,"ieeeAddr":"0x50325ffffea43243","lastSeen":1647343159704,"manufacturerID":4098,"manufacturerName":"_TZ3210_psgq7ysz","model":"TS0502B","modelID":"TS0502B","networkAddress":49494,"powerSource":"Mains (single phase)","type":"Router","vendor":"TuYa"}]'

### Comments

I tried a light converter but it still says unsupported.  This is a 'curly filament' large globe zigbee bulb by Zemismart.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0501B'],
    model: '_TZ3210_nehayyhx',
    vendor: '_TZ3210_nehayyhx',
    description: 'Zemismart filament lamp',
    extend: extend.light_onoff_brightness

	// Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    //extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

### Supported color modes

brightness only

lukejagger, 2022-03-15:
Also tried:

> zigbeeModel: ['TS0501B'],
> model: 'TS0501B',
> vendor: '_TZ3210_nehayyhx',

in the converter.

Koenkk, 2022-03-15:
Does the following external converter work?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0501B'],
    model: '_TZ3210_nehayyhx',
    vendor: '_TZ3210_nehayyhx',
    description: 'Zemismart filament lamp',
    extend: extend.light_onoff_brightness(),

	// Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    //extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

lukejagger, 2022-03-15:
No, just tried it, still unsupported.

lukejagger, 2022-03-15:
This is in my Z2M config:

> external_converters:
>   - TS0501B.js

and TS0501B.js is in config folder of HA.

lukejagger, 2022-03-15:
Yes, I realised that later and tried it in the Z2M folder, but that hasn't worked either

lukejagger, 2022-03-16:
Ah! I read the instructions wrong, and added it into the config in the HA addon UI.   Now I've added it into the actual file, it's working!  Here's the final JS...

> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
> 	fingerprint: [{modelID: 'TS0501B', manufacturerName: '_TZ3210_nehayyhx'}],
>         model: 'TS0501B',
>         description: 'Spiral filament globe bulb',
>         vendor: 'Zemismart',
>         extend: extend.light_onoff_brightness(),
> };
> 
> module.exports = definition;

Koenkk, 2022-03-16:
thanks, added it

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

lukejagger, 2022-03-16:
Thanks very much!