Koenkk/zigbee2mqtt #11856: Brightness regulation on Tuya wall dimmer TS0601

ceromanc, 2022-03-15:
### What happened?

Hello, I use the following rotary dimmer switch:

https://aliexpress.com/item/1005002692699078.html?gatewayAdapt=glo2deu&spm=a2g0o.order_list.0.0.21ef5c5fuEptj2

The dimmer was recognized as:

Zigbee model: 			TS0601
Zigbee manufacturer: 		_TZE200_dfxkcots
Manufacturer 				[TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Model 					[TS0601_dimmer](https://www.zigbee2mqtt.io/devices/TS0601_dimmer.html#tuya-ts0601_dimmer)

and the status is: supported.

On/Off works perfectly. Only when I try to change the brightness via HA, the target value is reduced by 2. For example, I would like to change the brightness to 50, but the value immediately jumps to 48. Even if I just click the input field and then outside with the mouse, the actual value is reduced by 2 each time until the value becomes a minus value, after which an error message appears.

Any idea how to correct it?

Many thanks in advance.

### How to reproduce it (minimal and precise)

Just try to change the brightness vie input field or slider

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

SONOFF ZB Dongle-P CC2652P

### Debug log

[image]

ceromanc, 2022-04-19:
Added animated gif to show the issue

[image]

ceromanc, 2022-05-09:
After update to Zigbee2MQTT Version
 1.25.1 issue is solved.

closed 2022-05-09 by ceromanc

bort900, 2022-07-16:
I am having the exact same issue as described here.

I have a MOES branded Tuya TS0601 wall dimmer and it will respond to on/off commands but not brightness commands. You can manually adjust the dimmer at the switch and it will reflect the state properly, but you cant set brightness from Zigbee2MQTT

I am using Zigbee2MQTT 1.26.0 with zStack3x0 2022021 on a SONOFF Zigbee 3.0 USB adapter.