Koenkk/zigbee2mqtt #11863: [New device support]: Nous A1Z

GSzabados, 2022-03-16:
### Link

https://nous.technology/product/a1z-1.html

### Database entry

{"id":88,"type":"Router","ieeeAddr":"0xa4c1381477066fd5","nwkAddr":28357,"manufId":4417,"manufName":"_TZ3000_ksw8qtmt","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_ksw8qtmt","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1647421016348,"defaultSendRequestWhen":"immediate"}

### Comments

Nous, another Tuya company, has these plugs.
They seems to be the same TS011F as many others.

I couldn't find the `_TZ3000_ksw8qtmt` manufacturer in the code base yet.

GSzabados, 2022-03-16:
Tried this external converted from the Tuya TS011F:

Child lock, voltage, current and power works. Couldn't get yet an energy value, it still says `NULL`, but I haven't used it with any powerful device for a longer period yet, but I would assume that it should report at least 0 once. 

Haven't tried yet, indicator light options, and changing the power_outage_memory, but it looks like that the second one set to restore works as expected.

I will report back with further findings as well.

```
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const globalStore = require('zigbee-herdsman-converters/lib/store')
const ota = require('zigbee-herdsman-converters//lib/ota');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['TS011F'],
        model: 'A1Z',
        description: 'Smart plug (with power monitoring)',
        vendor: 'Nous',
        ota: ota.zigbeeOTA,
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory,
            fz.ts011f_plug_indicator_mode, fz.ts011f_plug_child_mode],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.ts011f_plug_indicator_mode, tz.ts011f_plug_child_mode],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage'),
            exposes.enum('indicator_mode', ea.ALL, ['off', 'off/on', 'on/off', 'on'])
                .withDescription('Plug LED indicator mode'), e.child_lock()],
    },
];

```

GSzabados, 2022-03-16:
A currentSumDelivered through the Dev Console updated the energy value as well.

Koenkk, 2022-03-21:
added the device, could you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

GSzabados, 2022-03-22:
> could you check if it works?

I haven't re-paired it, but it has picked up the Tuya  converter correctly and continued working without any issue. A bit later I will try to delete one of the plugs and re-pair it as well, just to make sure that it works.

Should be Nous A1Z added to the whitelabel list as well?

GSzabados, 2022-04-02:
Thanks @Koenkk! It is working well with 1.25.0.

rvmuilekom69, 2022-04-17:
Hi all,
I have this same socket, any idea when this device will be supported in the main version?
Thanks

GSzabados, 2022-04-17:
> when this device will be supported in the main version?

It is already supported from 1.25.0.

rvmuilekom69, 2022-04-17:
You are totally right. I did not have the latest version. Thanks