Koenkk/zigbee2mqtt #11864: All lights sets brightness to 1 after host restart

Mr-Groch, 2022-03-16:
### What happened?

Each time after host with z2m, ha and mosquitto docker containers is restarted, all lights will have brightness: 1 after turn on.

You can see in attached log that for example:
1. 'Lampka mała (szara) - sypialnia' has brightness 172, and next on 2022-03-16 09:34:58 it will have brightness 1
2. 'Lampka RGBW - sypialnia' has brightness 254, and next on 2022-03-16 09:35:00 it will have brightness 1
3. 'Lampa - kuchnia' has brightness 152, and next on 2022-03-16 09:35:09 it will have brightness 1
4. 'Taśma LED RGBW - zewnętrzna 1' has brightness 120, and next on 2022-03-16 09:35:16 it will have brightness 1
5. 'Listwa LED RGBW - kuchnia (lewa)' has brightness 155, and next on 2022-03-16 09:35:52 it will have brightness 1
And so on for all lights that supports brightness (some of them had brightness 1 on start, becouse this happens to them earlier, and I had not turned it on and adjusted till then)...

### What did you expect to happen?

Brightness should be properly restored from state.json

### How to reproduce it (minimal and precise)

Restart host machine with HA, Z2M and Mosquitto containers (this is Synology NAS - it takes about 5 minutes from reboot start to fully start all containers)

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

Sonoff CC2652p USB Stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8260670/log.txt)

Mr-Groch, 2022-03-16:
I have made additional tests, and I can confirm that to reproduce this, you need just to stop z2m container for abut 5 minutes start it and wait additional few minutes after start - all lights will respectively have brightness 1 after a while...

Mr-Groch, 2022-03-16:
> * What device/model is this?

Issue occurs on all devices - there are OSRAM bulbs ([AC03645](https://www.zigbee2mqtt.io/devices/AC03645.html#osram-ac03645), [AB32840](https://www.zigbee2mqtt.io/devices/AB32840.html#osram-ab32840), [AC03641](https://www.zigbee2mqtt.io/devices/AC03641.html#osram-ac03641)), IKEA bulbs ([LED1924G9](https://www.zigbee2mqtt.io/devices/LED1924G9.html#ikea-led1924g9)), LIDL LED strip ([HG06104A](https://www.zigbee2mqtt.io/devices/HG06104A.html#lidl-hg06104a)), LIDL LED bar ([14149505L/14149506L](https://www.zigbee2mqtt.io/devices/14149505L_14149506L.html#lidl-14149505l%252F14149506l)), OSRAM LED strips ([4052899926110](https://www.zigbee2mqtt.io/devices/4052899926110.html#osram-4052899926110) and [4058075036185](https://www.zigbee2mqtt.io/devices/4058075036185.html#osram-4058075036185))... 

> * When you set a brightness and read it again via the frontend, what does the device return?

When I turn it on and set brigthness it will report correctly:

[image]

But when I turn it off, it will always report 1:

[image]

I think that the issue is that z2m should not update brigthness from device that is turned off. As you can see in my logs, all devices that have brightness 1, reports earlier:

```
debug 2022-03-16 09:34:58: Retrieving state of 'Lampka mała (szara) - sypialnia' after reconnect
debug 2022-03-16 09:34:58: Received Zigbee message from 'Lampka mała (szara) - sypialnia', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
```

So if device after reconnect reports `"onOff":0`, brightness level should not be overwritten in states.json

sjorge, 2022-03-17:
@Koenkk don't we do some special things with brightness=1 when turning off and on the bulb? Or was that the other way around. I vaguely remember some sort of work around for that for some devices.

Koenkk, 2022-03-17:
@sjorge I guess you are talking about the [`turnsOffAtBrightness1`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/e104abdca8cb4b4c24d908dd9f66b9dd7d414250/devices/philips.js#L44) parameter, which is used to publish the correct state when you set brightness: 1. Thats something different.

@Mr-Groch I pushed a fix now, can you test it?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

sjorge, 2022-03-17:
> @sjorge I guess you are talking about the [`turnsOffAtBrightness1`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/e104abdca8cb4b4c24d908dd9f66b9dd7d414250/devices/philips.js#L44) parameter, which is used to publish the correct state when you set brightness: 1. Thats something different.

Ah yes! That was it.

Mr-Groch, 2022-03-18:
> @Mr-Groch I pushed a fix now, can you test it?

Thanks, looks that it is fixed now! After host reboot, lights have now proper brightness after turning on. I've also stopped z2m container, waited few minutes, started and brithness was also correct after turning lights on

closed 2022-03-18 by Mr-Groch