Koenkk/zigbee2mqtt #11867: Failed to configure Legrand 067775/741811

mducommun, 2022-03-16:
### What happened?

I added two power plug Legrand 067775/741811 in my installation. They do work well and I can switch them on or off, I also have the instantaneous measured power but the zigbee2mqtt log on Home Assistant report an error for both of them:

Zigbee2MQTT:error 2022-03-16 14:37:08: Failed to configure 'Prise TV 2', attempt 1 (Error: Read 0x000474000011b152/1 haElectricalMeasurement(["acFrequencyDivisor","acFrequencyMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at processTicksAndRejections (internal/process/task_queues.js:95:5)
    at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:25:5)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/legrand.js:209:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13)
    at Configure.start (/app/lib/extension/configure.ts:68:13)
    at Controller.callExtensions (/app/lib/controller.ts:316:17)
    at Controller.start (/app/lib/controller.ts:153:9)
    at start (/app/index.js:101:5))

I also have no voltage or current showed in z2m.
Is it a bug ?

### Zigbee2MQTT version

1.24.0-1

### Adapter firmware version

20211217

### Adapter

Zigstar Lan

Koenkk, 2022-03-16:
Could you check if this issue has been fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [' Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
    model: '067775/741811',
    vendor: 'Legrand',
    ota: ota.zigbeeOTA,
    description: 'Power socket with power consumption monitoring',
    fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement],
    toZigbee: [tz.on_off, tz.legrand_settingAlwaysEnableLed, tz.legrand_identify],
    exposes: [e.switch(), e.action(['identify']), e.power(), e.voltage(), e.current()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint);
        try {
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        } catch (error) {
            /* fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/11867 */
        }
        await reporting.activePower(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)
- check that the reported power is correct

mducommun, 2022-03-16:
Hi,
Thanks for your help, I'm not familiar with coding and I get some errors with the external converter you ask me to add:

(node:1424) UnhandledPromiseRejectionWarning: ReferenceError: ota is not defined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:1424) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:1424) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Expected ( instead of { at line 18
Expected an identifier at line 19
missing similcon at line 27

Koenkk, 2022-03-16:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/11867#issuecomment-1069428418, can you check again?

mducommun, 2022-03-16:
It works, no more errors in my logs.
But they are no longer OTA compatible.
[image]
Voltage and Current are still "Null"

Thanks

Koenkk, 2022-03-17:
`UNREPORTABLE_ATTRIBUTE` indicates that the device does not support current (note that there could be different version out for this device). You could check wether it does support reporting voltage by commenting `await reporting.rmsCurrent(endpoint);` (change it to `//await reporting.rmsCurrent(endpoint);`

Inlustra, 2022-04-09:
Hey guys, I'm also struggling with this device. I've got a weird version being reported also, this is from the OTA menu:
0x000474000009edc2 | Legrand | 067775/741811 | 20180417 | 00e��!u

I get an error stating that it's newer than the versions online.
I've tried your version above but I'm not getting any measurements at all, it is still in the OTA menu.

action
Triggered action (e.g. a button click)
Null
power
Instantaneous measured power
NullW
voltage
Measured electrical potential value
NullV
current
Instantaneous measured electrical current
NullA

mapo31, 2022-04-25:
> Hey guys, I'm also struggling with this device. I've got a weird version being reported also, this is from the OTA menu: 0x000474000009edc2 | Legrand | 067775/741811 | 20180417 | 00e��!����u�
>
> I get [...]

Hello,
Same issue here

closed 2022-06-03 by github-actions[bot]

capi92, 2022-08-16:
> > Hey guys, I'm also struggling with this device. I've got a weird version being reported also, this is from the OTA menu: 0x000474000009edc2 | Legrand | [067775/741811](tel:067775/741811) | 20180417 | 00e��!����u�
> > I get an error stating that it's newer than the versions online. I've tried your version above but I'm not getting any measurements at all, it is still in the OTA menu.
> > action Triggered action (e.g. a button click) Null power Instantaneous measured power NullW voltage Measured electrical potential value NullV current Instantaneous measured electrical current NullA
> 
> Hello, Same issue here

Same here

Date du firmware
20180605
Version du firmware
016  !����!
Constructeur
[Legrand](https://www.zigbee2mqtt.io/supported-devices/#v=Legrand)
Modèle
[067775/741811](https://www.zigbee2mqtt.io/devices/067775_741811.html#legrand-067775%252F741811)

novisys, 2022-08-19:
Hi,
i have the same problem here !
Any one solved this kind of probelem ?

mapo31, 2022-08-19:
Hello
I had the same issue and I solved it by setting force ota to true. This way it skip the version check.
https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files
[image]

capi92, 2022-08-19:
Hi @mapo31 ,

Could you describe how you set force ota to true ?

Thanks,
Pierre

novisys, 2022-08-19:
Hi ,
i managed to get the OTA update working but it's so sloooow !
Here is the step by step:
1- add this parametter to your configuration.yaml file:
```
ota:
    zigbee_ota_override_index_location: my_index.json
```

2- create an empty ".json" file (https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files)
   > IMPORTANT: put this file in a reachable directory anywhere on your server or on external host url.
   > add this lines on your ".json" file:
```
[
    {
        "fileVersion": 2966019,
        "fileSize": 242695,
        "manufacturerCode": 4129,
        "imageType": 16,
        "sha512": "fc3a7f49cef2a6847168cfe165b335feeecbcd64c8729afa4b6aea302988ebf5a17c85b285230ece7c9cc469017e8b694f5ef33481bdfce8d5fb9b5645f98c70",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0010-2d4203-NLM-45.zigbee",
        "path": "images/Legrand/1021-0010-2d4203-NLM-45.zigbee",
        "force": true
    },
    {
        "fileVersion": 4080131,
        "fileSize": 250647,
        "manufacturerCode": 4129,
        "imageType": 17,
        "sha512": "63169864311803b092291a093427fa7bc5d10d9bc044f3340b98b6c473ed4dd0f987fe775dfdf0c7f5f7e9525863ea8962515b804560076f16eb32fa1c83eb31",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0011-3e4203-NLP-62.zigbee",
        "path": "images/Legrand/1021-0011-3e4203-NLP-62.zigbee",
        "force": true
    },
    {
        "fileVersion": 3555843,
        "fileSize": 196871,
        "manufacturerCode": 4129,
        "imageType": 18,
        "sha512": "58290da7042fbcbc4fa759cd880bd687349025b3c54cbd7f5fc1ce1babd82fa3cccc890d536861b37c21ee7b302148c1e1124cc1ebdde217e3b6709147af5787",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Legrand/1021-0012-364203-NLT-54.zigbee",
        "path": "images/Legrand/1021-0012-364203-NLT-54.zigbee",
        "force": true
    }
]
```
3- Restart z2m 
4- Go to the OTA and check for the update 
5- Make the Update and Enjoy !

capi92, 2022-08-19:
> Hi , i managed to get the OTA update working but it's so sloooow ! Here is the step by step: 1- add this parametter to your configuration.yaml file:
>
> ```
> ota:
>     zigbee_ota_override_index_location: [...]

Working, OTA on going on my side. Thanks

novisys, 2022-08-19:
Ok, good!
Now any one know how to get the Voltage and Current to work ?
In développement section this two parameters can be asked succesfully but can't get it work adn give me 0 all time !
Maybe the device is different from this one [067775/741811] because mine is an [600491A] and need to be added on Z@M as new device with new parameters ? @Koenkk please help us !

novisys, 2022-08-19:
i have this message when i try with CUSTOM: 
2022-08-19 23:49:15Received Zigbee message from '0x000474000009cb5d', type 'raw', cluster 'greenPower', data '{"data":[25,69,2,8],"type":"Buffer"}' from endpoint 242 with groupID null

DejavuDmitry, 2023-03-04:
Okay, but now in [1.30.2] version I can't update my Legrand devices via OTA

LeorFinacre, 2023-03-12:
> Could you check with:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: [' Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
>     model: '067775/741811',
>     vendor: 'Legrand',
>     ota: ota.zigbeeOTA,
>     description: 'Power socket with power consumption monitoring CUSTOM',
>     fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement],
>     toZigbee: [tz.on_off, tz.legrand_settingAlwaysEnableLed, tz.legrand_identify],
>     exposes: [e.switch(), e.action(['identify']), e.power(), e.voltage(), e.current()],
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
>         await reporting.onOff(endpoint);
>         try {
>             await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
>         } catch (error) {
>             /* fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/11867 */
>         }
>         await reporting.activePower(endpoint);
>         await reporting.rmsCurrent(endpoint);
>         await reporting.rmsVoltage(endpoint);
>     },
> };
> 
> module.exports = definition;
> ```

Hi, i have the exact same device and trying to get the power consumption working.
I tried adding this file but zigbee2mqtt does not restart, I have this error :
`typeerror cannot read properties of undefined (reading 'options')`

Is there anything to do to fix this ?

Thank's