Koenkk/zigbee2mqtt #11870: TypeError: meta.message.state.toLowerCase is not a function

fzzybllz, 2022-03-16:
### What happened?

OSRAM AB3257001NJ is not changing state, if a boolean true/false is published to zigbee2mqtt/friendlyname/set/state. An error occurs in the log, see below.
Tried it in two different environments with different coordinators and different devices.

### What did you expect to happen?

The device should change it's state

### How to reproduce it (minimal and precise)

Publishing "on"/"off" to zigbee2mqtt/friendlyname/set/state -> works fine
Publishing true/false or 0/1 to zigbee2mqtt/friendlyname/set/state by mqttexplorer -> error
Publishing true/false or 0/1 to zigbee2mqtt/friendlyname/set/state by node-red -> error
Publishing true/false or 0/1 to zigbee2mqtt/friendlyname/set/state by iobroker -> error

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20201127

### Adapter

usb-Texas_Instruments_TI_CC2531_USB_CDC or SONOFF Zigbee 3.0 USB

### Debug log

ERROR: Publish 'set' 'state' to 'DeviceABC' failed: 'TypeError: meta.message.state.toLowerCase is not a function'
DEBUG: TypeError: meta.message.state.toLowerCase is not a function at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:266:85) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (events.js:412:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:122:27) at MqttClient.emit (events.js:400:28) at MqttClient._handlePublish (/app/node_modules/mqtt/lib/client.js:1547:12) at MqttClient._handlePacket (/app/node_modules/mqtt/lib/client.js:535:12) at work (/app/node_modules/mqtt/lib/client.js:438:12) at Writable.writable._write (/app/node_modules/mqtt/lib/client.js:452:5)

fzzybllz, 2022-03-16:
Similar to #212 https://github.com/Koenkk/zigbee-herdsman-converters/issues/212

Koenkk, 2022-03-16:
true/false/0/1 are not allowed, only `ON` and `OFF`

fzzybllz, 2022-03-18:
Is there a special reason for that? I just want to understand that.
Most switches are controlled by booleans, which also means less traffic regarding bit-length

Koenkk, 2022-03-18:
> Most switches are controlled by booleans, which also means less traffic regarding bit-length.

What you send to z2m is translated to the zigbee spec (bits)

> Is it planned to enable this in the future?

No

closed 2022-03-21 by fzzybllz