Koenkk/zigbee2mqtt #11872: Hive radiator valve TRV fails to configure

Westcott1, 2022-03-16:
### What happened?

 When restarting Z2M I get -

Failed to configure 'Lilac radiator', attempt 3 (Error: Read 0x842e14fffe98d8e4/1 hvacThermostat(["danfossWindowOpenFeatureEnable","danfossWindowOpenExternal","danfossDayOfWeek","danfossTriggerTime","danfossAlgorithmScaleFactor","danfossHeatAvailable","danfossMountedModeControl","danfossMountedModeActive","danfossExternalMeasuredRoomSensor","danfossRadiatorCovered","danfossLoadBalancingEnable","danfossLoadRoomMean"], {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4678,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/danfoss.js:120:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:115:13))

### What did you expect to happen?

Configuration to succeed

### How to reproduce it (minimal and precise)

Restart Z2M, or click its "Reconfigure" button in the console.

### Zigbee2MQTT version

 1.24.0-dev commit: 5b730575

### Adapter firmware version

 zStack3x0 -  20211207

### Adapter

zzh

Koenkk, 2022-03-16:
Could you check if this issue has been fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // eTRV0100 is the same as Hive TRV001 and Popp eT093WRO. If implementing anything, please consider
    // changing those two too.
    zigbeeModel: ['eTRV0100', 'eTRV0101'],
    model: '014G2461',
    vendor: 'Danfoss',
    description: 'Ally thermostat CUSTOM',
    whiteLabel: [{vendor: 'Danfoss', model: '014G2463'}],
    fromZigbee: [fz.battery, fz.thermostat, fz.thermostat_weekly_schedule, fz.hvac_user_interface, fz.danfoss_thermostat],
    toZigbee: [tz.danfoss_thermostat_occupied_heating_setpoint, tz.thermostat_local_temperature, tz.danfoss_mounted_mode_active,
        tz.danfoss_mounted_mode_control, tz.danfoss_thermostat_vertical_orientation, tz.danfoss_algorithm_scale_factor,
        tz.danfoss_heat_available, tz.danfoss_heat_required, tz.danfoss_day_of_week, tz.danfoss_trigger_time,
        tz.danfoss_window_open_internal, tz.danfoss_window_open_external, tz.danfoss_load_estimate,
        tz.danfoss_viewing_direction, tz.danfoss_external_measured_room_sensor, tz.danfoss_radiator_covered,
        tz.thermostat_keypad_lockout, tz.thermostat_system_mode, tz.danfoss_load_balancing_enable, tz.danfoss_load_room_mean,
        tz.thermostat_weekly_schedule, tz.thermostat_clear_weekly_schedule, tz.thermostat_programming_operation_mode,
        tz.danfoss_window_open_feature],
    exposes: [e.battery(), e.keypad_lockout(), e.programming_operation_mode(),
        exposes.binary('mounted_mode_active', ea.STATE_GET, true, false)
            .withDescription('Is the unit in mounting mode. This is set to `false` for mounted (already on ' +
                'the radiator) or `true` for not mounted (after factory reset)'),
        exposes.binary('mounted_mode_control', ea.ALL, true, false)
            .withDescription('Set the unit mounting mode. `false` Go to Mounted Mode or `true` Go to Mounting Mode'),
        exposes.binary('thermostat_vertical_orientation', ea.ALL, true, false)
            .withDescription('Thermostat Orientation. This is important for the PID in how it assesses temperature. ' +
                '`false` Horizontal or `true` Vertical'),
        exposes.binary('viewing_direction', ea.ALL, true, false)
            .withDescription('Viewing/Display Direction. `false` Horizontal or `true` Vertical'),
        exposes.binary('heat_available', ea.ALL, true, false)
            .withDescription('Not clear how this affects operation. However, it would appear that the device does not execute any ' +
                'motor functions if this is set to false. This may be a means to conserve battery during periods that the heating ' +
                'system is not energized (e.g. during summer). `false` No Heat Available or `true` Heat Available'),
        exposes.binary('heat_required', ea.STATE_GET, true, false)
            .withDescription('Whether or not the unit needs warm water. `false` No Heat Request or `true` Heat Request'),
        exposes.enum('setpoint_change_source', ea.STATE, ['manual', 'schedule', 'externally'])
            .withDescription('Values observed are `0` (manual), `1` (schedule) or `2` (externally)'),
        exposes.climate().withSetpoint('occupied_heating_setpoint', 5, 32, 0.5).withLocalTemperature().withPiHeatingDemand()
            .withSystemMode(['heat']).withRunningState(['idle', 'heat'], ea.STATE),
        exposes.numeric('external_measured_room_sensor', ea.ALL)
            .withDescription('If `radiator_covered` is `true`: Set at maximum 30 minutes interval but not more often than every ' +
            '5 minutes and 0.1 degrees difference. Resets every 35 minutes to standard. If `radiator_covered` is `false`: ' +
            'Set at maximum 3 hours interval but not more often than every 30 minutes and 0.1 degrees difference. ' +
            'Resets every 3 hours to standard. Value 21C = 2100 (-8000=undefined).')
            .withValueMin(-8000).withValueMax(3500),
        exposes.binary('radiator_covered', ea.ALL, true, false)
            .withDescription('Set if the TRV should solely rely on external_measured_room_sensor or operate in offset mode. ' +
                '`false` = Auto Offset Mode or `true` = Room Sensor Mode'),
        exposes.binary('window_open_feature', ea.ALL, true, false)
            .withDescription('Whether or not the window open feature is enabled'),
        exposes.numeric('window_open_internal', ea.STATE_GET).withValueMin(0).withValueMax(4)
            .withDescription('0=Quarantine, 1=Windows are closed, 2=Hold - Windows are maybe about to open, ' +
                '3=Open window detected, 4=In window open state from external but detected closed locally'),
        exposes.binary('window_open_external', ea.ALL, true, false)
            .withDescription('Set if the window is open or close. This setting will trigger a change in the internal ' +
                'window and heating demand. `false` (windows are closed) or `true` (windows are open)'),
        exposes.enum('day_of_week', ea.ALL,
[... cut ...]
        const values = {timeStatus: 3, time: time, timeZone: ((new Date()).getTimezoneOffset() * -1) * 60};
        endpoint.write('genTime', values);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)

Koenkk, 2022-03-17:
At the top of the file add `const ota = require('zigbee-herdsman-converters/lib/ota');`

Westcott1, 2022-03-18:
The description is still "Radiator valve based on Danfoss Ally (014G2461)"
How do I override the built-in config for mine, which is a Hive -
Model
: UK7004240
Zigbee Model: TRV001

Thanks.

Koenkk, 2022-03-18:
@Westcott1 updated https://github.com/Koenkk/zigbee2mqtt/issues/11872#issuecomment-1069434808 , could you check again/

Westcott1, 2022-03-18:
Yes, that's fixed it!
It took me a while to realise that you'd edited the config above.

Koenkk, 2022-03-19:
At the top of the file, add `const constants = require('zigbee-herdsman-converters/lib/constants');`

Westcott1, 2022-03-19:
Yes, that's fixed it!

Koenkk, 2022-03-20:
great, pushed the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-20 by Koenkk

robertalexa, 2022-04-01:
@Koenkk Thanks for sorting this. I have noticed this too a few releases ago when Hive and Pop have been merged into the Danfoss Ally converter. The issue with that is they do their own version of the firmware, and the firmware is not public to download/update. So yes, some of the command will not work on Hive for example or even on Danfoss if they were on older firmware. Unfortunately didn't get a chance to patch it, despite experiencing this myself. Fortunately it was doing 3 attempts to configure and then stopping, so not a deal breaker.

kuduacz, 2022-12-07:
Trying to pair two new hive trv's. one of them paired successfully but 2nd not:
[image]

Finally its added but not working at all. can change anything
[image]

and when try to change temperature setpoint its doesnt accepted. i mean when i run to z2m again there is old temp setpoint. Changed only when change it manually on trv.

robertalexa, 2022-12-07:
@kuduacz Looks like a network quality issue at your end. If you are pairing through a coordinator, bring the device as close as possible to the coordinator. If you are pairing through a router device, bring it close to that. If pairing fails through a router device you may be affected by an issue that was recently resolved (depending on your coordinator firmware) where some security values were not being passed through. If that is the case have a look at the last few releases of z2m, there is an explanation and a solution.

As to after the device gets paired, not all attributes are supported hy Hive TRV. Hive TRV is a rebranding of the Danfoss Ally thermostat (which is what z2m supports much better) with a custom Hive firmware. If you get UNSUPPORTED_ATTRIBUTE it means they are not supported by Hive, and there is nothing we can do. (trust me, i have 8 of them and i tried everything, incuding flashing them with Danfoss firmware) we are at the mercy of Hive releasing updates, us pairing our trvs with hive just to update, and then hope for the best. But Hive are terrible at release notes.... or anything else for that matter.

[image]
Same thing at my end, but it is expected unfortunately. Call Hive and tell them how much they suck :) It took me ages for them to even admit their device is a rebrand with custom firmware, despite it being freaking obvious :laughing:

kuduacz, 2022-12-07:
try to pair it 0.5m from coordinator (sonfoo zigbee dondle plus
[image]
with that error. 
try to pair to zigbee router. maybe then it gonna work.
You know, i just wont to set target temperature throu my HA. nothing more.
one of my trv that work:
Firmware build date
20190411
Firmware version
01.15.0008 02.02

and 2nd one no have that values.

BTW ota dont work too.

robertalexa, 2022-12-07:
@kuduacz It works perfectly fine to set the temperature via HA :). you just need to use the correct attribute.
This is what you should change
[image]

Here is my HA (works natively, no custom work needed) - ignore the unavailable TRV, it is not on my network at the minute, been doing some tests.
[image]

OTA is not supported. z2m provides OTA for the original Danfoss Ally firmware, which works on Ally and Popp trv (another rebrand). But like i side, Hive are using a custom one, so it is incompatible - like i said, i know, i tried, a lot.

Do a factory reset on your TRV. Remove the battery. hold the button, put the battery in, keep holding until the device reboots 3 times. Enable pairing after that in z2m

kuduacz, 2022-12-07:
Try to change it via z2m panel and via ha thermostat card, both ways failture (when re-enter to z2m device there is nothing changed on temperature setoint)

Thaks for tip about factory reset.
1: remove battery
2: press and hold button
3: put battery
4: wait to 3 time reset
5: enable pairing mode in z2m
??

and than it shout pair correctly.?

I'am using latest version of z2m so try to update my coordinator.

robertalexa, 2022-12-07:
The above will only work if the device pairs correctly (try on the second one that said UNSUPPORTED_ATTRIBUTE).

Tthe one that said failed to interview you will need to remove it, first and then re-attemp pairing it with a factory reset of the device.

kuduacz, 2022-12-07:
Yes. As i said before 1st work as expected. can change setpoint temperature and few other things. there is problem just with 2nd.

Anyway can You tell me which function work?
Wont to compare with my 1st trv. 
keypadk work
programming operation mode not work
external measures room sensor - work but cant see any differences.
radiator covered - not work
window open external work

robertalexa, 2022-12-07:
I do not remember on top of my head. Press the refresh icon. If you hget Unsupported attribute it doesn't work, if you get a value, then it works. Whichever wants you will get UNSUPORTED is what Hive doesn;t support in the firmware of the actual device. z2m just exposes them as they shoul dhave been from the Danfoss firmware.This was done as a choice because they share pretty much everything, and it is much easier to manage

kuduacz, 2022-12-07:
Ok. Try factory reset and there is something strange. IT show on devices even when there pairing mode is disabled.
Remove IT again enable pairing mode and see that:[image]

pnieweglowski, 2022-12-17:
@kuduacz I get my Hive valve and had many problems with pairing of this device. At some point, I was not able to join it to my network. I had to reset my server. After that, I reset device as described above and it appears in z2m.
Firmware build date
    20220425
Firmware version
    02.33.0008 02.33
I'm able to set temperature from HA interface at this moment. 
I can not see local_temperature in HA and programming_operation_mode operations finish with unsupported_attributes errors.