Koenkk/zigbee2mqtt #11873: zigbee2mqtt - Restart, resets add-on settings

sambeetm, 2022-03-16:
### What happened?

Everytime, I restart the add-on, the custom settings are reset or lost. Basically, I have 3 settings 1) An external converter file 2) Settings-->availability-->Advanced 3) Settings-->Advanced-->Last Seen-->ISO_8601. I have to re-do the above everytime, I restart the add-on.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

SONOFF ZBBridge

Koenkk, 2022-03-16:
This should be fixed in the edge version of the addon already.

mducommun, 2022-03-17:
I didn't understand if it was already fixed in our version or if it will be available but I have the same issues. My availability settings are lost each time the addon is restarted or home assistant is restarted (for an update for example).

sambeetm, 2022-03-17:
looks like this is fixed in the Dev branch. There is an add-on in HA "zigbee2mqtt Edge" which corresponds to Dev branch. If you are willing to remove the production add-on and install the Edge add-on, you will get the fix. But there are few steps involved to remove and install edge. I will prefer to wait until the Edge becomes production...

mducommun, 2022-03-17:
ok get it... I'm gonna wait for the prod release too.
Thanks for the explanation.

Scoff123, 2022-03-21:
Also using a Sonoff ZBBridge and seeing this same behaviour - hopefully the fix will be released into the production version soon

vandad51, 2022-04-14:
> This should be fixed in the edge version of the addon already.

@Koenkk How long until this fix makes it into the main release? I have the same issue and finally seen I'm not the only one that's experiencing it.

Koenkk, 2022-04-14:
@vandad51 it is in 1.25.0-1 (all settings except serial and mqtt are managed via the frontend now).

vandad51, 2022-04-24:
> @vandad51 it is in 1.25.0-1 (all settings except serial and mqtt are managed via the frontend now): https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/releases/tag/v1.25.0-1

Yup, updating fixed the problem. Thanks for all your work!

closed 2022-04-24 by Koenkk