Koenkk/zigbee2mqtt #11875: Problem with Group and Color Temp

UweLoyal, 2022-03-16:
### What happened?

Please see:
https://github.com/ioBroker/ioBroker.zigbee/issues/1390

### Zigbee2MQTT version

2-1.2.7.1.

### Adapter firmware version

20210708

### Adapter

Zigbee Coordinator in ioBroker

Koenkk, 2022-03-16:
Since you are using iobroker, this issue doesn't belong in the zigbee2mqtt issue tracker.

closed 2022-03-16 by Koenkk

UweLoyal, 2022-03-16:
I's realy funny, asgothian (https://github.com/ioBroker/ioBroker.zigbee/issues/1390) say ask the hardware producer, slaesh (https://slae.sh/projects/cc2652/#). slaesh say, ask Koenkk (https://github.com/Koenkk/zigbee2mqtt/issues/11875) and Koenkk, sais ask asgothian.

Who can now help me after staying again at the beginning?

In hope for Help!

Rgds
Uwe Loyal

Koenkk, 2022-03-16:
- You are not using zigbee2mqtt, so this is definitely not a z2m issue
- Looking at https://github.com/ioBroker/ioBroker.zigbee/issues/1390 this doesn't looks to be an issue with the hardware (seems software related)
- It could be relatted to zigbee-herdsman-converter/zigbee-herdsman, but then @[asgothian](https://github.com/asgothian) should explain what the problem is here