Koenkk/zigbee2mqtt #11876: LED1925G6 firmware update not working properly

marthubner, 2022-03-16:
### What happened?

I have IKEA bulb LED1925G6 which should have a new firmware [(V-1.1.018)](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) update available since yesterday. My bulb is on firmware 1.0.021 and when I check for update in OTA section I only get message saying "Firmware on '0x0c4314fffe8d7354' is newer than latest firmware online." which I assume is not true.

### What did you expect to happen?

Firmware update to start.

### Zigbee2MQTT version

1.24.0-dev commit: 83f8250

### Adapter firmware version

20210319

### Adapter

Slaesh's cc2652rb stick

MattWestb, 2022-03-16:
Its looks IKEA or Z2M **and** ZHA is parsed the version number wrongly the the old one is looking like this in ZHA: `Firmware: 0x10021655` and then forcing updating to the new one its looks like this: `Firmware: 0x00011018` = lower version.

I have patching ZHA and forcing updating the firmware (was taking under 10 minutes updating it) and its looks working OK but i have not time testing it so much.

Have requesting help in ZHA for checking if its the OTA file that have wrong metadata or its parsed wrongly from it of the system.

T-IOOO, 2022-03-17:
I am experiencing the same issue with LED1925G6 and LED1923R5

malaakso, 2022-03-17:
I think the problem is that bulb uses 0x10021655 in the Query Next Image Request as file version, and new metadata on IKEA server says 0x00011018. File version is up to manufacturer to define, so IKEA has changed their versioning here.

T-IOOO, 2022-03-17:
Is it possible to update the bulbs with ikeas gateway? If so, how could that be possible?

malaakso, 2022-03-17:
Zigbee spec says that upgrade server (i.e. gateway) is free to push also older (according to file version) image to the device. So probably it just has special handling for it that needs to be included also in zigbee2mqtt.

MattWestb, 2022-03-17:
I have downgrading many IKEA light and there is not blocked for downgrading as all IKEA controllers is (I have not trying 100% of them but most).

If using deCONZ its only picking the old firmware and doing up/downgrade and if its OK for the device is doing it.

ZHA is possible changing its default only upgrade thru patching the HA container (ZHA WiKi).

We have looking in the metadata and and its looks like IKEA have doing this wrong but the OTA file is OK if getting it on the device.

I have not time testing IKEA GW but i very likely doing it in the weekend and reporting back.

T-IOOO, 2022-03-17:
I’m not surprised If IKEA is the ones to ”blame” here. I hope that this will be resolved

MattWestb, 2022-03-17:
My thinking its no so many manufactures that have updating all there current and most of its discontinued the last year (i think its zero but Philips have doing most but not all legacy devices and controllers).
Comparing with OSRAM that still have devices that is knowing corrupting packages and nearly is not working at all with open systems.
Also releasing the firmware for open system is great comparing with most that can sending one update file that the user cant using in the system and is not allowed sharing it public.
Sad that the controller updates and the CWS3 was not working well but perhaps they shall listening more on the community for getting feed back.

tashda, 2022-03-23:
I no longer receive the error stating that the currently installed firmware is newer than the one available, but now it simply returns "no update available".

That's _one_ problem solved, I guess.. still doesn't work properly.

T-IOOO, 2022-03-23:
Same issue for me. Says no update available and still on the old firmware.

MattWestb, 2022-03-23:
IKEA have reverting the update of the CWS3 from the reddit: https://www.reddit.com/r/tradfri/comments/tk3gq3/recent_firmware_update_causing_issues_with/

closed 2022-04-30 by github-actions[bot]