Koenkk/zigbee2mqtt #11878: GL-C-009 does not respond to brightness_move_onoff or brightness_move

MoosJ80, 2022-03-16:
### What happened?

I have added the GL-C-009 to Zigbee to MQTT.
When i publish "brightness_step" the GL-C-009 responds ok
When publish  brightness_move_onoff or brightness_move nothing happens

### What did you expect to happen?

I expected the light turn brighter

### How to reproduce it (minimal and precise)

I send the messages with MQTTlens

### Zigbee2MQTT version

1.18.1 commit: 1.18.1

### Adapter firmware version

20210120

### Adapter

ZZH!

Koenkk, 2022-03-17:
Probably not supported by the device.

Koenkk, 2022-03-17:
This part of the docs is the same for all lights, I've added a note to the page.

closed 2022-04-24 by github-actions[bot]