Koenkk/zigbee2mqtt #11885: SmartThings 3321-S --- No Vibration ??

sambeetm, 2022-03-17:
### What happened?

I have few Smarthings Multi Sensors. When configured, they are falling under "SmartThings 3321-S" device profile. But this profile does not expose the Vibration capability. All my sensors have the vibration sensing capability. How do I get around it ?

### What did you expect to happen?

Expect the Vibration capability

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

SONOFF ZBBridge

markmghali, 2022-03-17:
Yes, same question! Had the same issue with deconz too

johnggreen, 2022-03-30:
Similar question here also. Not familiar with the zigbee2mqtt development cycle but hoping to get to grips with it soon. While running HA and it works fine with ZHA but unfortunately my network is now using Zigbee2MQTT which mitigates some other issues I was having. Looking at the cluster the device seems to have ```manuSpecificSamsungAccelerometer```.  

This looks to require a PR into zigbee-herdsman-converters to resolve as the [code ](https://github.com/Koenkk/zigbee-herdsman-converters/blob/e9292187c98f62ca686a84fea25338fb5d82c883/devices/smartthings.js#L72) does not seem to be aware of this. 

I noticed that a couple of other models 'multiv4' and 'multi' also share also make reference to this functionality so I'm hoping that we're not far from a fix. I'm wondering whether anyone has any guides to testing changes locally before raising a PR or is in a better position than me to test?

johnggreen, 2022-03-30:
EDIT: Got it working locally after figuring out the documentation. Will raise a PR once I know the functionality is working as expected.

markmghali, 2022-03-30:
Is there a way I can test with my conbee II stick? I'd love to help if I can. currently running z2m in a docker on unraid,

johnggreen, 2022-03-30:
Currently getting the below error, the values look to be pulling through correct within the Web interface.

```

Failed to configure 'doorbell', attempt 3 (Error: Write 0x000d6f000bbd7acf/1 manuSpecificSamsungAccelerometer({"0":{"value":1,"type":32}}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4362,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.configure (/app/dist/util/externally-loaded.js:21:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))

```

sambeetm, 2022-03-30:
Thanks johnggreen. I will also configure locally and update here.

markmghali, 2022-03-30:
Same Error 

Error Failed to configure 'Garage Door', attempt 3 (Error: Write 0x000d6f000bbcc16d/1 manuSpecificSamsungAccelerometer({"0":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4362,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22) at Object.configure (/app/dist/util/externally-loaded.js:21:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))

johnggreen, 2022-03-31:
There's two properties that are set on the newer multipurpose sensor that I'm trying to apply without much success. These are 

```
await endpoint.write('manuSpecificSamsungAccelerometer', {0x0000: {value: 0x01, type: 0x20}}, options);
await endpoint.write('manuSpecificSamsungAccelerometer', {0x0002: {value: 0x0276, type: 0x21}}, options);
```

I presume these are for setting motionThresholdMultipler and motion threshold whether these are even an issue for now?

Below is my current version of the file without errors
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;

const definition = {
        zigbeeModel: ['3321-S'],
        model: '3321-S',
        vendor: 'SmartThings',
        description: 'Multi Sensor (2015 model)',
        fromZigbee: [fz.temperature, fz.ias_contact_alarm_1_report, fz.ias_contact_alarm_1, fz.battery, fz.smartthings_acceleration],
        toZigbee: [],
        meta: {battery: {voltageToPercentage: '3V_2100'}},
        configure: async (device, coordinatorEndpoint, logger) => {

            const endpoint = device.getEndpoint(1);
            const options = {manufacturerCode: 0x104E};
            await reporting.bind(endpoint, coordinatorEndpoint,
                ['msTemperatureMeasurement', 'genPowerCfg', 'manuSpecificSamsungAccelerometer']);
            await reporting.temperature(endpoint);
            await reporting.batteryVoltage(endpoint);
            const payloadA = reporting.payload('acceleration', 10, constants.repInterval.MINUTE, 1);
            await endpoint.configureReporting('manuSpecificSamsungAccelerometer', payloadA, options);
            const payloadX = reporting.payload('x_axis', 10, constants.repInterval.MINUTE, 1);
            await endpoint.configureReporting('manuSpecificSamsungAccelerometer', payloadX, options);
            const payloadY = reporting.payload('y_axis', 10, constants.repInterval.MINUTE, 1);
            await endpoint.configureReporting('manuSpecificSamsungAccelerometer', payloadY, options);
            const payloadZ = reporting.payload('z_axis', 10, constants.repInterval.MINUTE, 1);
            await endpoint.configureReporting('manuSpecificSamsungAccelerometer', payloadZ, options);
            // Has Unknown power source, force it.
            device.powerSource = 'Battery';
            device.save();
        },
        exposes: [e.temperature(), e.contact(), e.battery_low(), e.tamper(), e.battery(),
            e.moving(), e.x_axis(), e.y_axis(), e.z_axis()],
};
module.exports = definition;
```

Is motion working fine for yourselves?

krazos, 2022-03-31:
Just tried the most recently posted external converter file, and the `moving`, `x_axis`, `y_axis` and `z_axis` values are all showing up for me in Zigbee2mqtt and Home Assistant.  Many thanks!

Happy to test any further revisions to the converter file, if helpful.

johnggreen, 2022-04-02:
That's all the changes I can perform without getting some further hardware to debug. I'll mark the PR as active for the changes to be merged in a couple of days if no issues are detected.

johnggreen, 2022-05-03:
These changes were merged in and should just work going forward. This issue can likely be closed.

closed 2022-05-03 by Koenkk

bmmiller, 2022-09-09:
Does this apply to the STSS-MULT-001 sensor as well?  I'm seeing the same sort of behavior with a lot of missing reports.  I tried applying the change outlined above here and it doesn't seem to have any effect.

mcnannay, 2023-01-25:
> Does this apply to the STSS-MULT-001 / PGC313 sensor as well?
> 
> I've got the extra reports added, but everything is Null.
> 
> [image]

I'm getting the same results with the STSS-MULT-001 (PGC313). I have no idea how to implement this unfortunately, I wonder if this sensor works with ZHA.