Koenkk/zigbee2mqtt #11889: Publishing to homeassistant topic not happening

MGlants, 2022-03-17:
### What happened?

Publishing to homeassistant MQTT topics done only after restarting of z2m. Updating of sensors not posting to those topics. 

### What did you expect to happen?

Updating homeassistant/ toppics acording to zigbee2mqtt/ topic.

### How to reproduce it (minimal and precise)

`advanced:
  channel: 11
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: homeassistant/status
  last_seen: ISO_8601
  log_level: info
  log_output:
    - console
  network_key:
   - HIDDEN
  pan_id: 6754
  homeassistant_legacy_entity_attributes: true
  homeassistant_legacy_triggers: false
  ikea_ota_use_test_url: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
experimental:
  new_api: true
  transmit_power: 20
homeassistant: true
frontend:
  port: 8080
mqtt:
  base_topic: zigbee2mqtt
  include_device_information: true
  server: mqtt://emqx
  client_id: zigbee2mqtt
  keepalive: 60
  password: HIDDEN
  reject_unauthorized: true
  user: HIDDEN
  version: 4
  force_disable_retain: true
permit_join: false
serial:
  port: tcp://172.16.10.20:20108
  disable_led: false
  adapter: auto
devices:
HIDDEN
availability: true`

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20210813

### Adapter

zStack3x0

MGlants, 2022-03-17:
Probably longer period for posting it

MGlants, 2022-03-18:
after ha reboot 
[image]

va4ok, 2022-03-30:
Hi, looks i have the same issue.
From time to time zigbee2mqtt addon hangs until reboot plugin.

No new values get or set, plugin CPU load > 20%, RAM > 11%, Embeded UI not responding.

Looks like some memory leak

Gaps on card in HA
[image]

RAM and CPU usages by plugin
[image]

CPU for HA system
[image]

RAM for HA system
[image]

va4ok, 2022-03-30:
I've removed and install addon one more time and looks it works.

closed 2022-05-08 by github-actions[bot]