Koenkk/zigbee2mqtt #11890: MAC channel access failure (225)

Joly0, 2022-03-17:
### What happened?

After some time of running well, i now get the error message "MAC channel access failure (225)" constantly in zigbee2mqtt whenever i want to do anything with my devices. Furthermore, i cant control them anymore, they dont react to anything, they dont give back any information nor do they show a lqi.

As a coordinator i have this Sonoff Zigbee3.0 USB Dongle and it worked great since i setup this configuration a few weeks ago, but now i cant do anything anymore.

### What did you expect to happen?

My devices should work as intended

### How to reproduce it (minimal and precise)

i dont know

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee3.0 USB Dongle

Koenkk, 2022-03-18:
MAC channel access failure is caused by interference, see https://www.zigbee2mqtt.io/guide/faq/#common-error-codes

Joly0, 2022-03-21:
Looks like that fixed the issue

closed 2022-03-21 by Joly0

Dekyi, 2023-03-12:
I have a rock solid Zigbee mesh but every time I add a new controller to my pS5 or update the firmware on a controller my network crashes and gives me the "MAC channel access failure". The controller a connected via bluetooth and maybe they are useing it the 2.4 GHz band very aggressive. Has someone else noticed this and maybe found a solution to tame the PS5?

dirkjan2302, 2023-04-11:
Also over here with a "slae.sh cc2652rb stick - slaesh's iot stuff" but only the 230V powered devices are not responsive after 3 hours fully functional.
What  can this be and how to solve this, no new devices added for months.

Please advice.

marcgarciamarti, 2023-09-16:
Hello everyone,

I've spent a few hours today trying to figure out what is going on with my HA instance. So far I would say that everything is fine, but I'm having interference issues, as, among all that I see, I tend to observe ```MAC channel access failure``` error in the logs.

Repairing all my devices sounds daunting and I would rather avoid going down that path. Is there anyway I can identify what the offending wifi network is, and nuke it somehow? My wireless router is sitting a few meters away from my coordinator but it has been there for years and I have not experience these issues in the past, so I suspect one of my neighbors may have recently installed a new wireless router that is now causing the issues. I read @Koenkk article above and that has helped me get a better understanding of the issue.

Any tips would be most appreciated.

marcgarciamarti, 2023-09-16:
So, it looks like I managed to solve my issues by unplugging my dongle and plugging it back into my NUC. Apparently, cutting the power supply for a while did good. Sometimes the most stupid thing is what solves the issue. Changing the channels in my 3 wireless networks was not necessary and also my neighbors seem to not have deployed any noisy device either. Rebooting HA and my NUC multiple times did not address this issue as this dongle continued to be powered at all times...

ovidiusas, 2023-10-14:
> So, it looks like I managed to solve my issues by unplugging my dongle and plugging it back into my NUC.
Thank you for the fix! I had the same issue after I powered off/on my raspberry pi (Zigbee2MQTT
 1.33.1-1).

jonnylangefeld, 2024-05-08:
> So, it looks like I managed to solve my issues by unplugging my dongle and plugging it back into my NUC.

Wow I'm so glad, this is what helped me too. After restarting my server a couple of times and even after physically switching off power and on again it wouldn't work. But just unplugging and plugging back in the dongle (and restarting zigbee2mqtt afterwards) all zigbee devices came back 🎉

atle85, 2024-09-24:
Just want to chime in if someone else has the same problem. Zigbee stick is Sonoff P on a extended USB cord. My system also suddenly 90% blacked out. All connections dropped. Lot's of MAC errors. My wifi was running on channel 1 while Zigbee on 25. My neighbours wifi might affect slightly, but not THAT much. I tried restarting the system multiple times (both Zigbee2Mqtt module of Home Assistant and the entire device rebooted). No effect. Let the system run over night - no effect. I don't know exactly WHAT solved it, but I did the following the last time:
1.) Plugged out and in the Zigbee stick (thought it might power the internals of the Zigbee stick totally off and on)  
2.) Restarted all mains powered Zigbee devices (5 seconds off, then power on again)
3.) Restarted my wifi network AND hard restarted the more dumb mesh extenders by plugging out and didn't plug them in before Zigbee was up and running again (Amplifi HD)
  
If I had to guess, it might be the last part - one of the mesh extenders going crazy. However, I'm really not sure. Now everything is up and beating again. Been running for several hours now.

mikkelsiggaard, 2024-11-14:
> So, it looks like I managed to solve my issues by unplugging my dongle and plugging it back into my NUC. Apparently, cutting the power supply for a while did good. Sometimes the most stupid thing is [...]

I just had similar issues. Disconnected and connected Zigbee coordinator, restarted Z2M and some guy named Bob is now my uncle.

marcgarciamarti, 2024-11-15:
> I just had similar issues. Disconnected and connected Zigbee coordinator, restarted Z2M and some guy named Bob is now my uncle.

Glad it helped!!

michaelh99, 2024-12-01:
Ran into the same message after having trouble controlling some devices, saw this thread and "fixed" the problem by rebooting the proxmox instance where HA is running

rolfvreijdenberger, 2025-03-28:
just an FYI for others finding this thread.

interference is electromagnetic, so it can come from many sources. 
logically check sources and begin with the zigbee adapter itself.
I recently unintendedly tested interference for an slzb-06 coordinator.
Having it close to a cpu, a power supply, a wifi access point, microwave etc might all cause it.

I found out because 'when cleaning up' my cable mess, I decided to stick my slzb-06's on a 24 port switch. I have 2 zigbee networks and thus 2 coordinators. both of them on different zigbee channels. I stuck them both on the switch with double sided tape. only 1 started giving errors, and this was stuck right on the internal power supply of the switch. moved it away (ethernet cable of 50cm) and immediately all was fine. The other one is still stuck to the side of the switch where the powercord is not entering the switch and is doing just fine.