Koenkk/zigbee2mqtt #11891: Value of measured Air Pressure is over 1100

T-IOOO, 2022-03-17:
### What happened?

I have two WSDCGQ11LM temperature sensors. Homebridge log is giving me a warning from the characteristic for ’Air Pressure’. In zigbee2mqtt the value is sometimes way too high 
[image]

### What did you expect to happen?

Values lower than 1100 hpa 

### How to reproduce it (minimal and precise)

It seems to behave randomly. Homebridge log is beeing flooded with characteristic warnings

### Zigbee2MQTT version

1.24.0-dev commit: cae83c90

### Adapter firmware version

20220219

### Adapter

Slae.sh

Koenkk, 2022-03-18:
Could you provide the debug log when incorrect measurements are published?

Koenkk, 2022-03-18:
Could you check if this has been fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4016 ?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

T-IOOO, 2022-03-18:
I will check as soon as u

closed 2022-03-18 by T-IOOO

T-IOOO, 2022-03-18:
> Could you check if this has been fixed by [Koenkk/zigbee-herdsman-converters#4016](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4016) ?
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Running 1.24.0-dev commit: 0cd02c74
I can confirm it's working now. No warnings in Homebridge!

Good Job!