Koenkk/zigbee2mqtt #11893: Wrong device information

dexsta, 2022-03-18:
### What happened?

So I'm trying to add this device which says it is supported.

[image]

However, the device is coming up as a Water leak detector ?

[image]

The Model #'s are nearly the same, except my device has `_repeater` at the end.

[image]

And in the Map option shows this

[image]

How would I go about fixing this, or is it a Bug ? Tried many times removing/re-adding the device. 

Thanks.

### Zigbee2MQTT version

1.18.1 commit: 1.18.1

### Adapter firmware version

20211217

### Adapter

Sonoff  ZigBee 3.0 USB Dongle plus

Koenkk, 2022-03-18:
you are using an outdated z2m version.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2022-03-18 by Koenkk

dexsta, 2022-03-18:
@Koenkk thanks for replying. Please excuse my dumbness, couple questions. I just installed this like yesterday, would it not already come with the latest version? I have turned on Auto update but still the same. Or I need to do something else?

[image]

So 124.0-1 is that the latest ? Big difference to what I have at the moment.

[image]

Thanks for your assistance 👍

dexsta, 2022-03-18:
Ok, so I had to change to the new Repo. 👌 Can close.