Koenkk/zigbee2mqtt #11895: [New device support]:

frankis78, 2022-03-18:
### Link

https://www.connecte.no/termostat

### Database entry

{"id":29,"type":"Router","ieeeAddr":"0x2c1165fffe9d2f68","nwkAddr":42065,"manufId":4098,"manufName":"_TZE200_4hbx5cvx","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,4,5,513,3,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_4hbx5cvx","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b002390da35","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1647595450552,"defaultSendRequestWhen":"immediate"}

### Comments

Hi,
Can someone help me to create a converter for This Underfloor Thermostat?
https://www.connecte.no/termostat
[For.utviklere.og.avanserte.smarthus.systemer.som.krever.koding.av.Connecte.termostat.docx](https://github.com/Koenkk/zigbee2mqtt/files/8303328/For.utviklere.og.avanserte.smarthus.systemer.som.krever.koding.av.Connecte.termostat.docx)

I think maybe this converter can be used as a base.
https://gist.github.com/geoffreylagaisse/bb90214dbaf687465e5841f49685cf66?fbclid=IwAR15VWWydzMFwawYACieviTSE4DtvjEaMOQ0WqZHJ-kVlK5gHHF4VPV_gaI#file-tv01-zg-js

The Thermostat went into system as unsupported
Device type: Router
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_4hbx5cvx
Support status: Not supported

Thank you

frankis78, 2022-03-23:
In manual https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#instructions
it tells you to add datapoint into tuya.js file, and add the special commands into the fromzigbee and tozigbee file, and then update your external converter file.

Can somebody tell me if i need to add data to the 3 files (tuya.js, fromzigbee.js, tozigbee.js) while testing, or can i use only the external converter file until im finished with testing, and then add from external converter file in to the 3 base files?

frankis78, 2022-03-24:
I don't get anything in the log other than Received Zigbee message from 'Termostat stue', type 'attributeReport', cluster 'genBasic', data '{"appVersion":67}' from endpoint 1 with groupID 0

Can someone help out?
This is what I'm using right now.
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const tuyaLocaldataPoints = {
        connecte_Switch: 1,
        connecte_Mode: 2,
        connecte_Temp_set: 16,
        connecte_Temp_current: 24,
        connecte_Temp_correction: 28,
        connecte_Child_lock: 30,
        connecte_Temp_floor: 101,
        connecte_Sensor_type: 102,
        connecte_Temp_activate: 103,
        connecte_Load_status: 104,
        connecte_Temp_program: 105,
        connecte_Open_window: 106,
        connecte_Max_protect_temper: 107,
        connecte_Sn: 199,
};

const fzLocal = {
    connecte_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
        const dpValue = tuya.firstDpValue(msg, meta, 'connecte_thermostat');
        const dp = dpValue.dp;
        const value = tuya.getDataValue(dpValue);
            switch (dp) {
            case tuyaLocaldataPoints.connecte_Switch:
                console.log("Connecte Switch " + value + "\n");
                return {switch: {true: "True", false: "False"}[value]};
            case tuyaLocaldataPoints.connecte_Mode:
                console.log("Connecte Mode" + value + "\n");
                return {mode: value};
            case tuyaLocaldataPoints.connecte_Temp_set:
                console.log("Connecte Temp Set " + value + "\n");
                return {temp_set: value};
            case tuyaLocaldataPoints.connecte_Temp_current:
                console.log("Connecte Temp Current " + value + "\n");
                return {temp_current: value};
            case tuyaLocaldataPoints.connecte_Temp_correction:
                console.log("Connecte Temp Correction " + value + "\n");
                return {temp_correction: value};
            case tuyaLocaldataPoints.connecte_Child_lock:
                console.log("Connecte Child Lock " + value + "\n");
                return {child_lock: {true: "True", false: "False"}[value]};
            case tuyaLocaldataPoints.connecte_Temp_floor:
                console.log("Connecte Temp Floor " + value + "\n");
                return {temp_floor: value};
            case tuyaLocaldataPoints.connecte_Sensor_type:
                console.log("Connecte Sensor Type " + value + "\n");
                return {sensor_type: value};
            case tuyaLocaldataPoints.connecte_Temp_activate:
                console.log("Connecte Temp Active " + value + "\n");
                return {temp_active: value};
            case tuyaLocaldataPoints.connecte_Load_status:
                console.log("Connecte Load Status " + value + "\n");
                return {load_status: {true: "True", false: "False"}[value]};
            case tuyaLocaldataPoints.connecte_connecte_Temp_program:
                console.log("Connecte Temp Program " + value + "\n");
                return {temp_program: value};
            case tuyaLocaldataPoints.connecte_Open_window:
                console.log("Connecte Open Window " + value + "\n");
                return {open_window: {true: "True", false: "False"}[value]};
            case tuyaLocaldataPoints.connecte_Max_protect_temper:
                console.log("Connecte Max Protect Temper " + value + "\n");
                return {max_protect_temper: value};
            case tuyaLocaldataPoints.connecte_Sn:
                console.log("Connecte SN Nr " + value + "\n");
                return {sn: value};

            default:
                meta.logger.warn(`zigbee-herdsman-converters:connecte_thermostat: Unrecognized DP #${ dp} with data ${JSON.stringify(msg.data)}`);
            }
        },
    },
};

const definition = {
    zigbeeModel: ['hbx5cvx'],
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_4hbx5cvx'
        },
    ],
        model: 'Smart-Termostat',
        vendor: 'Connecte',
        description: 'Underfloor heating thermostat',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
        fzLocal.connecte_thermostat
    ],
    toZigbee: [
        tz.tuya_data_point_test // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [],
    //[
    //    e.switch(), e.mode(), e.temp_set(), e.temp_current(), e.temp_correction(), e.child_lock(),e.temp_floor(),
    //    e.sensor_type(),e.temp_active(),e.load_status(),e.temp_program(),e.open_window(),e.max_protect_temper(),e.sn()
    //],
};       
module.exports = definition;`

owangen, 2022-03-30:
I have create support for fromZigbee for this device, but have problem with toZigbee.
When trying to run a converter for toZigbee for a value, I'm using tuya.sendDataPointValue, that uses tuya.sendDataPoints. This gives me this error: "Error: Cluster 'manuSpecificTuya' has no command 'dataRequest'"
I have looked at all of the different converters for other Tuya device, but no one gives me any hint on how to fix this.

The documentation [here](https://github.com/Koenkk/zigbee2mqtt/files/8303328/For.utviklere.og.avanserte.smarthus.systemer.som.krever.koding.av.Connecte.termostat.docx) points to the following:

"In short, you can set a datapoint value by using the `TY_DATA_REQUEST` command (0x0 or 0), with as payload a sequence number and a data unit. And yes, I agree that the command name is confusing.
Below is how I defined the `TY_DATA_REQUEST` command for the Homey platform. Note that this is a TS abstraction, but the defined `args` are serialized into their bitwise representation and pasted after each other.
 
  ```
dataRequest: {
    id: 0x0,
    frameControl: ['clusterSpecific', 'disableDefaultResponse'],
    args: {
      sequenceNumber: ZCLDataTypes.uint16,
      dataPoint: ZCLDataTypes.uint8,
      dataType: ZCLDataTypes.uint8,
      length: ZCLDataTypes.data16,
      data: ZCLDataTypes.buffer,
    },
```
  },
"
The documentation referenced above also states:
        a. TY_DATA_ REQUEST command is used to write data point values
        b. TY_DATA_REPORT command is used by the thermostat to report a data point value to the gateway (marked as ` 
            directionToClient`).
        c. The “DP Data Format” is important, note that all number are transferred as Big-Endian.
        d. TUYA_MCU_SYNC_TIME to set the thermostat time. Needs to have both UTC Unix timestamp in seconds and that same 
            timestamp but then adjusted to the local timezone.

Questions:
1. Is this "id: 0x0" something special for this Tuya device? Or do anyone else see what causes the error `Error: Cluster 'manuSpecificTuya' has no command 'dataRequest'`
2. Setting the time for this device gives does not work, nothing happens when using `onEvent: tuya.onEventSetLocalTime` or `onEvent: tuya. onEventSetTime`. Any clue why?
3. Any pointers on how I could proceed to be able to send commands to this device?

owangen, 2022-04-03:
@hacker-cb could you help me on what I could try to get further?
1. onEvent: tuya.onEventSetLocalTime or onEvent: tuya. onEventSetTime seemingly does nothing. I notice "No converter available for 'TS0601' with cluster 'manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":0}'" in the logs.
2. I'm getting "Error: Cluster 'manuSpecificTuya' has no command 'dataRequest'" when sending zigbee to this device. According to the developer information for this device, dataRequest should be correct as far as I can understand? https://github.com/Koenkk/zigbee2mqtt/files/8303328/For.utviklere.og.avanserte.smarthus.systemer.som.krever.koding.av.Connecte.termostat.docx

I have ordered a tuya gateway so I can sniff the traffic, but it takes time before it arrives.

owangen, 2022-04-06:
Note to self: Update to latest zigbe2mqtt before developing converts

closed 2022-04-07 by frankis78