Koenkk/zigbee2mqtt #11899: Issue with Lidl HG06336 (window and door sensor) after changing batteries

geolin, 2022-03-18:
### What happened?

After joining the above mentioned window and door sensor everything worked fine. After pulling out the batteries, the sensor won't rejoin again. It keeps sending "device_announce" and "device_announced".
I enabled the availability setting but that didn't change anything. 
The only way I managed to get the sensor working again is to reset the sensor and rejoin it to the network.
But that shouldn't be the way to go for when the batteries have been replaced.

### What did you expect to happen?

After changing the batteries the sensor should rejoin the network.

### How to reproduce it (minimal and precise)

Join the sensor. Take out the batteries. Wait for a couple of minutes. Put in the batteries again. The adapter won't rejoin the network but will send the "device_announce" "device_announced" messages. For sending these messages you have to remove the magnet from the sensor and then push it back to the sensor.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26610700

### Adapter

RaspBee2

### Debug log

Info MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"eiac2-1"}'
Debug Device '0x842e14fffeef6080' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842e14fffeef6080","ieee_address":"0x842e14fffeef6080"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x842e14fffeef6080"},"type":"device_announced"}'
Debug Retrieving state of '0x842e14fffeef6080' after reconnect

closed 2022-04-25 by github-actions[bot]

McMornan, 2022-11-07:
Hi, I second that. I have exactly the same issue. I use a slae.sh stick as coordinator. Latest development version of zigbee2mqtt running in a docker container. (as of nov, 6th, 2022)

The door contact runs fine for months. When the battery needs replacement the sensor does not join the network on his own. I need to manually delete the sensor using the zigbee2mqtt gui. Repairing works without issues. till next time ...

Any idea what I could check ?

Bumekye, 2023-05-27:
Hi,
Got the same issue. Has anyone solved it yet?
Delete it and repairing works, but… you know…

robertklep, 2023-09-19:
I don't see how z2m can force the device to rejoin the network, the issue seems to be with the device's firmware (I also have one that exhibits the same behaviour).

kBrausew, 2024-02-05:
The same behavior occurs for me with ioBroker and Herdsman Converter

Mangar2005, 2024-02-17:
Same issue for me with the Lidl/Silvercrest door sensors.
Sonoff Dongle Plus + Zigbee2MQTT.  All my other Zigbee-devices working perfect.
I just changed to HA from Homey Pro, where I didn´t had any issues with them. This gives some hope the issue could be fixed?

jo8be, 2025-02-04:
I can confirm the problem.
I have coneccted the switch to Home Assistant.

Looking at the Exposes tab of the device under Zigbee2MQTT:

Once configured it says:
Contact: Open
Tamper: Clear

With magnet it goes to:
Contact: Closed
Tamper: Clear

Removing the mounting plate:
Contact: Open
Tamper: Tampered

With magnet and removed plate:

Contact: Closed
Tamper: Tampered

Mounting tha plate again:

Contact: Open
Tamper: Clear

With magnet it goes again to:
Contact: Closed
Tamper: Clear

Removing the batteries for 30 sec and re-inserting:

Device lost connection. Only deleting it from HA and reconnecting it will help.

I also did a test with fixing the tamper switch with tape while removing/readding the batteries, but that did not change the behavior.

My conlclusion:

The problem comes from the loss of supply voltage, not from the tamper switch.
The device is garbage. That is probably the reason, why it is no longer available at Lidl.