Koenkk/zigbee2mqtt #11903: TuYa garage door opener PJ-ZGD01 does not work as intended

cody82, 2022-03-19:
### What happened?

When I trigger the garage door opener the relay only activates if the sensor has contact. So you can only open your garage door.

### What did you expect to happen?

I would expect the relay to trigger regardless if the sensor is contacted or not. I want to open and close the door.

### How to reproduce it (minimal and precise)

* Pair the device
* Close the contact sensor
* Trigger the device, e.g. {"trigger":True}
* The relay clicks
* Open the contact sensor
* Trigger the device, e.g. {"trigger":True}
* The relay does not do anything

### Zigbee2MQTT version

1.23.0 commit: 6ea61bae

### Adapter firmware version

20210120

### Adapter

zzh!

Koenkk, 2022-03-20:
@[octa22](https://github.com/octa22) could you check this?

jensstruemper, 2022-03-22:
Hi! This was a known issue in version 1.23 but has been fixed in 1.24.

closed 2022-03-22 by cody82