Koenkk/zigbee2mqtt #11904: paulmann-500.45 'blinks' every 400s

uafcybs, 2022-03-19:
### What happened?

Once integrated into the system, the [Paulmann 500.45 led controller](https://www.zigbee2mqtt.io/devices/500.45.html#paulmann-500.45) works fine, except that it turns on at full brightness for 0.5 seconds every 400 seconds...
The problem is not in the controller because:
a) it does not "blink" if not connected to the zigbee network
b) this is the second such controller with which I have this problem

### What did you expect to happen?

Controller does not do what is not requested.

### How to reproduce it (minimal and precise)

1) Integrate  [Paulmann 500.45 led controller](https://www.zigbee2mqtt.io/devices/500.45.html#paulmann-500.45) (with attached led strip) into zigbee2mqtt
2) make sure the attached led is in "off"state
3) after some (several hours) time led strip will flash at full brightness for ~0.5. s. It will repeat every 400s

### Zigbee2MQTT version

1.24.0-1

### Adapter firmware version

20201026

### Adapter

slaesh's cc2652rb stick

### Debug log

[zigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/8310113/zigbee2mqtt.log)
**KitchenWorktopLight** - is the Paulmann 500.45 controller

panther-sh, 2022-03-27:
Hi uafcybs,
the same device works in my z2m environment as expected. For me it seems not possible to reproduce your described problem.

zb Device:	Paulmann 500.45 Maxled Schalt/Dimm Controller max. 144W 24V DC
z2m Ver.:	1.24.0
Adapter:		SONOFF Zigbee 3.0 USB Dongle Plus
Typ:			zStack3x0 
FW:			20211217

uafcybs, 2022-04-29:
Just received a SONOFF stick - hope it will work better

RikScheffer, 2022-05-03:
I have recently bought 15 of these lamps, and experience it with all of them. Funny thing is, they all do this at almost exactly the same time. However, not all of them blinks every 400 seconds, sometimes it is just a single one or several and sometimes almost all of them.

uafcybs, 2022-05-03:
I replaced the slaesh’ stick with SONOFF. Did not help …

RikScheffer, 2022-05-03:
I just did some tests.
My results:
- turning off zigbee2mqtt stopped all the blinks. So that suggests the cause is somewhere in the controller part of the setup.
- after turning it back on, the number of blinks is greatly reduced, it now is about 1 lamp every 400 seconds, in stead of ~8 lamps. Before I already had the feeling that over time it got worse, that now seems confirmed.

One more remark:
I also occasionally see the lights blink `off` when they are turned on. But that happens a lot less often then them blinking on when they are turned off.

closed 2022-06-11 by github-actions[bot]

tbrtje, 2022-10-03:
I noticed the same issue for my device. Were you able to resolve it?

RikScheffer, 2022-10-03:
Not yet

yrobert, 2022-11-28:
I got the same issue with my CC2531

mpmario, 2023-09-03:
I experience the same behaviour with paulmann 500.47. Found no solution yet...

RikScheffer, 2023-09-22:
my solution was to disable the `permit_join` option in the configuration, that way the blinks are gone