Koenkk/zigbee2mqtt #11908: Wrong device icons since update to  1.24.0-dev commit: 02bc325

nohn, 2022-03-20:
### What happened?

Since updating to 1.24.0-dev commit: 02bc325, device Icons are wrong:

[image]

### Zigbee2MQTT version

 1.24.0-dev commit: 02bc325

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB Stick

mpuff, 2022-03-20:
I have this problem also with the stable version 1.24.0

GerdRuetten, 2022-03-20:
The webserver under https://www.zigbee2mtqq.io is down and this response to wrong devie icons

nohn, 2022-03-20:
Is there a reason, the icons aren't distributed with the software and loaded locally?

GerdRuetten, 2022-03-20:
> Is there a reason, the icons aren't distributed with the software and loaded locally?

Imho the reason is the fast integration of new devices.

So it's easier to serve the images and other informations from a central server, than allways update a local instance.

nohn, 2022-03-20:
That doesn't make sense to me: For new device support the local instance has to be updated anyway.

sjorge, 2022-03-20:
Having the option to fetch the icons locally, say by installing an extra npm module or something would indeed be great.
Even if the missing icons don't bother you, the delay before it gives up trying to fetch them slows down the entire frontend.

Koenkk, 2022-03-20:
zigbee2mqtt.io seems up again so this should be fixed.

itec78, 2022-03-21:
I think that getting images from zigbee2mqtt.io could be used for tracking and a profiling users and devices.
It would be better to have all the files locally.

itec78, 2022-04-21:
I don't know what are the next steps to follow.
Maybe a new issue can be created, asking to serve images locally instead of using zigbee2mqtt.io website?
I found that there are 2054 images, for a total of 12,4 MB. I think it's not a problem serving them locally.
https://github.com/Koenkk/zigbee2mqtt.io/tree/master/public/images/devices

WoLpH, 2022-04-25:
Another vote here for serving them locally...

The current implementation is quite annoying in my case:
My modem is near the cable connection which is outside of my house. To be able to power cycle the device remotely, I've connected it to a zigbee switchable plug. As you can probably guess... this needs to happen when my internet is down so I can't reach the server at that point and the web interface stalls for a few minutes because of it.

As a workaround I've blocked the image url with umatrix, but that's just annoying :)

WoLpH, 2022-05-26:
bump for a fix that only includes local resources

itec78, 2022-05-26:
I found that the code that points to external address is here:
https://github.com/nurikk/zigbee2mqtt-frontend/blob/dev/src/components/device-image/index.tsx#L19
Is ok to have the issue here, or is it better to open it in frontend? (https://github.com/nurikk/zigbee2mqtt-frontend)

I think that 2 steps are required:
1) copy the images from https://github.com/Koenkk/zigbee2mqtt.io/tree/master/public/images/devices to local path. (could be done somewhere here https://github.com/nurikk/zigbee2mqtt-frontend/blob/dev/.github/workflows/release.yml)
2) change the getZ2mDeviceImage variable to point local resources

After that the changes will be deployed to zigbee2mqtt automatically
https://github.com/Koenkk/zigbee2mqtt/blob/master/.github/workflows/update_frontend.yml#L17

Am I wrong?

nohn, 2022-05-26:
There is already an issue open and the frontend project: https://github.com/nurikk/zigbee2mqtt-frontend/issues/1477

WoLpH, 2022-06-27:
@nohn does the other issue make this one obsolete?

nohn, 2022-06-27:
.

itec78, 2022-06-27:
I think this issue is never obsolete until it will be solved. There's a privacy problem. And doesn't make sense to have remote contents on a fully local application.

nohn, 2022-06-27:
@itec78 you are 100% correct. But technically speaking this isn't an issue in zigbee2mqtt and can't be fixed there but needs to be fixed in [zigbee2mqtt-frontend](https://github.com/nurikk/zigbee2mqtt-frontend): https://github.com/nurikk/zigbee2mqtt-frontend/issues/1477

closed 2022-08-06 by github-actions[bot]

Hypfer, 2023-09-27:
https://github.com/nurikk/zigbee2mqtt-frontend/blob/a2e3dc31110b33aa5fe38e5e8ec7970aabf315bf/src/components/device-image/index.tsx#L19

What's stopping anyone from extending this function (that already exists!) to take an optional configuration parameter into account where you can specify a custom asset host?
That can't be more annoying than fighting with the stale bot