Koenkk/zigbee2mqtt #11909: Lixee No price exposed

kevincaradant, 2022-03-20:
### What happened?

I don't have price displayed in the payload

### What did you expect to happen?

[image]

I want a value on the 'current_price' field

### How to reproduce it (minimal and precise)

Just waiting info from Linky but nothing happens

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211210

### Adapter

Launchpad CC1352-P2

### Debug log

```
 Received Zigbee message from 'linky', type 'readResponse', cluster 'liXeePrivate', data '{"currentTarif":"BASE\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'linky', type 'readResponse', cluster 'haElectricalMeasurement', data '{"rmsCurrentMax":90}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'linky', type 'readResponse', cluster 'seMetering', data '{"activeRegisterTierDelivered":"TH..","meterSerialNumber":{"data":[48,51,49,57,54,49,56,57,51,52,56,48],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'linky', type 'readResponse', cluster 'haMeterIdentification', data '{"availablePower":30}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'linky', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"apparentPower":1410,"rmsCurrent":6}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'linky', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,8443110]}' from endpoint 1 with groupID 0

Info MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_power":-32768,"active_power_max":-32768,"active_power_ph_b":-32768,"active_register_tier_delivered":"TH..","apparent_power":1410,"available_power":30,"current_date":"","current_price":"","current_summ_delivered":8443.1,"current_tarif":"BASE","current_tier10_summ_delivered":0,"current_tier1_summ_delivered":8443.1,"current_tier2_summ_delivered":0,"current_tier3_summ_delivered":0,"current_tier4_summ_delivered":0,"current_tier5_summ_delivered":0,"current_tier6_summ_delivered":0,"current_tier7_summ_delivered":0,"current_tier8_summ_delivered":0,"current_tier9_summ_delivered":0,"linkquality":116,"message1":"","message2":"","meter_serial_number":"xxxxxx","rms_current":6,"rms_current_max":90,"rms_voltage":65535,"site_id":"","status_register":"","update":{"state":"idle"},"update_available":false,"warn_d_p_s":null}'

Koenkk, 2022-03-20:
@vk496 could you check this?

vk496, 2022-03-20:
@kevincaradant Linky mode? Price is available under standard, not historique

kevincaradant, 2022-03-20:
Not sure to understand when should I use standard mode. 
I thought it was for people who want to sell their energy solar panel ect. 

I'm wrong on this?

If I want to know how much I spent by day in electricity, I need to go on standard mode?

Thbak you @vk496

kevincaradant, 2022-03-20:
And @vk496, do you have any idea on how to limit the number of fields exposed by Z2M?
Because in Standard mode, there is so many empty fields that I don't want because I don't need them?
Thanks
Regards

vk496, 2022-03-20:
We are not talking about your zlinky config. We talk about your Linky. The TIC mode by default is historique, which is limited in what it sends.

The zlinky configuration should match your Linky device, but information is provided by Linky

kevincaradant, 2022-03-20:
Oh OK thank you :) 

but for me the price could be added locally ( manually ) and using the real index from linky we should not need to have to use the standard mode like that and it could be possible to have the cost by minute, hour or day :)

Si if I understand well, everything around the price is not possible if we are on historic mode ?

closed 2022-03-22 by kevincaradant