Koenkk/zigbee2mqtt #11910: zigbee2mqtt.io DNS is down

fsevilla3, 2022-03-20:
### What happened?

The zigbee2mqtt.io domain fails to resolve, rendering the website is inaccessible. Even the initial name server lookup doesn't return any results. In addition to a local query that uses Cloudflare DNS (1.1.1.1), I also checked this independently using https://mxtoolbox.com/SuperTool.aspx?action=dns%3azigbee2mqtt.io&run=toolpage.

### What did you expect to happen?

zigbee2mqtt.io to resolve and website to be accessible

### How to reproduce it (minimal and precise)

dig zigbee2mqtt.io ns

### Zigbee2MQTT version

N/A

### Adapter firmware version

N/A

### Adapter

N/A

Koenkk, 2022-03-20:
Seems to work again:

[image]

GerdRuetten, 2022-03-20:
Yes allways work well

closed 2022-03-21 by fsevilla3