Koenkk/zigbee2mqtt #11911: Upgrading from 1.22.2-1 means I cannot add new devices.

technikly, 2022-03-20:
### What happened?

After upgrading from 1.22.2-1 to either 23 or 24, I'm unable to add new new Sonoff ZigBee buttons. Sonoff ZigBee outlet/plugs also fail to add. I also cannot reconfigure existing devices. This is with the conbee 2.

Zigbee2MQTT

Settings
Tools
About
Raw
Translate
Donate
Zigbee2MQTT version
1.22.2 commit: unknown
Coordinator type
ConBee2/RaspBee2
Coordinator revision
0x26720700
Coordinator IEEE Address

Frontend version
0.6.67
Stats
Total 18
By device type
End devices: 15
Router: 3
By power source
Battery: 15
Mains (single phase): 3
By vendor
eWeLink: 13
SONOFF: 3
Samjin: 1
Sunricher: 1
By model
WB01: 9
MS01: 4
S26R2ZB: 2
01MINIZB: 1
multi: 1
ZGRC-KEY-013: 1

### What did you expect to happen?

I expect the device to be recognised when attempting to pair or reconfigure

### How to reproduce it (minimal and precise)

Enter pair mode with conbee 2 home assistant supervisor. Put sonoff ZigBee device in pair mode. Nothing with .23 or .24. works as intended for .22

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

0x26720700

### Adapter

Cobee2

gyver, 2022-03-29:
I might have the same problem : one of my Sonoff Zigbee temp/humidity sensor dropped from my network recently and I couldn't add it back (there is no message in log either, no failure to configure, no trace that the device was even detected). Suspecting a device failure I tried a spare one which failed to pair too.

I tried both devices (original and spare) on another network managed by ZHA and they were both detected (interviewed and configured) properly.

This was on 1.23.0 and I tried upgrading to 1.24.0 (always using git checkout <tag> and npm ci) which failed to detect them too.

More information :
- There are currently 48 devices on the network,
- the adapter used is PiZigate+ on a Raspberry Pi 4.

I'm a bit hesitant trying to revert to 1.22.2 : not sure if downgrading is supported.

I'll try adding another device (not Sonoff) and report if it fails too.

gyver, 2022-03-29:
Note about previous comment : adding an OSRAM [AC03645](https://www.zigbee2mqtt.io/devices/AC03645.html#osram-ac03645) just worked without any issue. So maybe the problem is with Sonoff devices specifically.

technikly, 2022-04-06:
I've upgraded to 1.25. Problem persists. It is all standard SONOFF devices. Buttons and Motion sensors. With the CONBEE II. Problem still doesn't occur at 1.22. The only change is the Zigbee2MQTT version so I think the devices are fine.

technikly, 2022-04-07:
Update - The Samsung MultiSensor also doesn't properly report using the CONBEE II after v.22 
Every time I restore to version.22, all problems are fixed. Is there something I'm doing wrong or is there a firmware I'm missing? Or is this just a bug with Zigbee2Mqtt?

closed 2022-05-16 by github-actions[bot]