Koenkk/zigbee2mqtt #11915: Device Eglo connect.2 12253 is identified as AwoX 33943/33944/33946

itec78, 2022-03-20:
### What happened?

I bought a Eglo connect.2 bulb. https://www.eglo.com/it/lampadine-12253.html
After joining it appears as AwoX 33943/33944/33946.
It's working bad, color temperature max/min setting probably are wrong, and there are problems with color shades (goes from red to violet to blue without intermediate steps .
How can I fix it? I tried retrieving color temperature range like explained in documentation, but I got a standard answer without useful data.
This is the database.db entry:

`{"id":8,"type":"Router","ieeeAddr":"0xa4c13850f407e37a","nwkAddr":44051,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,242,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorMode":1,"currentX":25231,"currentY":10158,"colorTemperature":156}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1647797203832,"defaultSendRequestWhen":"immediate"}
`
What can I do?
Thanks

### What did you expect to happen?

That it was recognized correctly

### How to reproduce it (minimal and precise)

Pair it.

### Zigbee2MQTT version

1.24.0-dev commit: 0cd02c7

### Adapter firmware version

20210708

### Adapter

SONOFF ZigBee 3.0 USB Dongle Plus

Koenkk, 2022-03-21:
Can you confirm the bulb works correctly with another hub? (e.g. hue hub)

itec78, 2022-03-21:
I don't have another hub to test the bulb.
I made some tests using the official app, and all is working fine. But I think it uses bluetooth instead of zigbee

itec78, 2022-03-24:
Looking at this page https://zigbee.blakadder.com/Eglo_12253.html it seems that device was supported properly, but link to page https://www.zigbee2mqtt.io/devices/TLSR82xx.html is broken.

itec78, 2022-03-24:
Yes, colors are working fine in "color_mode":"hs". I think this was the problem.
Probably in temperature mode min and max should be set to 153 and 370.

Just for info, in xy color mode I was getting these colors:
[image]

itec78, 2022-03-24:
I did more tests for brightness, it's working with this sentence:
`{"brightness":127,"color":{"h":170,"s":100}}`
and the answer is
`{"brightness":127,"color":{"h":170,"hue":170,"s":100,"saturation":100,"x":0.153,"y":0.3744},"color_mode":"hs","color_temp":153,"color_temp_startup":65535,"last_seen":"2022-03-24T23:26:30+01:00","linkquality":132,"state":"ON"}`

Koenkk, 2022-03-26:
Cool, enabled the HS color mode (so you will be able to set it from the frontend) and updated the color temp range.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

itec78, 2022-03-26:
Hi.
The color temp is working fine, but I have still problems with colors.
I can set the color from frontend, and using color_hs works fine.
[image]
From desktop and Home Assistant however color_xy is still the default mode.
[image]
How can I set hs as default mode?

Koenkk, 2022-03-26:
You can do this by overriding the HA discovery payload from the zigbee2mqtt `configuration.yaml`, example:

```yaml
devices:
  "0x12345678":
    friendly_name: my_light
    homeassistant:
      light:
        supported_color_modes: ['hs', 'color_temp']
```

itec78, 2022-03-26:
Now it's ok.
Thanks a lot

closed 2022-03-26 by itec78