Koenkk/zigbee2mqtt #11923: [New device support]:

unpack8866, 2022-03-22:
### Link

https://www.spotmau.cn/products-switch.html

### Database entry

MQTT publish: topic 'zigbee2mqtt/0x000d6f000b86628a', payload '{"linkquality":51,"state":"ON","state_16":"OFF","state_l1":"OFF","state_switch":"OFF"}'

### Comments

[image]
[image]

Request to add device support; One gang and two gang switches with neutral line.

### External converter

```shell
const device = {
    zigbeeModel: ['1719SP-PS1-02'],
    model: 'SP-PS1-02',
    vendor: 'Spotmau',
    description: 'Smart wall switch - 1 gang',
    fromZigbee: [fz.ignore_basic_report, fz.on_off,],
    toZigbee: [tz.on_off,],
    exposes: [e.switch().withEndpoint('l1')],
    endpoint: (device) => {
        return {'l1': 16};
    },    
    meta: {configureKey: 1, multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(16), coordinatorEndpoint, ['genOnOff', 'genBasic']);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genLevelCtrl']);        
    },
module.exports = device;

const device = {
    zigbeeModel: ['1719SP-PS2-02'],
    model: 'SP-PS2-02',
    vendor: 'Spotmau',
    description: 'Smart wall switch - 2 gang',
    fromZigbee: [fz.ignore_basic_report, fz.on_off,],
    toZigbee: [tz.on_off,],
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right')],
    endpoint: (device) => {
        return {'left': 16, 'right': 17};
    },    
    meta: {configureKey: 1, multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(16), coordinatorEndpoint, ['genOnOff', 'genBasic']);
        await reporting.bind(device.getEndpoint(17), coordinatorEndpoint, ['genOnOff', 'genBasic']);
    },
module.exports = device;
```

Koenkk, 2022-03-22:
Does it work with the external converter you posted?

unpack8866, 2022-03-23:
Yes, it works perfectly under the external converter!
[image]

[image]

closed 2022-03-23 by Koenkk

Koenkk, 2022-04-02:
The line you crossed out is not there: https://github.com/Koenkk/zigbee-herdsman-converters/blob/1cd9cd749caf36d43307cd4ca94fa48385c509e2/devices/spotmau.js#L14 , what should be changed?

unpack8866, 2022-04-03:
Update the picture of 2 gang switch

[image]

Koenkk, 2022-04-03:
updated the image