Koenkk/zigbee2mqtt #11924: Setting position of IKEA blinds result in endless GET-requests

gukj, 2022-03-22:
### What happened?

So I've got 7 Fyrtur blinds setup using Zigbee2MQTT and Homebridge. The blinds work fine most of the time, but seems to sometimes end up in a weird loop where Z2M constantly sends (receives?) a request to get the current position.

I've tried to find a reliable way to reproduce the issue. but there are a lot of moving parts so I'm not sure. My feeling is that it's got something to do with the blinds being set to a position and then all communication with that blind is stopped as its just communicating its current position. I've managed to "freeze" it at other positions as well.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

 0x26580700

### Adapter

ConBee II

### Debug log

Debug log from Z2M:
```
Debug Received MQTT message on 'zigbee2mqtt/0x842e14fffe5a15de/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to 'BL04'
Debug Received MQTT message on 'zigbee2mqtt/0x842e14fffe5a15de/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to 'BL04'
Debug Received MQTT message on 'zigbee2mqtt/0x842e14fffe5a15de/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to 'BL04'
Debug Received MQTT message on 'zigbee2mqtt/0x842e14fffe5a15de/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to 'BL04'
```

Corresponding debug log from Homebridge:
```
[22/03/2022, 14:36:35] [zigbee2mqtt] Publish to 'zigbee2mqtt/0x842e14fffe5a15de/get': '{"position":0}'
[22/03/2022, 14:36:39] [zigbee2mqtt] Publish to 'zigbee2mqtt/0x842e14fffe5a15de/get': '{"position":0}'
[22/03/2022, 14:36:43] [zigbee2mqtt] Publish to 'zigbee2mqtt/0x842e14fffe5a15de/get': '{"position":0}'
[22/03/2022, 14:36:47] [zigbee2mqtt] Publish to 'zigbee2mqtt/0x842e14fffe5a15de/get': '{"position":0}'
```

gukj, 2022-03-22:
That does make sense. I was looking at them backwards. Thanks!

closed 2022-03-22 by gukj

gukj, 2022-03-22:
This might be a really stupid question, but looking at it from Homebridge, it seems like Homebridge thinks the blind is in position:0 (when its actually in position: 100). Does Z2M not publish the location as 0%? Or does it simply echo what Homebridge asks? I'm not sure I "get" what the get-command is for if its not a question from HB to Z2M.

gukj, 2022-03-22:
@sjorge

sjorge, 2022-03-22:
It depends on the blind unfortunately, e.g. Ubisys J1 will publish update via reporting even when it's moving, but the Azis GEAR for example will only publish an update via reporting once it has stopped moving. These are the 2 cover devices I own, there might be some that don't report there position at all.

That for the device -> z2m updates via reporting. However ...

When z2m received a `zigbee2mqtt/<device>/set/position` it will will send the message to the device to update it's position, but if `optimistic` device option (see https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#common-device-options) is set to true, it will also publish a update with the position that was just send.

To further complicate things, not all devices agree with the ZCL which means open vs closed, some have it 0=open,100=closed and other have it 100=open,0=closed (fun times), a lot of cover devices have a device option to invert it called `invert_cover`.

When it z2m receives a `zigbee2mqtt/<device>/get/position` if the device supports it, it will try to read the actual position from the device, and then publish an update.

But based on your logs, something is asking z2m to do a get for the position, presumable homebridge.

gukj, 2022-03-22:
Thank you for taking the time to look into this!

I'm using these blinds (FYRTUR E1757). Both Homebridge and Z2M seems to agree on both target and current position and there doesn't seem to be a conflict in what percentage represents what - HB and Z2M both agree that 100% = open and 0% = closed.

I've tried setting them up with `optimistic: false` and `optimistic: true`. It doesn't seem to make any difference.

I'll try my luck over in the Homebridge repo and see if someone there might be able to help me look into it.

gukj, 2022-03-23:
For anyone stumbling upon this issue in the future:

It seems my issue was resolved by res-syncing the blinds and not giving them friendly names. There still seems to be messages going back and forth about some blinds being in position: 0 even when they're not, but it doesn't stop the blinds from getting new instructions.

This is what I'm seeing in the Z2M log:
’’’
Debug Received MQTT message on 'zigbee2mqtt/0x000d6ffffe8e9405/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to '0x000d6ffffe8e9405'
Debug Received MQTT message on 'zigbee2mqtt/0x680ae2fffe12a6e7/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to '0x680ae2fffe12a6e7'
Debug Received Zigbee message from '0x000d6ffffe8e9405', type 'readResponse', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe8e9405', payload '{"battery":85,"linkquality":255,"position":100,"update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from '0x680ae2fffe12a6e7', type 'readResponse', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe12a6e7', payload '{"battery":99,"linkquality":255,"position":100,"update":{"state":"idle"},"update_available":false}'
Debug Received MQTT message on 'zigbee2mqtt/0x000d6ffffe8e9405/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to '0x000d6ffffe8e9405'
Debug Received MQTT message on 'zigbee2mqtt/0x680ae2fffe12a6e7/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to '0x680ae2fffe12a6e7'
Debug Received Zigbee message from '0x000d6ffffe8e9405', type 'readResponse', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe8e9405', payload '{"battery":85,"linkquality":255,"position":100,"update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from '0x680ae2fffe12a6e7', type 'readResponse', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe12a6e7', payload '{"battery":99,"linkquality":255,"position":100,"update":{"state":"idle"},"update_available":false}'
Debug Received MQTT message on 'zigbee2mqtt/0x000d6ffffe8e9405/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to '0x000d6ffffe8e9405'
Debug Received MQTT message on 'zigbee2mqtt/0x680ae2fffe12a6e7/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to '0x680ae2fffe12a6e7'
Debug Received Zigbee message from '0x000d6ffffe8e9405', type 'readResponse', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe8e9405', payload '{"battery":85,"linkquality":255,"position":100,"update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from '0x680ae2fffe12a6e7', type 'readResponse', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe12a6e7', payload '{"battery":99,"linkquality":255,"position":100,"update":{"state":"idle"},"update_available":false}'
Debug Received MQTT message on 'zigbee2mqtt/0x000d6ffffe8e9405/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to '0x000d6ffffe8e9405'
Debug Received MQTT message on 'zigbee2mqtt/0x680ae2fffe12a6e7/get' with data '{"position":0}'
Debug Publishing get 'get' 'position' to '0x680ae2fffe12a6e7'
Debug Received Zigbee message from '0x000d6ffffe8e9405', type 'readResponse', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe8e9405', payload '{"battery":85,"linkquality":255,"position":100,"update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from '0x680ae2fffe12a6e7', type 'readResponse', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe12a6e7', payload '{"battery":99,"linkquality":255,"position":100,"update":{"state":"idle"},"update_available":false}'
```

alex3305, 2022-09-12:
I was having the same issue(s) with these blinds. Especially one of them. However not giving them friendly names - as suggested by @gukj - makes the device unusable for me, as Z2M reports the blind as _Offline_.

However, force removing them and restarting Zigbee2MQTT and than adding blinds has resolved this. Making the blinds controllable from Z2M and Home Assistant.

alex3305, 2022-09-18:
About a week later and I'm having the same issues again with one of my curtains. So it hasn't been solved I suppose. I have seen about the same messages in the logging:

```
Zigbee2MQTT:info  2022-09-18 12:25:46: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":76,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:25:52: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":78,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:18: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":92,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:18: MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"battery":100,"device_temperature":25,"illuminance":104,"illuminance_lux":104,"linkquality":255,"occupancy":false,"power_outage_count":79,"temperature":26,"voltage":3055}'
Zigbee2MQTT:info  2022-09-18 12:26:19: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":90,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:20: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":87,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:21: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":89,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:22: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":86,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:33: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffed2eed4","ieee_address":"0x804b50fffed2eed4"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-09-18 12:26:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x804b50fffed2eed4"},"type":"device_announced"}'
Zigbee2MQTT:info  2022-09-18 12:26:35: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":100,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:35: MQTT publish: topic 'zigbee2mqtt/Bedroom Repeater', payload '{"linkquality":255,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:37: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":99,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:38: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":94,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:39: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":89,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:40: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":100,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:41: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":94,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:42: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":88,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-18 12:26:43: MQTT publish: topic 'zigbee2mqtt/0x804b50fffed2eed4', payload '{"battery":85,"linkquality":255,"position":81,"state":"OPEN","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:error 2022-09-18 12:26:43: Publish 'set' 'state' to '0x804b50fffed2eed4' failed: 'Error: Command 0x804b50fffed2eed4/1 closuresWindowCovering.upOpen({}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:info  2022-09-18 12:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x804b50fffed2eed4' failed: 'Error: Command 0x804b50fffed2eed4/1 closuresWindowCovering.upOpen({}, {\"sendWhen\":\"active\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x804b50fffed2eed4"},"type":"zigbee_publish_error"}'
[... cut ...] nse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"0x804b50fffed2eed4"},"type":"zigbee_publish_error"}'
```

And as you can see I have removed the device on the end of the logging. I will again re-pair this curtain and try to see if that resolves it. Pretty annoying little issue 😒 .

gukj, 2022-12-30:
@alex3305 Not sure if this is still an issue for you, but I've found that simply removing the battery from the blind and waiting for a minute seems to convince the blind that it has to wake up and start talking to the system. I've also gotten most of our blinds that were a couple years old replaced at IKEA and I'm fairly certain that the newer blinds don't experience this issue. At least not as much.

syncnj, 2023-01-02:
@gukj interesting, I'm currently having this issue, where blinds go up and down after issuing a set position command, like https://github.com/Koenkk/zigbee2mqtt/issues/11941, however it does also happen to newer blinds (got replaced). So not sure if the issue is only affecting relatively old blinds? The official IKEA hub though, didn't have this issue and all position change finished without issue. It started happening after I tried to use these blinds with z2m & conbee2.

pnewell, 2023-04-23:
I am also experiencing this, has anyone found a solution?

alex3305, 2023-04-23:
@gukj Apparently I missed your message but unfortunately that didn't work for me. The solution for me was to migrate to a Sonoff ZBDongle-P (also for @pnewell). 

But another weird and probably similar issue popped up for me today after recharging the batteries of my blinds. I couldn't control the blinds until I restarted Z2M. Bit annoying but 🤷

gukj, 2023-05-05:
I ended up replacing most if not all blinds at IKEA (after listing all the things I did to troubleshoot, they were very ok with just replacing them). Haven't had the issue where they'll just go up and down by themselves since. I do still have some issues with them after restarting the raspbi running them, but most often it just requires me to remove the battery for a minute and then replacing it. Worst case I'll put it into pairing mode and open up the phoscon gw and add them again. Haven't had to reconfigure them yet so somehow they seem to just remember their old config.