Koenkk/zigbee2mqtt #11925: [New device support]:

unpack8866, 2022-03-22:
### Link

https://www.orvibo.com/cn/product/productList.html?wjxsafari=1

### Database entry

MQTT publish: topic 'zigbee2mqtt/0x00124b00092cb604', payload '{"linkquality":72,"state_l1":"ON","state_undefined":"ON"}'

### Comments

Request to add device support; Orvibo socket。
[image]

### External converter

```shell
const device = {
    zigbeeModel: ['ORVIBO Socket'],
    model: 'OR-ZB-S010-3C',
    vendor: 'ORVIBO',
    description: 'Smart Socket',
    fromZigbee: [fz.ignore_basic_report, fz.on_off,],
    toZigbee: [tz.ptvo_switch_trigger, tz.on_off,],
    exposes: [e.switch().withEndpoint('l1'),
],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l1: 1,
        };
    },
module.exports = device;
```

unpack8866, 2022-03-23:
Yes, it works perfectly!
[image]

Koenkk, 2022-03-23:
added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-23 by Koenkk

unpack8866, 2022-04-05:
There is another model of this device, which will be delivered in 2015; Please add, thank you very much!
zigbeeModel: '93e29b89b2ee45bea5bdbb7679d75d24'

Koenkk, 2022-04-05:
Added

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)