Koenkk/zigbee2mqtt #11928: New Device wrong

beandj, 2022-03-22:
### What happened?

I bought a Zemismart ZTB25 (https://pt.aliexpress.com/item/1005003821797706.html), I already knew it wasn't supported but after joining it appears as TS0003 (_TZ3000_vjhcenzo).

[image]

This is the database.db entry:

{"id":18,"type":"Router","ieeeAddr":"0xa4c13837b30f7d3b","nwkAddr":62972,"manufId":4417,"manufName":"_TZ3000_vjhcenzo","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,242,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAABAAA"}},"genBasic":{"attributes":{"modelId":"TS0003","manufacturerName":"_TZ3000_vjhcenzo","powerSource":1,"zclVersion":3,"appVersion":81,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1}},"manuSpecificTuya_3":{"attributes":{"53264":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccc7358","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53264":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccc7358","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"53264":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ccc7358","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":81,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1647983037700,"defaultSendRequestWhen":"immediate"}

### What did you expect to happen?

That it was recognized correctly or as unsupport device

### How to reproduce it (minimal and precise)

Pairing

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20180507

### Adapter

Sonoff Zigbee Cc2531

Koenkk, 2022-03-23:
This is a whitelabel device, besides the picture/description, does everything work correctly?

beandj, 2022-03-23:
I think not. When I turn on/off one switch, all switches turning on/off

Info MQTT publish: topic 'zigbee2mqtt/0xa4c13837b30f7d3b', payload '{"linkquality":49,"state_center":"ON","state_left":"OFF","state_right":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0xa4c13837b30f7d3b', payload '{"linkquality":49,"state_center":"ON","state_left":"ON","state_right":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0xa4c13837b30f7d3b', payload '{"linkquality":10,"state_center":"ON","state_left":"ON","state_right":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0xa4c13837b30f7d3b', payload '{"linkquality":23,"state_center":"ON","state_left":"ON","state_right":"ON"}'

Koenkk, 2022-03-23:
Could you check if this issue has been fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0003'],
    model: 'ZM-L03E-Z',
    vendor: 'Zemismart',
    description: 'Smart light switch - 3 gang with neutral wire CUSTOM',
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right')],
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    whiteLabel: [{vendor: 'BSEED', model: 'TS0003', description: 'Zigbee switch'}],
    meta: {multiEndpoint: true, disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)
- does it work correctly now?

beandj, 2022-03-24:
In the initial tests everything seems to be working as it should

rodrigogbs, 2022-03-24:
same problem here, i will test the procedure and give feedback soon.

rodrigogbs, 2022-03-24:
I tried the procedure, but this error is appearing in the log:

``
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[17:17:46] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[17:17:47] INFO: MQTT available, fetching server detail ...
[17:17:47] INFO: Previous config file found, checking backup
[17:17:47] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[17:17:47] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[17:17:49] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.24.0 start
> node index.js
(node:361) UnhandledPromiseRejectionWarning: SyntaxError: Invalid or unexpected token
(Use `node --trace-warnings ...` to show where the warning was created)
(node:361) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:361) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
``

beandj, 2022-03-24:
@rodrigogbs, it looks like syntax error. I got this when I forgot one ";"

rodrigogbs, 2022-03-24:
Yes @beandj you are right, I was facing a problem during the .JS file creation.

I recreated the file and now it is working like a charm !! Thanks @Koenk.

Will you put this in a version update or this converter will be the definitive solution?

rodrigogbs, 2022-03-24:
We only have an issue with the LED stats, it is only ON when the state_left is ON.

How to manage this?

Because i want it to be ON when all the switches are OFF.

beandj, 2022-04-03:
@Koenkk,

I can't use this device to pairing endpoint devices

Info Zigbee: allowing new devices to join via interruptor07.
Error Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: 'Data request failed with error: 'No network route' (205)'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{},"error":"Data request failed with error: 'No network route' (205)","status":"error","transaction":"9w05y-1"}'

rodrigogbs, 2022-04-03:
> @Koenkk,
>
> I can't use this device to pairing endpoint devices
>
> Info Zigbee: allowing new devices to join via interruptor07. Error Request 'zigbee2mqtt/bridge/request/permit_join' failed with error: [...]

@beandj are you trying to pair a device far from the coordinator or are you trying just to use mesh signal to an enddevice (already connected to the Zigbee network) from TB25Z as a router?

beandj, 2022-04-03:
@rodrigogbs, I don't know how to do this. I always exclude from network and try pairing closest router.

I got this error by click here
[image]

rodrigogbs, 2022-04-03:
> @rodrigogbs, I don't know how to do this. I always exclude from network and try pairing closest router.
> 
> I got this error by click here [image]

Ok, you are trying to pair a device from this router.

 If you try to do this by another router, does the same error happen?

beandj, 2022-04-04:
> > @rodrigogbs, I don't know how to do this. I always exclude from network and try pairing closest router.
> > I got this error by click here [image]
> 
> Ok, you are trying to pair a device from this router.
> 
> If you try to do this by another router, does the same error happen?

Sometimes I got different errors, but with 'cortina01' I can pair.

rodrigogbs, 2022-04-05:
> Sometimes I got different errors, but with 'cortina01' I can pair.

@beandj I suggest you to always try to pair new devices with (ALL) (Todos - in portuguese) option enabled and nearest as possible to Coordinator (< 1m).

closed 2022-05-14 by github-actions[bot]