Koenkk/zigbee2mqtt #11931: hold_modes in Thermostat TV02

SargonofAssyria, 2022-03-23:
### What happened?

See here https://github.com/home-assistant/core/issues/44255 and here https://github.com/Koenkk/zigbee2mqtt/discussions/11900
Please change code so that "hold*" topic are supported as normal mode topics by Domoticz.
Or that hold* topics are valid and the change could be made in Domoticz.

### Zigbee2MQTT version

1.24.0-dev

### Adapter firmware version

20220103

### Adapter

Texas Instruments LAUNCHXL-CC1352P-2

Koenkk, 2022-04-17:
I'm not sure what you mean, could you make a pull request?

SargonofAssyria, 2022-04-21:
What I mean is that there are some system_mode devices and preset devices for Thermostats. They are almost the same. I wonder if those 2 different devices will be joined to 1 type for both, or that it will remain as it is now? It has impact on Domoticz.

Koenkk, 2022-04-21:
It will remain as is, the reason for that is that HA requires it: https://www.home-assistant.io/integrations/climate.mqtt/#example. So some conversion at the Domoticz integration should be done here.

closed 2022-05-29 by github-actions[bot]