Koenkk/zigbee2mqtt #11934: Sonoff ZBMINI-L not working with groups

nauper, 2022-03-23:
### What happened?

I installed the ZBMINI-L and when using the physical switch or trying to toggle just the ZBMINI-L it works like expected. But when I add it to a group and try to toggle the group, it doesn't work. The other lights still toggle as normal.

### What did you expect to happen?

That the ZBMINI-L should work in a group

### How to reproduce it (minimal and precise)

Connect the ZBMINI-L
Add it to a group
Try to control the group

### Zigbee2MQTT version

 1.24.0

### Adapter firmware version

0x264a0700

### Adapter

Conbee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8332646/log.txt)

Koenkk, 2022-03-23:
is this and enddevice? (you can see that in the frontend) if yes, this device may not support group commands

nauper, 2022-03-24:
yes, it is an end device. Ah that suck, but good to know

Koenkk, 2022-03-24:
Added a note about this to the docs.

closed 2022-03-24 by Koenkk