Koenkk/zigbee2mqtt #11935: Aqara Presence Detector FP1 not updating exposes after firmware update

piggydamn, 2022-03-23:
### What happened?

Updated the sensor when new firmware was reported by zigbee2mqtt.
Going from original version to 20210121.
all the exposes dosent work, presence_event always shows right_leave.
I overcome this problem by resetting the sensor.
This doesn't had any effect on the missing exposes.

### What did you expect to happen?

To work better with the new firmware :-)

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dong cc2562p

### Debug log

    "device": {
      "id": "9cccb6f01f8ec61be630d7e9e9fc79db",
      "name": " Presence_DiningRoom",
      "name_by_user": null,
      "disabled": false,
      "disabled_by": null,
      "entities": [
        {
          "device_class": null,
          "disabled_by": null,
          "disabled": false,
          "entity_category": null,
          "entity_id": "binary_sensor.0x54ef441000477ac7_presence",
          "icon": null,
          "original_device_class": "presence",
          "original_icon": null,
          "state": {
            "entity_id": "binary_sensor.0x54ef441000477ac7_presence",
            "state": "on",
            "attributes": {
              "approach_distance": "medium",
              "linkquality": 207,
              "monitoring_mode": "undirected",
              "power_outage_count": 1,
              "presence": true,
              "presence_event": "right_leave",
              "temperature": 24,
              "update": {
                "state": "idle"
              },
              "update_available": false,
              "device_class": "presence",
              "friendly_name": " Presence_DiningRoom_presence"
            },
            "last_changed": "2022-03-23T03:28:48.112081+00:00",
            "last_updated": "2022-03-23T09:59:47.615060+00:00"
          },
          "unit_of_measurement": null
        },
        {
          "device_class": null,
          "disabled_by": null,
          "disabled": false,
          "entity_category": null,
[... 1302 lines cut ...]
              "availability": [
                {
                  "topic": "zigbee2mqtt/bridge/state"
                }
              ],
              "device": {
                "identifiers": [
                  "zigbee2mqtt_0x54ef441000477ac7"
                ],
                "manufacturer": "Xiaomi",
                "model": "Aqara Presence Detector FP1 (regions not supported for now) (RTCZCGQ11LM)",
                "name": " Presence_DiningRoom",
                "sw_version": ""
              },
              "device_class": "update",
              "enabled_by_default": true,
              "entity_category": "diagnostic",
              "json_attributes_topic": "zigbee2mqtt/ Presence_DiningRoom",
              "name": " Presence_DiningRoom_update_available",
              "payload_off": false,
              "payload_on": true,
              "state_topic": "zigbee2mqtt/ Presence_DiningRoom",
              "unique_id": "0x54ef441000477ac7_update_available_zigbee2mqtt",
              "value_template": "{{ value_json['update']['state'] == \"available\" }}",
              "platform": "mqtt"

Otnow, 2022-03-23:
This is to be expected, since current stable version of zigbee2mqtt does not support the correct operation of the new sensor firmware, and before updating it, it was necessary to follow the [recommendation](https://github.com/Koenkk/zigbee2mqtt/issues/11019#issuecomment-1065971861):

> _Important: before updating the sensor firmware to version 53, you must install a zigbee2mqtt version that supports this firmware version, otherwise presence_event will report incorrect data due to the changed logic of the firmware._

Accordingly, in order to solve the problem, you need to install the [developer version](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) of zigbee2mqtt or wait for the release of the stable version at the beginning of next month.

piggydamn, 2022-03-23:
> This is to be expected, since current stable version of zigbee2mqtt does not support the correct operation of the new sensor firmware, and before updating it, it was necessary to follow the [...]

understood，
looking forward to the stable new version of zigbee2mqtt. 
thx!

maxi1134, 2022-03-31:
Hi there!

I installed the edge version of Z2M, and I still get this bug where it always shows right_leave and doesn't update states

Otnow, 2022-03-31:
> Hi there!
> 
> I installed the edge version of Z2M, and I still get this bug where it always shows right_leave and doesn't update states

Try re-pairing sensor.

piggydamn, 2022-04-03:
I installed the nwe version of Z2M, and I repaired the sensor。it works well at the begining，but after several houers it doesn't update states again

Otnow, 2022-04-03:
> I installed the nwe version of Z2M, and I repaired the sensor。it works well at the begining，but after several houers it doesn't update states again

First, try to reset sensor by power supply (off and then on), and if it does not help, then [write a log file](https://www.zigbee2mqtt.io/guide/configuration/logging.html) with `log_level: debug` setting, which will record moment of problem.

closed 2022-05-11 by github-actions[bot]