Koenkk/zigbee2mqtt #11936: Envilar ZGR904-S

th0str, 2022-03-23:
### Link

https://zigbee.blakadder.com/Envilar_ZGR904-S.html

### Database entry

N/A

### Comments

Almost identical to iCasa ICZB-RM11S witch you support. Only difference the buttons has swtiched side.
https://www.zigbee2mqtt.io/devices/ICZB-RM11S.html

### External converter

```shell
const definition = {
    zigbeeModel: ['ZGR904-S'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZGR904-S', // Vendor model number, look on the device for a model number
    vendor: 'Envilar', // Vendor of the device (only used for documentation and startup logging)
    description: 'Envilar Touchlink Remote', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    meta: {battery: {dontDividePercentage: true}},
    fromZigbee: [fz.command_recall, fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.battery],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.action(['recall_*', 'on', 'off', 'brightness_stop', 'brightness_move_up', 'brightness_move_down'])],
};
```

Koenkk, 2022-03-23:
Could you extend the actions such that it contains all possible recall options? So instead of `recall_*`;`recall_1`, `recall_2`, etc.

th0str, 2022-03-24:
I have added some actions now. is this enough?
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZGR904-S'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZGR904-S', // Vendor model number, look on the device for a model number
    vendor: 'Envilar', // Vendor of the device (only used for documentation and startup logging)
    description: 'Envilar Touchlink Remote', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    meta: {battery: {dontDividePercentage: true}},
    fromZigbee: [fz.command_recall, fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.battery],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.action(['recall_*', 'on', 'off', 'brightness_stop', 'brightness_move_up', 'brightness_move_down'])],
};

module.exports = definition;`

th0str, 2022-03-24:
Sorry, I'm not sure how to assist here.
All I did was take a copy from icasa remote https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/icasa.js.

If I can be of any assistance, please inform me how.

Koenkk, 2022-03-26:
Exactly what I needed, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-26 by Koenkk