Koenkk/zigbee2mqtt #11937: YSR-MINI-01_wwcw not fully turning off

Imaginous, 2022-03-23:
### What happened?

**YSR-MINI-01_wwcw**

The device can't be fully turned off when it is on the coolest or warmest color temperature.
Also when it is at the coolest color temperature it warm is still lit and vice versa.

Furthermore an X-Y color pallette is shown while you can only control warm and cold white colors.

### What did you expect to happen?

I expected that the device would turn on/off completely and that the on the warmest color temperature cold white would be off and vice versa.

### How to reproduce it (minimal and precise)

Select warmest or coolest color and turn device off.

The LEDs will still be lit but at a lower level. When the color temperature is neutral the device will turn off completely.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26720700

### Adapter

ConBeeII

Imaginous, 2022-03-23:
I did some further testing tonight.
I found out it only happens if the transition value is 1 or higher.

Also it seems setting it to 0 won't fix it.
You have to make the transition field empty to fix it again.

robertklep, 2022-04-14:
@Imaginous I have a somewhat similar issue with the dimmer-only device of the same brand, perhaps this works for you as well: https://github.com/Koenkk/zigbee2mqtt/issues/10741#issuecomment-1098724755

closed 2022-05-23 by github-actions[bot]