Koenkk/zigbee2mqtt #11942: [New device support]:

dariox64, 2022-03-23:
### Link

https://zigbee.blakadder.com/SWE_XZ-CGV3.html

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0x0c4314fffe6e0996","nwkAddr":17562,"manufId":4098,"manufName":"_TYZB01_vwqnz1sn","powerSource":"Battery","modelId":"TS0202","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,1280],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TYZB01_vwqnz1sn","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b0024bc1919","zoneState":1,"currentZoneSensitivityLevel":2}},"msIlluminanceMeasurement":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":262,"inClusterList":[1024],"outClusterList":[],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648063716462,"defaultSendRequestWhen":"immediate"}

### Comments

Currently the sensor is recognized as TS0202. Motion detection seems to be working properly. Unfortunately, there is no indication of the light intensity.

dariox64, 2022-04-23:
up

dareksm32, 2022-06-29:
up

dariox64, 2022-06-29:
up

dariox64, 2022-08-01:
up

dariox64, 2022-09-05:
up

dariox64, 2022-10-07:
up

closed 2022-12-23 by github-actions[bot]