Koenkk/zigbee2mqtt #11949: Alecto SMART-MOTION10 mislabeled as TuYa TS0201

RemcoSchrijver, 2022-03-24:
### What happened?

Adding the [Alecto Smart-Motion10 motion sensor](https://alectohome.nl/collections/alecto_smart_home/products/alecto-smart-motion10-smart-zigbee-bewegingssensor-wit) it is mislabeled as a [Tuya  TS0201](https://www.zigbee2mqtt.io/devices/TS0201.html#tuya-ts0201)

### What did you expect to happen?

I expected that either this device was supported or shown as not supported.

The database entry it gets is the following: 
`
{"id":16,"type":"EndDevice","ieeeAddr":"0xxxxxxxxxxxx","nwkAddr":37711,"manufId":4098,"manufName":"TUYATEC-dgtxmihe","powerSource":"Battery","modelId":"RH3040","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"RH3040","manufacturerName":"TUYATEC-dgtxmihe","appVersion":67,"stackVersion":0}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":182}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0018e86e7d","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"20180512","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1648153108115,"defaultSendRequestWhen":"immediate"}`

### How to reproduce it (minimal and precise)

Add the Alecto smart motion 10 to your zigbee network and it becomes mislabeled.

### Zigbee2MQTT version

1.23.0

### Adapter firmware version

20210120

### Adapter

Slaesh's CC2652RB stick

Koenkk, 2022-03-26:
This is a whitelabel TuYa device, meaning we cannot distinguish them from other whitelabeled TuYa devices (since the modelID is reused). As you can see the device discovers itself as a TuYa device: `"manufacturerName":"TUYATEC-dgtxmihe"`

RemcoSchrijver, 2022-03-28:
Thank you for your quick reply!
I already expected something like that was up, so can I just follow [the support new device instructions?](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html) Or do I need to remove some known TuYa devices from my setup as to get this one to show up as not supported?

Koenkk, 2022-03-28:
why do you want that? It already works right?

RemcoSchrijver, 2022-03-28:
Ohh sorry that I did not mention it in the issue, not only the name/icon is mislabeled but it also exposes the entities it expects from the temperature sensor so: 
`battery: 85%
temperature: N/A°C
humidity: N/A%
voltage: 2900mV
linkquality: 132lqi`
And not prescense/occupancy like I had expected.

Koenkk, 2022-03-29:
Could you make a screenshot of the z2m frontend -> click on device -> about and exposes pages?

Koenkk, 2022-03-30:
Could you check if it is recognized correctly now?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

RemcoSchrijver, 2022-03-31:
Yes that did the trick, it now registers as a [RH3040 PIR,](https://www.zigbee2mqtt.io/devices/RH3040.html#tuya-rh3040) thank you kindly for your help!

Koenkk, 2022-04-01:
perfect!

closed 2022-04-01 by Koenkk