Koenkk/zigbee2mqtt #11956: Niko 552-72101 + 552-72102 + 552-72301 + 552-72201

bjorndeboodt, 2022-03-26:
### Link

https://www.niko.eu/nl-be/producten/niko-home-control/producten-voor-traditionele-bekabeling/geconnecteerde-enkelvoudige-schakelaar-10-a-zigbee-productmodel-niko-6f86be91-a2f1-5421-a5ce-234092bfe9c3

### Database entry

{"id":20,"type":"Router","ieeeAddr":"0xcc86ecfffe2ac2a8","nwkAddr":21220,"manufId":4703,"manufName":"NIKO NV","powerSource":"Mains (single phase)","modelId":"Single connectable switch,10A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2821,64512,64513],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20210224","swBuildId":"313-687-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648230380156,"useImplicitCheckin":true}

### Comments

I tried to follow the supporting new device page but got stuck at point "2. Adding Your Device".
I try to connect 4 new devices of Niko: 552-72101 + 552-72102 + 552-72301 + 552-72201. I mannaged to pair these but when the programming part starts I end up nowere. I added the 4 devices to the existing converter-file "niko.js" but I am not sure I did it correctly. Also I have no clue of the things to write in the code. You can find my attempt in the "External converter" part.

Can someone help me so these devices can be added to the zigbee2mqtt supported devices?

Below I mentionned the links to the devices + the entry of each device from the database.

**Links + entries:**
1. 552-72101:  https://www.niko.eu/nl-be/producten/niko-home-control/producten-voor-traditionele-bekabeling/geconnecteerde-enkelvoudige-schakelaar-10-a-zigbee-productmodel-niko-6f86be91-a2f1-5421-a5ce-234092bfe9c3
Entry: {"id":20,"type":"Router","ieeeAddr":"0xcc86ecfffe2ac2a8","nwkAddr":21220,"manufId":4703,"manufName":"NIKO NV","powerSource":"Mains (single phase)","modelId":"Single connectable switch,10A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2821,64512,64513],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20210224","swBuildId":"313-687-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648230380156,"useImplicitCheckin":true}

2. 552-72102: https://www.niko.eu/nl-be/producten/niko-home-control/producten-voor-traditionele-bekabeling/geconnecteerde-enkelvoudige-schakelaar-10-a-zigbee-productmodel-niko-6f86be91-a2f1-5421-a5ce-234092bfe9c3
Entry: {"id":21,"type":"Router","ieeeAddr":"0xcc86ecfffe2ac022","nwkAddr":56467,"manufId":4703,"manufName":"NIKO NV","powerSource":"Mains (single phase)","modelId":"Double connectable switch,10A","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2821,64512,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Double connectable switch,10A","manufacturerName":"NIKO NV","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20210224","swBuildId":"313-688-00"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[0,3,4,5,6,2821,64512,64513],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20210224","swBuildId":"313-688-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648236197252,"useImplicitCheckin":true}

3. 552-72301: https://www.niko.eu/nl-be/producten/niko-home-control/producten-voor-traditionele-bekabeling/geconnecteerde-motorbediening-3-a-zigbee-productmodel-niko-7a97247a-5e33-5eaf-b21e-893ba9f15378
Entry: {"id":22,"type":"Router","ieeeAddr":"0xcc86ecfffe25d228","nwkAddr":14321,"manufId":4703,"manufName":"NIKO NV","powerSource":"Mains (single phase)","modelId":"Connectable motor control,3A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,4,5,258,2821,64512,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Connectable motor control,3A","manufacturerName":"NIKO NV","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":6,"hwVersion":0,"dateCode":"20200930","swBuildId":"313-690-00"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":0,"dateCode":"20200930","swBuildId":"313-690-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648236904583,"useImplicitCheckin":true}

4.  552-72201: https://www.niko.eu/nl-be/producten/niko-home-control/producten-voor-traditionele-bekabeling/geconnecteerde-dimmer-3-200-w-2-draads-zigbee-productmodel-niko-e61c5693-2d0c-5be5-8544-461983048049
Entry: {"id":23,"type":"Router","ieeeAddr":"0xcc86ecfffe25d5b5","nwkAddr":15836,"manufId":4703,"manufName":"NIKO NV","powerSource":"Mains (single phase)","modelId":"Connectable dimmer,3-200W,2-wire","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":267,"inClusterList":[0,3,4,5,6,8,2821,64512,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Connectable dimmer,3-200W,2-wire","manufacturerName":"NIKO NV","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20201010","swBuildId":"313-689-00"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20201010","swBuildId":"313-689-00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648237398277,"useImplicitCheckin":true}

### External converter

```shell
const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const reporting = require('../lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['Connected socket outlet'],
        model: '170-33505',
        vendor: 'Niko',
        description: 'Connected socket outlet',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
        toZigbee: [tz.on_off, tz.electrical_measurement_power, tz.currentsummdelivered],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.onOff(endpoint);

            // NOTE: we read them individually, acFrequency* is not supported
            //       so we cannot use readEletricalMeasurementMultiplierDivisors
            await endpoint.read('haElectricalMeasurement', ['acPowerMultiplier', 'acPowerDivisor']);
            await reporting.activePower(endpoint, {min: 5, max: 3600, change: 1000});
            await endpoint.read('haElectricalMeasurement', ['acCurrentDivisor', 'acPowerMultiplier', 'acPowerDivisor']);
            await reporting.rmsCurrent(endpoint, {min: 5, max: 3600, change: 100});
            await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor', 'acCurrentMultiplier']);
            await reporting.rmsVoltage(endpoint, {min: 5, max: 3600, change: 100});

            await reporting.readMeteringMultiplierDivisor(endpoint);
            await reporting.currentSummDelivered(endpoint, {min: 60, change: 1});
        },
        exposes: [
            e.switch(),
            e.power().withAccess(ea.STATE_GET), e.current(), e.voltage(),
            e.energy().withAccess(ea.STATE_GET),
        ],
[... 78 lines cut ...]
        description: 'Connectable motor control',
        fromZigbee: [],
[... cut ...]
        },
        exposes: [],
    },

// Niko 552-72201 - Connectable dimmer, 3-200W, 2-wire
// The programm-lines left blank/open is because I do not know what to write

    {
        zigbeeModel: ['Connectable dimmer, 3-200W, 2-wire'],
        model: '552-7221',
        vendor: 'Niko',
        description: 'Connectable motor control',
        fromZigbee: [],
        toZigbee: [], 
        configure: async (device, coordinatorEndpoint, logger) => {
        },
        exposes: [],
    },

];
```

sjorge, 2022-03-26:
For the connected switch you can try the code I posted in https://github.com/Koenkk/zigbee2mqtt/issues/11762, still waiting on one to finish that integration so for now It's just a guess based on the available info.

sjorge, 2022-03-29:
552-72101 + 552-72102  should now be supported in dev branch, I don’t own the other devices so I can’t add support for them.

closed 2022-05-06 by github-actions[bot]

bjorndeboodt, 2022-10-11:
Hello Jorge,

Apologies for the late response but this is a very busy period for me.
Thank you very much for your work regarding the issue of the Niko devices.

If you consider this an option, I could send you the other 2 devices
(552-72201 + 552-72301) by post so you can test them?
Feel free to let me know.

Best Regards,
Bjorn