Koenkk/zigbee2mqtt #11964: CC2652 Router get disconnected

celeroll, 2022-03-27:
### What happened?

I have a network which consists of coordinator = Slaesh's CC2652RB stick with firmware zStack3x0 20211217 and I'm on 1.24.0.
I have issues with my router CC2652  , which get disconnected from time to time and no end device is working.
CC2652 is on 20210128 version.
I tried the router dev version , but I all my end devices which were previously connected through this router were not reporting any more and I was not able to pair them again. 
So I returned back to 20210128.
I have created a CC2531 sniffer, so this I could also provide.
My "extended_pan_id": "dddddddddddddddd",
Below is a log attached.

### What did you expect to happen?

Since I removed all my old routers based on cc2530/cc2531 and upgraded to CC2652 I would expect to not have network disconnects at all.

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20210128

### Adapter

Texas Instruments LAUNCHXL-CC26X2R1

### Debug log

This are only the key logs. CC2652, if it is reporting it is reporting every 10 minutes. Have a look also at the link quality. It drops from 65 to 21 and now slowly goes back to ~60.
```
info  2022-03-26 21:20:34: MQTT publish: topic 'zigbee2mqtt/CC2652 Router', payload '{"linkquality":65}'
info  2022-03-26 23:10:32: MQTT publish: topic 'zigbee2mqtt/CC2652 Router', payload '{"linkquality":43}'
debug 2022-03-27 01:03:26: Active device 'CC2652 Router' was last seen '112.89' minutes ago.
debug 2022-03-27 01:13:27: Device 'CC2652 Router' reconnected
info  2022-03-27 01:13:27: MQTT publish: topic 'zigbee2mqtt/CC2652 Router/availability', payload 'online'
info  2022-03-27 01:13:27: MQTT publish: topic 'zigbee2mqtt/CC2652 Router', payload '{"linkquality":21}'
info  2022-03-27 07:40:35: MQTT publish: topic 'zigbee2mqtt/CC2652 Router', payload '{"linkquality":36}'
```

celeroll, 2022-04-08:
Any suggestion on this problem?

Chippi, 2022-05-13:
Your linkquality seems rather low. Mine are above 200. Do you live in a crowded area? Is there congestion on the 2.4Ghz band? 
I wrote this topic, you can maybe try that out. https://github.com/Koenkk/zigbee2mqtt/discussions/12467

closed 2022-05-21 by github-actions[bot]