Koenkk/zigbee2mqtt #11968: [Bug]: network map has no lines (at all)

TzachiGuetta, 2022-03-28:
### What happened?

The network map of my zigbee devices has no lines at all
I tried to:
1. switch to dev branch 
2. upgrade my adapter to latest version 
but the issue is still occurred 

In logs I have:
```
Error Failed to execute LQI for 'Coordinator'
Info Network scan finished
```

[image]

### What did you expect to happen?

to have lines between the cordinator and the endDevices 

### How to reproduce it (minimal and precise)

1. go to http://192.168.68.85:8080/#/map
2. click on Load map

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus (CC2652P)

### Debug log

Failed to execute LQI for 'Coordinator'
Info Network scan finished

Shred99, 2022-04-07:
I'm now running a second, much smaller Zigbee network in parallel with my main one and I'm having the same problem:

- ZigBee2MQTT version is 1.24.0
- Coordinator is a CC2531 (I know...) with firmware 20211115

The map worked perfectly while the coordinator was running firmware 20190608.  I swapped out the CC2531 for a different one with the 20211115 firmware and the problem with the map appeared.

It may be relevant that the network has no routers.

TzachiGuetta, 2022-04-11:
update:
I update my Adapter firmware version to: 20220219
and I updated my Zigbee2MQTT version to: 1.25.0
but the issue is still occurred

Shred99, 2022-04-11:
I think the common link between your network and mine is that we both have small networks with no routers.  It could be argued that the map functionality is pretty redundant in this sort of network anyway.

You could try adding a router, or downgrading your coordinator firmware to something ancient (mine worked perfectly until I upgraded the router firmware).  The problem with doing this is that you may need to re-pair the devices (I vaguely recall that downgrading below a certain version broken things?) and that if you have any sensors that need new firmware, they will break.

TzachiGuetta, 2022-04-15:
tried to:

1. add router (a usb stick of SONOFF Zigbee 3.0 USB Dongle Plus) 
2. using the shipped version of my Adapter 

but the issue is still occurred
@Shred99

Koenkk, 2022-04-17:
Could you check if this firmware fixes the `Error Failed to execute LQI for 'Coordinator'`? 
Note that even after this fix no lines might appear, see https://www.zigbee2mqtt.io/guide/faq/#why-are-some-links-missing-from-my-networkmap

[znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/8502097/znp_CC1352P_2_LAUNCHXL_tirtos_ccs.hex.zip) (only flash this fw on a SONOFF CC2652P coordinator)

TzachiGuetta, 2022-04-17:
@Koenkk, thank you for responding!
just tried what you suggested, but the issue is still occurred

```

- Debug Received MQTT message on 'zigbee2mqtt/bridge/request/networkmap' with data '{"routes":false,"transaction":"1ajh0-1","type":"raw"}'
- Info Starting network scan (includeRoutes 'false')
- Error Failed to execute LQI for 'Coordinator'

```

Koenkk, 2022-04-18:
@TzachiGuetta I cannot reproduce it, could you provide the herdmans debug logging starting from the point where you startm z2m until this error?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

TzachiGuetta, 2022-04-26:
@Koenkk, sorry for the late respond, missed your comment. 
here is it:
`Zigbee2MQTT:debug 2022-04-26 12:13:42: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-04-26 12:13:42: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-04-26.12-13-40' filename: log.txt
Zigbee2MQTT:debug 2022-04-26 12:13:42: Removing old log directory '/opt/zigbee2mqtt/data/log/2022-04-17.23-07-48'
Zigbee2MQTT:info  2022-04-26 12:13:42: Starting Zigbee2MQTT version 1.25.0 (commit #6f1460e4)
Zigbee2MQTT:info  2022-04-26 12:13:42: Starting zigbee-herdsman (0.14.20)
Zigbee2MQTT:debug 2022-04-26 12:13:42: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":true},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"auto","path":"/dev/ttyUSB0"}}'
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"auto"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":true,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +13ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625} +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +23ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,1,89,6,61] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,1,89,6,61] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 1 - [89,6] - 61 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - ping - {"capabilities":1625} +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +6ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,2,2,1,2,7,1,59,137,52,1,0,233] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,2,2,1,2,7,1,59,137,52,1,0,233] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 2 - [2,1,2,7,1,59,137,52,1,0] - 233 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219} +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 16 +0ms
  zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack3x0' ({"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20220219}) +0ms
  zigbee-herdsman:adapter:zStack:startup beginning znp startup +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - getExtAddr - {} +12ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,4,37] +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,8,97,4,176,191,212,28,0,75,18,0,243] +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,8,97,4,176,191,212,28,0,75,18,0,243] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 8 - 3 - 1 - 4 - [176,191,212,28,0,75,18,0] - 243 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - getExtAddr - {"extaddress":"0x00124b001cd4bfb0"} +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvLength - {"id":130} +10ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,33,19,130,0,178] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,2,97,19,24,0,104] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,2,97,19,24,0,104] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 2 - 3 - 1 - 19 - [24,0] - 104 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvLength - {"length":24} +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvReadExt - {"id":130,"offset":0} +9ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,33,28,130,0,0,0,187] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,97,28,0,24,0,1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13,229,1,0,71,176,4,0,107] +9ms
[... cut ...] tworkAddress":33143,"type":"EndDevice"},{"definition":{"description":"Aqara human body movement and illuminance sensor","model":"RTCGQ11LM","supports":"battery, occupancy, temperature, voltage, illuminance_lux, illuminance, linkquality","vendor":"Xiaomi"},"friendlyName":"Washroom motion sensor","ieeeAddr":"0x00158d000465f6db","lastSeen":1650963741158,"manufacturerName":"LUMI","modelID":"lumi.sensor_motion.aq2","networkAddress":38042,"type":"EndDevice"}]}},"status":"ok","transaction":"27dlt-1"}'`

TzachiGuetta, 2022-04-26:
@Koenkk new logs on dev branch:
[DEBUG=zigbee-herdsman* npm start.log](https://github.com/Koenkk/zigbee2mqtt/files/8566549/DEBUG.zigbee-herdsman.npm.start.log)

TzachiGuetta, 2022-04-26:
@Koenkk new logs on dev branch
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/8566556/z2m.log)
:

Koenkk, 2022-04-27:
@TzachiGuetta interesting, could you stop z2m and look what the `nwkAddr` of your coordinator is in the data/database.db? It should be `0`. If it is not `0` change it manually, start z2m and see if the issue still appears.

Example:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e1af91","nwkAddr":0,"manufId":0,"epList":........
```

TzachiGuetta, 2022-04-27:
@Koenkk FIXED!
thank you!
by the way, the value was: "nwkAddr":38978

closed 2022-04-28 by TzachiGuetta

fortellerq, 2022-07-26:
I am not sure what lead to this situation, but I can confirm that I changed cc2531 coordinator to SONOFF Zigbee 3.0 USB Dongle Plus (CC2652P) with the same firmware as OP. Then I started to receive this "Failed to execute LQI for 'Coordinator'" message during map update, but in my case I have routers. That lead to map showing this:
[image]
As you can see, LQI and lines are visible for routers, but not for end devices. That has been fixed by changing nwkAddr of coordinator to 0 as Koenkk suggested. In my case I had 21942 as nwkAddr of Coordinator. After chaning, map looks like this:
[image]
(had an issue with one sensor in the meantime, so "Czujnik temperatury zewnątrz" wasn't connected, even though it should on previous screenshot)

ampersandru, 2022-09-17:
i recently updated my ZBDongle-P to 20211217 and have the latest zigbee2mqtt and having the same issues. No issues with devices connecting to routers but nothing wants to connect to the coordinator which is an issue with a lot of aqara devices which don't like to reconnect to routers unless you initially pair it with the router.

How do I edit database.db to change this value?