Koenkk/zigbee2mqtt #11972: UnhandledPromiseRejectionWarning: TypeError: def.exposes is not iterable

peca89, 2022-03-28:
### What happened?

While running the 1.24 release on Windows, I get:
````
> zigbee2mqtt@1.24.0 start c:\zigbee2mqtt
> node index.js

(node:7028) UnhandledPromiseRejectionWarning: TypeError: def.exposes is not iterable
    at HomeAssistant.populateMapping (c:\zigbee2mqtt\lib\extension\homeassistant.js:590:38)
    at new HomeAssistant (c:\zigbee2mqtt\lib\extension\homeassistant.js:64:14)
    at new Controller (c:\zigbee2mqtt\lib\controller.js:83:34)
    at start (c:\zigbee2mqtt\index.js:100:18)
(node:7028) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch
block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:7028) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the
 Node.js process with a non-zero exit code.
````
1.21.2 is the last version that functions fine for me. Versions in between refuse to compile so I'm desperate for an update for new devices. 

### Zigbee2MQTT version

1.24

### Adapter firmware version

?

### Adapter

?

Koenkk, 2022-03-28:
Did you execute `npm run build` in the zigbeee2mqtt directory after updating?

peca89, 2022-03-29:
No, I didn't. That is not mentioned on https://www.zigbee2mqtt.io/guide/installation/04_windows.html
I only ran `npm ci` after updating.
Running `npm run build` throws whole bunch of errors:

````
c:\zigbee2mqtt>npm run build

> zigbee2mqtt@1.24.0 build c:\zigbee2mqtt
> tsc && node index.js writehash

lib/controller.ts:40:79 - error TS2304: Cannot find name 'Extension'.

40     (enable: boolean, name: string) => Promise<void>, () => void, (extension: Extension) => Promise<void>];
                                                                                 ~~~~~~~~~

lib/controller.ts:49:25 - error TS2304: Cannot find name 'Extension'.

49     private extensions: Extension[];
                           ~~~~~~~~~

<<<OUTPUT CUT>>>

Found 407 errors.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.24.0 build: `tsc && node index.js writehash`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.24.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Predrag\AppData\Roaming\npm-cache\_logs\2022-03-29T06_14_05_326Z-debug.log

c:\zigbee2mqtt>
````

In the meantime I have "patched" my working 1.21.2 by writing an external converter for my new device, but that's obviously not long term sustainable...

Koenkk, 2022-03-29:
What is the output of:
```bash
node --version
git status
```

peca89, 2022-03-30:
````
c:\zigbee2mqtt>node --version
v14.16.1
````
I'm not using git to update. I'm just overwriting with everything from release zip except \data folder. Therefore:
````
c:\zigbee2mqtt>git status
fatal: not a git repository (or any of the parent directories): .git

c:\zigbee2mqtt>
````
Might be worth noting that I'm using Windows 7.. :-/

Koenkk, 2022-03-30:
What are the contents of `lib/types/types.d.ts`?

peca89, 2022-03-30:
That file has the same contents as the one present in the release zip...

Anyway, I think I have repaired it. I have deleted everything from z2m folder leaving only data folder and its contents. Than I have copied fresh 1.24 contents, run `npm ci` and it compiles and runs now. I'm now really curious what has been causing the error in the original folder. Nothing was missing for sure as I just used to add new files and overwriting existing.

I still have the contents of the old folder backed up if you wish to troubleshoot. Otherwise, we may close this.

Koenkk, 2022-04-01:
@peca89 try doing a diff on the old vs new folder (with e.g. kdiff3)

closed 2022-05-09 by github-actions[bot]