Koenkk/zigbee2mqtt #11980: thermostat Moes BHT-002-GCLZB shows factor 10 too low temperature

regissg, 2022-03-29:
### What happened?

All is working fine except for the local_temperature that is 10 time lower than expected (1.8 instead of 18.5°).

### Zigbee2MQTT version

1.23.0-dev commit: 6394d0f4

### Adapter firmware version

20220103

### Adapter

Sonoff zStack3x0

### Debug log

`debug 2022-03-29 17:18:17: Received Zigbee message from 'sda', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,18],"type":"Buffer"},"datatype":2,"dp":24}],"seq":15109}' from endpoint 1 with groupID 0
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/state', payload 'ON'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/system_mode', payload 'heat'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/preset', payload 'hold'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/current_heating_setpoint', payload '20'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/child_lock', payload 'UNLOCK'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/sensor', payload 'IN'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/running_state', payload 'heat'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/heat', payload 'ON'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/kinds_of_period', payload 'OFF'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/local_temperature', payload '1.8'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/preset_mode', payload 'hold'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/local_temperature_calibration', payload '-1'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/max_temperature', payload '40'
info  2022-03-29 17:18:18: MQTT publish: topic 'zigbee2mqtt/sda/linkquality', payload '43'`

Koenkk, 2022-03-29:
Please check with z2m 1.24.0

regissg, 2022-03-30:
I have updated to 1.24.0 but I still have the same problem

Koenkk, 2022-03-30:
Could you provide the data/database.db entry of your device?

regissg, 2022-03-30:
The device modelID is a TS0601 but the manufacturerName is _TZE200_ye5jkfsb. I have edited directly the .../devices/moes.js file to add that device.

Here are the database.db as well as the moes.js
[moes.zip](https://github.com/Koenkk/zigbee2mqtt/files/8382810/moes.zip)

Koenkk, 2022-04-01:
Added it now, could you check if the reported `local_temperature` value is correct now?

regissg, 2022-04-01:
Your proposed fix does not work as you keep a  temperature = temperature / 10 for _TZE200_ye5jkfsb.

I do not understand why there was a /10 in the original fromZigbee.js file: 
```
            case tuya.dataPoints.moesLocalTemp:
                temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
                return {local_temperature: parseFloat((temperature / 10).toFixed(1))};

```
If I remove the /10 it does work well
```
            case tuya.dataPoints.moesLocalTemp:
                temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
                return {local_temperature: parseFloat(temperature.toFixed(1))}; 

```

adamaru, 2022-04-03:
Probably [this fix](https://github.com/Koenkk/zigbee-herdsman-converters/commit/946e42e74b826dbd1ae4be7bc340dd0d292a126c) broke my `_TZE200_aoclfnxz` . Now it's showing 10 times more than it's supposed to. `19.5°C` before fix now shows as `195°C`.

Koenkk, 2022-04-03:
Whoops; made a typo, fixed it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

adamaru, 2022-04-04:
It's working again, thanks.

chrisazzopardi, 2022-04-28:
any update on this?

closed 2022-06-06 by github-actions[bot]

VoltickVL, 2023-01-14:
I still also get have problem with temperature.... I am using HA Edge Addon v1.29.0-dev commit: [3351bc2]

{
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 190,
    "deadzone_temperature": 1,
    "heat": "OFF",
    "linkquality": 116,
    "local_temperature": 275,
    "local_temperature_calibration": -2,
    "max_temperature": 30,
    "max_temperature_limit": 350,
    "min_temperature_limit": 1,
    "preset": "hold",
    "preset_mode": "hold",
    }

Manufacturer: [Moes](https://www.zigbee2mqtt.io/supported-devices/#v=Moes)
Model: [BHT-002-GCLZB](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html#moes-bht-002-gclzb)
Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_5toc8efa