Koenkk/zigbee2mqtt #11981: Ikea devices no updates found

Thuurke, 2022-03-29:
### What happened?

The ikea site http://fw.ota.homesmart.ikea.net/feed/version_info.json shows there is an update for the shortcutbuttons.
3 shortcutbuttons of mine are on version 2.3.015. After clicking the check update button under OTA the update is not found. I have a spare shortcutbutton and added this one to zigbee2mqtt. This button is updated to 2.3.080.
I also have bulbs, remote control, switches, motion sensor, control outlets and none of them get any updates.

### What did you expect to happen?

The troubleshooting tips on https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#automatic-checking-for-available-updates don't help either

device to be updated

### How to reproduce it (minimal and precise)

I don't know

### Zigbee2MQTT version

1.24.0 commit: 7a2ddf2

### Adapter firmware version

0x26580700

### Adapter

ConBee2

yawor, 2022-03-30:
Check the firmware version on devices manually. Z2M not always refreshes information about the actual firmware version and date code.
In frontend, go to Devices, select your device and open Console tab. Select endpoint 1, genBasic cluser and swBuildId attribute and then click load. If this is a remote (shortcut button) then press it before loading the value as the device must be woken up.

Thuurke, 2022-03-30:
> Check the firmware version on devices manually. Z2M not always refreshes information about the actual firmware version and date code. In frontend, go to Devices, select your device and open Console [...]

I followed you're instructions. In the log I see the button press but when I press Read (instead of Load) I don't get an reaction. When I take the same actions from a light bulb or control outlet I get the firmware version.

yawor, 2022-03-30:
Here's the result from E1743 On/Off shortcut button:
[image]

Thuurke, 2022-03-30:
Oke after clicking on Read and several presses on the shortcut button (which shows on OTA page [2.3.015]) it shows Info Read result of 'genBasic': {"swBuildId":"2.3.080"}. 
Does this mean that the button is on the newest firmware (2.3.080) and the OTA page is incorrect? Can this be corrected?

Second thing is that when I go to one device and click read and the info is shown and I navigate to another device and dev console the info of genBasic is already shown, so it is not cleared.

yawor, 2022-03-30:
Yes. You have the latest FW, but Z2M didn't refresh that information. It may refresh it by itself after some time I think. You can also try remove the device and pair it again with your network (the results may vary, it seems that Z2M may still keep some information about removed devices in its cache?). I couldn't find a 100% working solution for Z2M to refresh the FW version and date code - it just does it whenever it likes to :).

Re-pairing the device after FW update is required for some devices to get full functionality. For example the Tradfri 5-button remote (E1524/E1810) has Scenes cluster available for binding after update to 2.3.080, which can be bound to LED bulbs for example to change color of the bulb, but it only shows up after re-pairing the remote after FW update.

closed 2022-05-08 by github-actions[bot]