Koenkk/zigbee2mqtt #11990: [New device support]: Aqara P1 Motion Sensor (MS-S02)

backlink-market, 2022-03-30:
### Link

https://www.aqara.com/en/product/motion-sensor-p1

### Database entry

{"id":24,"type":"EndDevice","ieeeAddr":"0x54ef44100043e1a1","nwkAddr":380,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.motion.ac02","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,64704],"outClusterList":[3,25,64704],"clusters":{"genBasic":{"attributes":{"appVersion":5,"modelId":"lumi.motion.ac02","manufacturerName":"LUMI","powerSource":3,"zclVersion":3,"hwVersion":1,"dateCode":"Dec  7 2021","swBuildId":"0.0.0_0005"}},"aqaraOpple":{"attributes":{"252":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":5,"hwVersion":1,"dateCode":"Dec  7 2021","swBuildId":"0.0.0_0005","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648630655174,"defaultSendRequestWhen":"immediate"}

### Comments

The new Aqara P1 Motion Sensor (MS-S02) is nearly the same as the Aqara RTCGQ11LM but with two Batteries and a possibilty to change the timeout and constant LUX updates.

Koenkk, 2022-03-30:
@z-x-f @otnow , managed to get this device already? 😄

Otnow, 2022-03-30:
🙂 Unfortunately, I do not have this sensor, but judging by description, functionally it is a symbiosis of _[RTCGQ12LM](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html)_ and _[RTCGQ13LM](https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html)_.

Accordingly, it can be assumed that this sensor uses the logic of its predecessors, i.e. the [external converter](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters) suggested below might work:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.motion.ac02'],
        model: 'RTCGQ14LM',
        vendor: 'Xiaomi',
        whiteLabel: [{vendor: 'Xiaomi', model: 'MS-S02'}],
        description: 'Aqara P1 human body movement and illuminance sensor',
        fromZigbee: [fz.RTCGQ12LM_occupancy_illuminance, fz.aqara_opple, fz.battery],
        toZigbee: [tz.aqara_detection_interval, tz.aqara_motion_sensitivity],
        exposes: [e.occupancy(), e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'),
            exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']),
            exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
                .withDescription('Time interval for detecting actions'), e.temperature(), e.battery()],
        meta: {battery: {voltageToPercentage: '3V_2850_3000_log'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage']);
            await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
            await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
        },
        ota: ota.zigbeeOTA,
};

module.exports = definition;
```

_**Important**: to check the operation of proposed converter, latest stable versions of zigbee2mqtt and firmware of coordinator (at least as [indicated here](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html#adapter-firmware)) must be installed._

backlink-market, 2022-03-30:
I am using the Z-Stack firmware with my ONOFF Zigbee 3.0 USB Dongle with this https://github.com/ioBroker/ioBroker.zigbee.
So how can I test your suggestion? Where to add this code?

asgothian, 2022-03-31:
The iobroker allows the use of external converters, but the feature is not well documented. As such, i need to verify the exact steps which are required and will provide them later today.

A.

backlink-market, 2022-03-31:
> 🙂 Unfortunately, I do not have this sensor, but judging by description, functionally it is a symbiosis of _[RTCGQ12LM](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html)_ and _[RTCGQ13LM](https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html)_.
> 
> Accordingly, it can be assumed that this sensor uses the logic of its predecessors, i.e. the [external converter](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters) suggested below might work:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>         zigbeeModel: ['lumi.motion.ac02'],
>         model: 'RTCGQ14LM',
>         vendor: 'Xiaomi',
>         whiteLabel: [{vendor: 'Xiaomi', model: 'MS-S02'}],
>         description: 'Aqara P1 human body movement and illuminance sensor',
>         fromZigbee: [fz.RTCGQ12LM_occupancy_illuminance, fz.aqara_opple, fz.battery],
>         toZigbee: [tz.aqara_detection_interval, tz.aqara_motion_sensitivity],
>         exposes: [e.occupancy(), e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'),
>             exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']),
>             exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
>                 .withDescription('Time interval for detecting actions'), e.temperature(), e.battery()],
>         meta: {battery: {voltageToPercentage: '3V_2850_3000_log'}},
>         configure: async (device, coordinatorEndpoint, logger) => {
>             const endpoint = device.getEndpoint(1);
>             await endpoint.read('genPowerCfg', ['batteryVoltage']);
>             await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
>             await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
>         },
>         ota: ota.zigbeeOTA,
> };
> 
> module.exports = definition;
> ```
> 
> _**Important**: to check the operation of proposed converter, latest stable versions of zigbee2mqtt and firmware of coordinator (at least as [indicated here](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html#adapter-firmware)) must be installed._

Unfortunatly this code is not working. There is a TypeError: TypeError: Cannot read property 'options' of undefined
Please see  [https://github.com/ioBroker/ioBroker.zigbee/issues/1402#issuecomment-1084971902](url)

Otnow, 2022-04-01:
> Unfortunatly this code is not working. There is a TypeError: TypeError: Cannot read property 'options' of undefined
> Please see [https://github.com/ioBroker/ioBroker.zigbee/issues/1402#issuecomment-1084971902](url)

I got acquainted with [your ioBroker PR](https://github.com/ioBroker/ioBroker.zigbee/issues/1402) and it seems that I understood a little how _ioBroker.zigbee_ module works.

According to [ioBroker changelog](https://github.com/ioBroker/ioBroker.zigbee/blob/master/README.md#changelog), _ioBroker.zigbee 1.6.16_ was released on _2022-02-16_ and, accordingly, I can assume that version of _[zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters)_ used in it corresponds to this date.

My initial version of external converter was designed for _[development version](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) of zigbee2mqtt_ (not stable version, as I mistakenly indicated), so it did not work for you, because at the moment, significant changes have been made to _zigbee-herdsman-converters_.

Try to test this variant adapted, it seems to me, to your version of _zigbee-herdsman-converters_:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: ['lumi.motion.ac02'],
        model: 'RTCGQ13LM',
        vendor: 'Xiaomi',
        whiteLabel: [{vendor: 'Xiaomi', model: 'MS-S02'}],
        description: 'Aqara P1 human body movement and illuminance sensor',
        fromZigbee: [fz.RTCGQ12LM_occupancy_illuminance, fz.aqara_opple, fz.battery],
        toZigbee: [tz.aqara_detection_interval, tz.RTCGQ13LM_motion_sensitivity],
        exposes: [e.occupancy(), e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'),
            exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']),
            exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
                .withDescription('Time interval for detecting actions'), e.temperature(), e.battery()],
        meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage']);
            await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
            await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
        },
        ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Changes:
- `tz.aqara_motion_sensitivity` replaced by `tz.RTCGQ13LM_motion_sensitivity`
- `3V_2850_3000_log` replaced by `3V_2850_3200`
- also for testing needs, model `RTCGQ14LM` is replaced by `RTCGQ13LM`, so that `motion_sensitivity` is initialized when pairing

_Accordingly, it will be necessary to re-pair the sensor._

backlink-market, 2022-04-01:
Great, this works well except the motion_sensitivity does not gets verified and so its not set correctly, I guess?
http://prntscr.com/D4pFQpLqfhB5

Otnow, 2022-04-01:
> Great, this works well except the motion_sensitivity and detection_interval. Both values are not respected by the sensor when I chage them. Its always the default value.

Because since sensor is battery-powered, it is in sleep mode most of the time to save energy and therefore does not respond to commands.

In order for sensor to accept new settings, immediately after changing them in web interface, _press button on it to wake it up_.

Otnow, 2022-04-01:
> Ok, did it and the error does not pop up in the log but the sensor still does not respect that value. Occupany is still 60s true before it falls back to false when no motion happens.

To better understand what's going on, we now need a _debug log_.

Enable debug messages (it seems to me that this option is visible in [this screenshot](https://github.com/ioBroker/ioBroker.zigbee/issues/1402#issuecomment-1084916391)) and provide a log in which _process of pairing sensor and an attempt to change its settings will be visible_.

asgothian, 2022-04-01:
This is correct. Please note: In order to see the zigbee-herdsman-debug messages, the adapter has to be run in debug mode as well.

A.

backlink-market, 2022-04-01:
the logs are quite big for pasting so I uploaded the log files:
Pairing log: https://file.io/EZ3AG2BdNS9e
changing detection_interval to 10s: https://file.io/aOixzZmlgaMc

note: There might be some unuseful logs stuff before and after but I thought its better to add more than missing something.

Otnow, 2022-04-01:
> Pairing log: https://file.io/EZ3AG2BdNS9e
> changing detection_interval to 10s: https://file.io/aOixzZmlgaMc

The file.io service reports:
> The file you requested has been deleted

There is no need to use a third party service as files can be attached here.

backlink-market, 2022-04-01:
Pairing log: [pairing_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8399570/pairing_log.txt)
changing detection_interval to 10s: [changed detection_interval to 10s.txt](https://github.com/Koenkk/zigbee2mqtt/files/8399571/changed.detection_interval.to.10s.txt)

Otnow, 2022-04-01:
> Occupany is still 60s true before it falls back to false when no motion happens.

> Pairing log: [pairing_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8399570/pairing_log.txt)

It's strange that your _occupancy_ is 60 seconds, although during pairing, the sensor reports that `detection_interval` = 30 seconds:

```javascript
2022-04-01 18:20:53.350 - debug: zigbee.1 (15728) Received Zigbee message from '0x54ef44100043f0b3', type 'readResponse', cluster 'aqaraOpple', data '{"258":30}' from endpoint 1 with groupID 0
...
2022-04-01 18:20:53.379 - debug: zigbee.1 (15728) Publish {"detection_interval":30} to "54ef44100043f0b3"
```
> changing detection_interval to 10s: [changed detection_interval to 10s.txt](https://github.com/Koenkk/zigbee2mqtt/files/8399571/changed.detection_interval.to.10s.txt)

I did not find any entry in log about pressing button on sensor after trying to change `detection_interval`.

Try to write a log with 2-3 attempts to change `detection_interval` and `motion_sensitivity`, as well as the start and end of motion detection.

backlink-market, 2022-04-02:
okay, here we go:
changed the interval to 13,14,15 seconds and pushed the button several times.
[changed detection_interval 13,14,15s.txt](https://github.com/Koenkk/zigbee2mqtt/files/8402143/changed.detection_interval.13.14.15s.txt)

changed the motion_sensitivity several times and pushed the button several times:
[changed motion sensitivity .txt](https://github.com/Koenkk/zigbee2mqtt/files/8402148/changed.motion.sensitivity.txt)

Otnow, 2022-04-02:
> changed the interval to 13,14,15 seconds and pushed the button several times. [changed detection_interval [...]

These logs show that sensor accepts changed `detection_interval` and `motion_sensitivity` settings.

However, I did not see in logs "end of motion detection" event (`"occupancy":false`), which I asked to record:

> as well as the start and end of motion detection.

Try to write a log that will contain following actions:

- press button on sensor once
- change `motion_sensitivity` to **high** and immediately press button on sensor once (_no need to press it several times_)
- change `detection_interval` to **2 seconds** and immediately press button on sensor once (_no need to press it several times_)
- wait until sensor stops detecting movement (`"occupancy":false`)
- create an active continuous movement in front of sensor for 15-20 seconds
- do not create any movement in front of sensor until sensor stops detecting movement (`"occupancy":false`)
- press button on sensor once

backlink-market, 2022-04-03:
Ok, this is a exact instructions 👍 
Here is the log file:
[Logfile.txt](https://github.com/Koenkk/zigbee2mqtt/files/8404401/Logfile.txt)

Otnow, 2022-04-03:
> Here is the log file:
> [Logfile.txt](https://github.com/Koenkk/zigbee2mqtt/files/8404401/Logfile.txt)

Very good. Judging by log, sensor accepts settings and follows them, but for some reason converter timeout does not correspond to `detection_interval` and is equal to default value (60 seconds), although there is a record of state change:

```javascript
2022-04-03 13:40:23.153 - debug: zigbee.1 (8604) convert result {"state":{"detection_interval":2}}
```

This is strange and I still can not understand reason for discrepancy.

I would like to take a look at values of states of this sensor.
Is it possible in _ioBroker_ to see current states of sensor like in zigbee2mqtt web-interface?:

[image]

I also noticed in log that you have a `lumi.motion.agl04` (**RTCGQ13LM**) sensor on network. The timeout algorithm in its converter is same as in what we are currently using for `lumi.motion.ac02` (**MS-S02**).
Hence the question - does `lumi.motion.agl04` work correctly for you, i.e. does timeout match value of `detection_interval`?

backlink-market, 2022-04-03:
Do you mean this: 
[image]

When I change the detection_interval in the RTCGQ13LM settings it shows the timeout error: 
Send command to 0x54ef441000398e05 failed with no error code (Timeout - 19045 - 1 - 241 - 64704 - 4 after 10000ms) 

When I try to wake up the sensor via button on the back it does not work. So the value is not set.

Otnow, 2022-04-03:
> Do you mean this:

This is a little different, it seems to me that this is an analogue of _Exposes_ in the zigbee2mqtt web-interface.

In zigbee2mqtt, device states can be found in `state.json` file in `data` subdirectory. Maybe in _ioBroker_ there is something similar?

> When I change the detection_interval in the RTCGQ13LM settings it shows the timeout error: Send command to 0x54ef441000398e05 failed with no error code (Timeout - 19045 - 1 - 241 - 64704 - 4 after [...]

Also strange. Unable to set `detection_interval` on both `lumi.motion.agl04` sensors?

@asgothian, can you tell us where in _ioBroker_ we can see device states and can there be a problem with writing or reading states (in our case it is `detection_interval`)?

asgothian, 2022-04-03:
The way it works in iobroker:

- each expose will generate one or multiple states, as visible in the screenshot @backlink-market posted 2 hrs ago.
- if a device reports any piece of information linked to an expose, this state will be updated.
- if the operator changes a state, the respective "set" command will be issued.
- ioBroker does not support reading individual states, but a trigger on the "device query" will attempt to read all states which offer a "read" function in the converter.

If a device offers a read function but no value is read, a message is left in the log. The same is true if a device does not actually acknowledge a "set" command.

A.

Otnow, 2022-04-03:
@asgothian, ok, thanks.
How can we find out the exact version of `zigbee-herdsman-converters` that is used in @backlink-market's _ioBroker_?

backlink-market, 2022-04-04:
I am using Zigbee Adapter v1.6.16. Maybe @asgothian can help

Otnow, 2022-04-07:
> So I think the version you are looking for is 14.0.422 , right?

Yes, that's right.

I installed _ioBroker (v5.3.4)_ + _ioBroker.zigbee (v1.6.16)_ on Windows and found cause of problem.

The following code is used to set `timeout` value in [fz.RTCGQ12LM_occupancy_illuminance](https://github.com/Koenkk/zigbee-herdsman-converters/blob/5b63e9377d4577ff557ebbbb3c47c42065c32d37/converters/fromZigbee.js#L5619-L5645):

```javascript
const timeout = meta && meta.state && meta.state.hasOwnProperty('detection_interval') ? meta.state.detection_interval : 60;
```

In _zigbee2mqtt_, this code works correctly, but in _ioBroker.zigbee_ it does not work correctly, because **`meta.state` object does not exist in it** and the condition check always returns false, so default value (60 sec) is set regardless of value `detection_interval`.

This issue affects motion sensors [RTCGQ12LM](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html), [RTCGQ13LM](https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html) and current _RTCGQ14LM_ and solving it **requires adding support for `meta.state` object to _ioBroker.zigbee_**.

As a result, judging by logs, converter works correctly, so I created a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4099) to add support for _RTCGQ14LM_ to _zigbee2mqtt_.

closed 2022-04-07 by Koenkk

ScottG489, 2022-04-18:
For anyone coming across this thread, to get this device recognized, the first config offered in https://github.com/Koenkk/zigbee2mqtt/issues/11990#issuecomment-1083365730 is what worked for me, not the amended one mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/11990#issuecomment-1085301841. The amended one actually caused the error mentioned (`Cannot read property 'options' of undefined`) it was purportedly supposed to fix.

I was able to get the sensitivity and timeout working as well by pressing the button around the time I was sending the update. However, I'm not exactly sure how to "submit" these updates via the UI since there are just text fields and no "submit" button, so I don't know the exact ordering or timing to get the setting to pick up. I'm also not sure if I should be using the "Exposes" or "Settings (specific)" tab in the UI since both seem to let you enter the value. After a little testing, pressing the button and then updating the value via Home Assistant works best for me (though doing this in the zigbee2mqtt interface probably has the same effect).

I have also seen the following error show up from time to time which I assume is a failed update:
```
Publish 'set' 'detection_interval' to '0x<redacted>' failed: 'Error: Write 0x<redacted>/1 aqaraOpple({"258":{"value":[5],"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
```

Also a question for the maintainers: I saw the PR mentioned above (https://github.com/Koenkk/zigbee-herdsman-converters/pull/4099) was merged. I assume once this makes it's way into the stable version and the HA add-on updates to use that then I can remove this external converter? Any estimate on about when that would be?

I'm new to this since I just migrated to zigbee2mqtt from ZHA yesterday :) Really love it so far! My Aqara vibration sensor has a lot more info coming into HA!

Otnow, 2022-04-18:
> The amended one actually caused the error mentioned (`Cannot read property 'options' of undefined`) it was purportedly supposed to fix.

You didn't read the discussion carefully.

The modified version of converter was not intended for the current version of _zigbee2mqtt_, but for a specific version of third-party software (_ioBroker_) built on its basis, so it is logical that it did not work for you.

> I assume once this makes it's way into the stable version, the HA add-on updates to use that, then I can remove this external converter? Any estimate on about when that would be?

Yes, is now available in the _[dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) zigbee2mqtt_, and will appear in the _master branch (stable)_ at the beginning of next month.

ScottG489, 2022-04-19:
You're right that I didn't read carefully :) Regardless, thought I'd call out which one worked for me for the next person coming to this thread to sift through easier.

Thanks again for helping with this. It's been working great!

digiblur, 2022-04-22:
> You didn't read the discussion carefully.
>
> The modified version of converter was not intended for the current version of _zigbee2mqtt_, but for a specific version of third-party software [...]

Thanks for your work on this sensor.  Working great!  Only thing I noticed is the interval slider in Z2M allows the scale to be set further than allowed so it throws an error.  Easy fix to just punch in the number you want.

spn91, 2022-04-23:
Should this device already work with the latest version of the Home Assistant Edge Addon? I added it today to Z2M, but it didn't report motion or illuminance. This was just a short test (I am evaluating switching from deconz), so I have no logs at the moment. Just wanted to ask if it should work with the Addon.

Otnow, 2022-04-23:
> Should this device already work with the latest version of the Home Assistant Edge Addon? I added it today to Z2M, but it didn't report motion or illuminance.

Yes, it should work.

Verify that the coordinator firmware version you are using meets the [following requirements](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html#adapter-firmware).

spn91, 2022-04-23:
That's exactly the firmware which is installed on my Conbee II

Otnow, 2022-04-23:
> That's exactly the firmware which is installed on my Conbee II

Try to re-pair the sensor and if it doesn’t help, then [turn on the debug mode](https://www.zigbee2mqtt.io/guide/configuration/logging.html) (`log_level: debug`) and write a log in which the _pairing process_ will be visible, then _pressing the button once_ and the _moment of movement in front of the sensor_.

spn91, 2022-04-24:
Very strange, I have  re-paired it multiple times yesterday. Now I have switched to the stable addon and added it via external converter and it is working now. I don't know what was wrong with it yesterday. Thanks for the help and the great software 👍

Otnow, 2022-05-07:
Judging by the screenshot of the sensor settings from the _Aqara Home_ app, there is also the `trigger indicator` option, the support of which has not yet been implemented, so I [made](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4222) an appropriate note in the description.

[image]

Otnow, 2022-05-10:
During the [discussion](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4222#issuecomment-1121276271), it was decided to [add](https://github.com/Koenkk/zigbee2mqtt.io/pull/1377) information about feature support to the documentation instead of a sensor description.

TheJulianJES, 2022-05-13:
Sorry for this just being an image for now -- I've read out the first 400 attributes on the Aqara Opple cluster (0xfcc0) on endpoint 1:
[image]

So far, only the attributes 258/0x0102 (detection interval) and 268/0x010c (motion sensitivity) are known.
I'll try to play around with writing these attributes and will report back.

TheJulianJES, 2022-05-13:
After some trial and error, it seems to be attribute 338/0x0152.

0: trigger indicator LED disabled
1: trigger indicator LED enabled

Otnow, 2022-05-13:
@TheJulianJES, thanks a lot 👍

@ScottG489, @digiblur, @spn91 and other owners of this sensor, please test the `trigger indicator` option using the following [external converter](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters):

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;
const herdsman = require('zigbee-herdsman');
const manufacturerOptions = {
    xiaomi: {manufacturerCode: herdsman.Zcl.ManufacturerCode.LUMI_UNITED_TECH, disableDefaultResponse: true},
};

const fzLocal = {
    RTCGQ14LM_trigger_indicator: {
        cluster: 'aqaraOpple',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data.hasOwnProperty('338')) {
                return {trigger_indicator: msg.data['338'] === 1};
            }
        },
    }
}

const tzLocal = {
    RTCGQ14LM_trigger_indicator: {
        key: ['trigger_indicator'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {true: 1, false: 0};
            await entity.write('aqaraOpple', {0x0152: {value: lookup[value], type: 0x20}}, manufacturerOptions.xiaomi);
            return {state: {trigger_indicator: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('aqaraOpple', [0x0152], manufacturerOptions.xiaomi);
        },
    }
}

const definition = {
    zigbeeModel: ['lumi.motion.ac02'],
    model: 'RTCGQ14LM',
    vendor: 'Xiaomi',
    whiteLabel: [{vendor: 'Xiaomi', model: 'MS-S02'}],
    description: 'Aqara P1 human body movement and illuminance sensor',
    fromZigbee: [fz.aqara_occupancy_illuminance, fz.aqara_opple, fz.battery, fzLocal.RTCGQ14LM_trigger_indicator],
    toZigbee: [tz.aqara_detection_interval, tz.aqara_motion_sensitivity, tzLocal.RTCGQ14LM_trigger_indicator],
    exposes: [e.occupancy(), e.illuminance_lux().withProperty('illuminance'),
        e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'),
        exposes.enum('motion_sensitivity', ea.ALL, ['low', 'medium', 'high']),
        exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
            .withDescription('Time interval for detecting actions'),
        exposes.binary('trigger_indicator', ea.ALL, true, false).withDescription('When this option is enabled then ' +
            'blue LED will blink once when motion is detected'), e.temperature(), e.battery()],
    meta: {battery: {voltageToPercentage: '3V_2850_3000_log'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genPowerCfg', ['batteryVoltage']);
        await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
        await endpoint.read('aqaraOpple', [0x010c], {manufacturerCode: 0x115f});
        await endpoint.read('aqaraOpple', [0x0152], {manufacturerCode: 0x115f});
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

_**Important**: to check the operation of proposed converter, zigbee2mqtt 1.25.1 and latest firmware of coordinator (at least as [indicated here](https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html#adapter-firmware)) must be installed._

In case of successful testing, please write a [debug log](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee2mqtt-debug-logging) with the following actions:

- sensor pairing
- press button on sensor once and wait 10 sec.
- enable `trigger indicator` option and immediately press button on sensor once
- press button on sensor once and wait 10 sec.
- disable `trigger indicator` option and immediately press button on sensor once
- press button on sensor once and wait 10 sec.

bzeiss, 2022-05-20:
Sorry for being a newbie. I wanted to try this external converter as I want to mount the P1 in my bedroom and the led would be annoying. I have tried to add this external converter to the zigbee2mqtt 1.25.1-1 config file (installed as Home Assistant addon) as well as through the zigbee2mqtt setting UI and have placed the aqara_p1.js file in the /config/zigbee2mqtt (=data directory). It seems that on pairing the already existing internal converter is used. I have altered the description in the .js file a little so that I could see the difference. Is there any way I can control which converter is used?

Otnow, 2022-05-20:
@bzeiss, as far as I understand, you tried to use an external converter according to the [instructions](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters). Right?

After changing the settings, you need to restart _zigbee2mqtt_ and, as you correctly noted, you can understand that it is the external converter that is working according to the _modified description_.

bzeiss, 2022-05-20:
@Otnow Thanks for answering. Yes, I somehow made it work. Not sure why it didn't work before, it was probably related to a missing restart as you suggested. The trigger setting seems to work for me. It's not blinking any more.

digiblur, 2022-05-20:
@bzeiss same kind of oddness I found at times https://youtu.be/qWVbsZCj6CY

bzeiss, 2022-05-20:
@Otnow Great! 

About the sensitivity, I'm not sure. It behaves really odd. I have tried multiple timeout settings and occupancy timeouts along with different sensitivity values. I think low is still worse than high. No matter what configuration, the maximum detection distance is like 2m for me (with all my four P1 motion sensors).

backlink-market, 2022-05-21:
I can confirm. Most P1 sensor does only have a detection distance of 2m or below.....don't know why

Otnow, 2022-05-21:
It seems that the matter is not in the sensitivity settings, but this option can be definitely excluded only by testing the sensor with the native Aqara hub.

Does anyone from the owners of the sensor have such an opportunity?

***

Also on the official website of the manufacturer there is such a comparison table:

[image]

As far as I understand, the old model **RTCGQ11LM** is compared with **RTCGQ14LM**.

This is the detection area for the **RTCGQ11LM** from the [user manual](https://cdn.cnbj2.fds.api.mi-img.com/cdn/aqara/website/manuals_en/Motion%20Sensor%20Quick%20Start%20Guide.pdf) (_by the way, the maximum distance is 7 m, not 4 m as noted in the comparison table above_):

[image]

And this is the detection area for **RTCGQ14LM** from the [user manual](https://cdn.aqara.com/cdn/website/mainland/static/docs/Motion-Sensor-P1_User-Manual.pdf):

[image]

Looking at the detection areas, it is completely incomprehensible why the **RTCGQ14LM** has problems with this.

As one of the reasons, some kind of manufacturing defect is possible, for example, with a lens.

By the way, is there an inconspicuous protective film on the lens that can interfere with the normal operation of the sensor?

backlink-market, 2022-05-22:
Does low or high mean full distance of 7m?

bzeiss, 2022-05-22:
My P1 motion sensors don't have a protective film. I think I have also placed them pretty well according to the instructions. What I have noticed, having tested them for a few days now, is that one single of my P1 sensors seems to overperform all the time (i.e. it triggers at a distance of maybe 4-5 meters) even on low sensitivity while the others don't. Unfortunately, I don't have the Aqara hub to test them with. I hope that someone else here is able to help with this.

Otnow, 2022-05-22:
> Does low or high mean full distance of 7m? can't really see a difference when changing

Theoretically yes, but apparently there is some other reason for the problem.

***

@bzeiss, thanks for the info. Strange problem. Let's wait for feedback from other owners.

TheJulianJES, 2022-05-22:
Using ZHA here with a slightly modified version of [this WIP quirk](https://github.com/zigpy/zha-device-handlers/pull/1541).
I can confirm that the detection range is only around 2 meters on Medium (2). Setting it to High (3) seems to increase the range a bit (sometimes?).

spn91, 2022-05-23:
It seems that this a problem with the retail version of the P1 motion sensor. I didn't noticed it, because I have only one in use and this one doesn't need to detect motion farther away than 2 meters.
This problem is already discussed in the Aqara Beta Test Channel and reported from users which are using the aqara zigbee gateway.

backlink-market, 2022-05-23:
Is it fixable via software update? Or do we need to send the sensor back to aqara?

spn91, 2022-05-23:
We don't know it yet. On the retail version is a newer firmware installed, then on the beta version of the sensor. Maybe this is the reason.

bzeiss, 2022-05-31:
@spn91 Is that beta test channel you are referring to something public? As I have four of the p1 sensors, I'm really interested in a solution.

TheJulianJES, 2022-05-31:
@bzeiss I think spn91 was talking about the pre-release firmware installed on sensors that were for testing.
Z2M has a fix for getting P1 sensors completely working that should be releasing with the next Z2M release (I think it should be tomorrow).

spn91, 2022-05-31:
@bzeiss no, it isn’t. There’s is also no solution for this problem at the moment, so access to this channel wouldn’t help. It has already been escalated to aqara, but no answer from them yet.

TheJulianJES, 2022-05-31:
See https://github.com/Koenkk/zigbee-herdsman-converters/pull/4099 for the initial device support and https://github.com/Koenkk/zigbee-herdsman-converters/pull/4299 for the high illuminance fix.

spn91, 2022-05-31:
Hmm I think I can’t follow you. The device was already supported for a longer time now and the other problem is for the problem with illumination and not the detection range 🤔

TheJulianJES, 2022-05-31:
@spn91 Sorry, the problem I was talking about was with the high illuminance and not the detection range. That will probably really need a firmware upgrade (if it's not a hardware issue)

bzeiss, 2022-07-10:
So I have fetched an Aqara Hub E1 (which seems to be the cheapest Aqara hub) to check whether there is a firmware update that fixes things.  There is indeed an update available:

current version: 0.0.0_0005
Latest version: version 0.0.0_00006

"Optimize illumination reporting".

I'm updating two of my sensors now and repair them with Zigbee2MQTT to find out whether the reporting has improved.

By the way: in case anyone wants to try this too, it has taken quite a while for me to get to this point.

1) Pairing the Aqara Hub E1 with Aqara home: being in Europe I could safely ignore most comments on the Internet that suggested switching modes from Xiaomi Home to Aqara home, resetting everything by pressing the button 10 times etc. Two things were most relevant to me: switching to a 2.4ghz wifi network and actually understanding how the Aqara Home App on Android works. The screen where I was expecting the pairing to happen wasn't even the pairing screen, but a screen before that. A very tiny checkbox (that I didn't identify as something that had to be checked) had to be checked in order to proceed to the actual pairing screen.

2) I thought one firmware update for the hub was enough and after that started the pairing process with the P1. This pairing process failed all the time. It tuned out I had to do like 4 or 5 firmware updates to the Aqara Hub E1 before I had the latest version which could actually pair the motion sensor P1.

I'll report back to this thread as soon as I have a result with the new firmware version. The firmware update of the sensors seems to take a long time.

spn91, 2022-07-10:
I don’t think that a firmware update which fixes this problem is already available. I just got the info this week (after a long time), from an aqara representative, that they began to work on it.

But your work was not for nothing, because may you can provide us with the firmware if it is available in the near future ;)

[4 more replies]