Koenkk/zigbee2mqtt #11995: [New device support]: ADEO/Lexman ENKI Lexman E27 LED white - ZBEK-5

Imbuzi, 2022-03-30:
### Link

https://www.leroymerlin.fr/produits/decoration-eclairage/ampoule-et-led/ampoule-led/ampoule-e27/ampoule-connectee-led-edison-e27-806-lm-variation-de-blancs-intensite-calex-82590434.html

### Database entry

No idea how to get it sry :(

### Comments

The reference on the box is : IST-CDZFB2AS007NA-MZN-01.

External converter tested and fully working.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZBEK-5'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'IST-CDZFB2AS007NA-MZN-01', // Vendor model number, look on the device for a model number
    vendor: 'ADEO', // Vendor of the device (only used for documentation and startup logging)
    description: 'ENKI LEXMAN E27 LED white', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [153, 454]})// Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

### Supported color modes

color temperatur

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}

Koenkk, 2022-04-01:
Could you provide the data/database.db entry of this device?

Imbuzi, 2022-04-05:
I have no idea where I can get it ... :(

Imbuzi, 2022-04-05:
Ok found it ... I use VSCode addon on HA, and by default, it hides all .db files ...
I'm editing the first issue message now.

Koenkk, 2022-04-05:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-04-05 by Koenkk