Koenkk/zigbee2mqtt #12000: Heiman HS2SK power consumption not working anymore

smulle48, 2022-03-31:
### What happened?

The power measurement somehow stopped working. I know it's been working before, but haven't used it lately. Not its reporting nothing. I have tried to remove it from the network and add it again, but still no success.

I've tried both the stable and edge version of zigbee2mqtt, with same result.

### Zigbee2MQTT version

1.24.0-dev commit: 85308da

### Adapter firmware version

20211217

### Adapter

slae.sh_cc2652rb_stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8386614/log.txt)

smulle48, 2022-04-03:
Updated to latest release. Still no success

Apr  3 09:25:44 homeassistant 0368ab1b7151[531]: Zigbee2MQTT:debug 2022-04-03 09:25:44: Received Zigbee message from '0x000d6f0014957975', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acCurrentMultiplier":1,"acVoltageDivisor":100,"acVoltageMultiplier":1}' from endpoint 1 with groupID 0
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]: Zigbee2MQTT:debug 2022-04-03 09:25:44: Received Zigbee message from '0x000d6f0014957975', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acCurrentDivisor":100,"acPowerDivisor":10,"acPowerMultiplier":1}' from endpoint 1 with groupID 0
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]: Zigbee2MQTT:debug 2022-04-03 09:25:44: Received Zigbee message from '0x000d6f0014957975', type 'readResponse', cluster 'haElectricalMeasurement', data '{}' from endpoint 1 with groupID 0
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]: Zigbee2MQTT:error 2022-04-03 09:25:44: Failed to configure '0x000d6f0014957975', attempt 1 (Error: Read 0x000d6f0014957975/1 haElectricalMeasurement(["acFrequencyDivisor","acFrequencyMultiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:25:5)
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/heiman.js:48:13)
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at Configure.configure (/app/lib/extension/configure.ts:115:13)
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at Configure.start (/app/lib/extension/configure.ts:68:13)
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at Controller.callExtensions (/app/lib/controller.ts:316:17)
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at Controller.start (/app/lib/controller.ts:153:9)
Apr  3 09:25:44 homeassistant 0368ab1b7151[531]:     at start (/app/index.js:101:5))

corneyl, 2022-04-03:
I've the same problem, reading the attributes from the device using the dev console seems to work fine though. I get this response: `Read result of 'haElectricalMeasurement': {"rmsCurrent":0,"activePower":17,"rmsVoltage":23300}`.

From the logging above it seems not to support reading `acFrequencyDivisor`, which I indeed can't read using the dev console. I get the same `UNSUPPORTED_ATTRIBUTE` response. Will a failed attribute read prevent reading the remaining attributes? See https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/heiman.js#L48.

smulle48, 2022-04-03:
@corneyl @Koenkk I can confirm that IS indeed that PR that mess up the plug.

Just powered up my test bench and removed the PR, and now it looks like this :-
[image]
[image]
)

I will try later to fix the broken PR :-)

smulle48, 2022-04-03:
I'm out on deep water here :-D

@Koenkk maybe you can help here :-D

Is's line 25 in lib/reporting.js that gives trouble

`await endpoint.read('haElectricalMeasurement', ['acFrequencyDivisor', 'acFrequencyMultiplier']);`

Don't know if other plugs that don't support this, also got problems. If I remove this one line, everything works, so maybe there should be some sort of check, to only use this, if the plug is one specific model?

corneyl, 2022-04-04:
I guess there are two options:

1. Add a parameter to `readEletricalMeasurementMultiplierDivisors` on whether it should read frequency related info.
2. Move the frequency related attribute reads to a separate method.

As `readEletricalMeasurementMultiplierDivisors` is named in a generic way, adding the parameter maybe makes more sense. Something like this?
```js
async function readEletricalMeasurementMultiplierDivisors(endpoint, readFrequencyAttrs = false) {
    // Split into three chunks, some devices fail to respond when reading too much attributes at once.
    await endpoint.read('haElectricalMeasurement', ['acVoltageMultiplier', 'acVoltageDivisor', 'acCurrentMultiplier']);
    await endpoint.read('haElectricalMeasurement', ['acCurrentDivisor', 'acPowerMultiplier', 'acPowerDivisor']);
    // Only read frequency multiplier/devisor when enabled as not all devices support this.
    if (readFrequencyAttrs) {
        await endpoint.read('haElectricalMeasurement', ['acFrequencyDivisor', 'acFrequencyMultiplier']);
    }
}
```

Then for e.g. for the Develco plug we can do:
```
configure: async (device, coordinatorEndpoint, logger) => {
    ...
    await reporting.readMeteringMultiplierDivisor(endpoint, true);
    ...
},
```

It seems that up until now there are only 2 devices using the frequency attributes, see https://github.com/Koenkk/zigbee-herdsman-converters/search?q=e.ac_frequency%28%29.
(I've not set-up a development environment yet, else I could also give it a try 😃)

smulle48, 2022-04-04:
> I guess there are two options:
>
>
>
> 1. Add a parameter to `readEletricalMeasurementMultiplierDivisors` on whether it should read frequency related info.
>
> 2. Move the frequency related attribute reads [...]

Sounds like an great idea. I'll test it later on my test bench :-)

smulle48, 2022-04-04:
@corneyl PR committed :-D

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4086

corneyl, 2022-04-08:
@smulle48 This issue can be closed since the PR is merged, right?

smulle48, 2022-04-09:
> @smulle48 This issue can be closed since the PR is merged, right?

Yep. Closing it :-D

Tnx all for the help :-D

closed 2022-04-09 by smulle48