Koenkk/zigbee2mqtt #12005: TuYa WHD02 gets paired as TS0001

vindolin, 2022-03-31:
### What happened?

I have linked some 10A versions of the WHD02 relay switches in my network and now bought the 16A version.
The 10A all got recognized as WHD02 but the 16A get recognized as a wall switch (device photo) TS0001.
When I go into the details of the 10A version they have TS0001 as the Zigbee Model but WHD02 under Model.
[image]

The 16A version shows both Zigbee Model and Model as TS0001.
[image]

Here's my database.db
[zigbee2mqtt_database.zip](https://github.com/Koenkk/zigbee2mqtt/files/8388431/zigbee2mqtt_database.zip)

### What did you expect to happen?

Should be recognized as WHD02

### Zigbee2MQTT version

1.24.0-dev

### Adapter firmware version

20211217

### Adapter

Sonoff CC2652P

vindolin, 2022-03-31:
I copied duplicated a line from a working WHD02 in my database.db and  only changed the ieeeAddr and nwkAddr to the one of the 16A version and it now works including the power_on_behavior.

teoconnect, 2022-04-16:
I have resolved wen i passed to the edge version, the resulting database.db for the WHD02 (that works) attached below.

`{"id":6,"type":"Router","ieeeAddr":"0xa4c138848885231d","nwkAddr":39512,"manufId":4417,"manufName":"_TZ3000_mx3vgyea","powerSource":"Mains` (single `phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_mx3vgyea","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024cbd5c3","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1650102127370,"defaultSendRequestWhen":"immediate"}`

thijsdejong, 2022-04-28:
Is there a guide available on how to fix this? Would be really appreciated so I can use my momentary switches with these!

Koenkk, 2022-04-29:
Both `_TZ3000_npzfdcof` and `_TZ3000_mx3vgyea` will be recognised as WHD02 in the latest dev and next release on 1 May.

thijsdejong, 2022-05-02:
@Koenkk mine are _TZ3000_rmjr4ufz, could those please also be recognized as WHD02

Koenkk, 2022-05-02:
@thijsdejong added. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

thijsdejong, 2022-05-02:
@Koenkk it looks like your change is incorrect. It should be under WHD02 model, not TS0202.

Koenkk, 2022-05-03:
@thijsdejong you are right 👍 fixed it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

thijsdejong, 2022-05-04:
@Koenkk I have just confirmed it works as expected, thanks! Now I can fit all of my devices with pulse switches around the house 😄

pannal, 2023-11-23:
Another WHD02: _TZ3000_skueekg3

jaclas, 2023-12-18:
Another case: _TZ3000_kycczpw8

[image]

[image]

casamyj, 2024-02-02:
> Another case: _TZ3000_kycczpw8 [...]

I have the same problem with this version

PVerburg, 2024-02-29:
have a couple here :_TZ3000_fdxihpp7 
should be Aubess TMZ02
[image]