Koenkk/zigbee2mqtt #12006: Develco Air quality sensor AQSZB-110

RitaSghe, 2022-03-31:
### What happened?

Hi,
I have this device but I have incorret Voc value (too high).
Frient support denies any responsabilty and says zigbee2mqtt the probable responsible of issue.
Somebody have this device ?
The Voc values are correct?

Example, yesterday after 2 hours of window open the value was 39
I close the window and the value are:
16:26                 Voc value      39
16:44                                       587
16:50                                       1094
17:03                                       1537
17:24                                       1901
19:30                                       4497

The windows was close, no person into the room .....

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26720700 - 0x00212effff07359e - 0.6.71

### Adapter

Conbee II

sjorge, 2022-03-31:
I added support for this device, zigbee2mqtt uses the `measuredValue` without any transformations applied.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/e9292187c98f62ca686a84fea25338fb5d82c883/converters/fromZigbee.js#L1774-L1801

deCONZ is also using the raw value without any transformations applied: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/8b07851c9ad431a3fafa4dee05d671b7ee8b8e94/devices/frient/aqszb-110_voc_sensor.json#L98

Looking at their own technical docs https://www.develcoproducts.com/media/1674/aqszb-110-technical-manual-air-quality-sensor-04-08-20.pdf 

[image]

[image]

Mentions it's a raw ppb value and that is what we present in the `voc` payload.

sjorge, 2022-03-31:
That being said, personally also struggle with getting the value below 100ppb, even with all windows open and a fan creating airflow.

[image]

These are the number for today, I got up around 6:00 and left the house to go to work, both rooms have one of the windows on the small opening stand to air them.

RitaSghe, 2022-03-31:
Ok, thank you ....

closed 2022-05-09 by github-actions[bot]

fdemmer, 2022-12-05:
I have three of those sensors sitting next to each other on my desk and those are the state readouts I get after letting them sit there for an hour:

[image]

Temperature variation is good, humidity barely within tolerance and VOC is all over the place.

I guess the idea of air quality measurements is a good one, but the cheap sensors are not quite there.

jessegabriels, 2023-11-04:
> I have three of those sensors sitting next to each other on my desk and those are the state readouts I get after letting them sit there for an hour:
>
> [image]
>
> Temperature variation is good, humidity [...]

Hey I have 3 of these sensors too. I am having trouble setting them up. Is there any chance that you could help me?