Koenkk/zigbee2mqtt #12007: [New device support]: Zigbee device Humidity, Temperature, clock with clusters (000A Time) and (EF00 Tuya), TS0601,_TZE200_c7emyjom

Bibyte, 2022-03-31:
### Link

https://www.tuya.com/

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0xa4c1385e5e9baf30","nwkAddr":33269,"manufId":4417,"manufName":"_TZE200_c7emyjom","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"C\u0000\u0000\u0000eC\u0000\u0000\u0000\u0012��\u0007\u0000e��\u0007\u0000\u0012","65506":34,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_c7emyjom","powerSource":3,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0e96a","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648741927268,"defaultSendRequestWhen":"immediate"}

### Comments

This is Tuya zigbee device Humidity, Temperature, clock with clusters (000A Time) and (EF00 Tuya), TS0601,_TZE200_c7emyjomI I try to do similar other external converter TuYa Humidity, Temperature sensor.
but it doesn't show temperature, humidity and battery. It does not work :( 
Can you help me to setup this device?

### External converter

```shell
{
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_c7emyjom'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_bjawzodf'}],
        model: 'TS0601_temperature_humidity_sensor',
        vendor: 'TuYa',
        description: 'Temperature & humidity sensor',
        fromZigbee: [fz.tuya_temperature_humidity_sensor],
        toZigbee: [],
        exposes: [e.temperature(), e.humidity(), e.battery()],
    }
```

Koenkk, 2022-04-01:
Could you try with the following external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bq5c8xfe'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_bjawzodf'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_c7emyjom'}],
    model: 'TS0601_temperature_humidity_sensor',
    vendor: 'TuYa',
    description: 'Temperature & humidity sensor',
    fromZigbee: [fz.tuya_temperature_humidity_sensor],
    toZigbee: [],
    exposes: [e.temperature(), e.humidity(), e.battery()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;

```

Bibyte, 2022-04-02:
> Could you try with the following external converter?

Thank you! It works!
But this device have a clock. On screen the clock is disabled. Is it possible to enable it and update the time on device screen?

Koenkk, 2022-04-02:
Try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_bq5c8xfe'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_bjawzodf'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_c7emyjom'}],
    model: 'TS0601_temperature_humidity_sensor',
    vendor: 'TuYa',
    description: 'Temperature & humidity sensor',
    fromZigbee: [fz.tuya_temperature_humidity_sensor],
    toZigbee: [],
    onEvent: tuya.onEventSetTime,
    exposes: [e.temperature(), e.humidity(), e.battery()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
    },
};

module.exports = definition;
```

Bibyte, 2022-04-04:
> Try with:

I have add, and reboot HA. The time is not shows on the device. Not work(

Aaike, 2022-04-04:
Hi, I seem to have a device with the same model number and manufacturer but it looks different.
see: https://www.aliexpress.com/item/1005003340156712.html

I tried with the external converter suggested by @Koenkk above. I seem to have the same issue.
I can see the temperature and humidity being exposed (battery level is null though) and the clock does not appear on the device screen.

[image]

Bibyte, 2022-04-05:
> but it looks different

It looks exact like my.  I still didn't solve this problem for me too.

xxxTODxxx, 2022-04-06:
Hey, maybe I have the same device name but a little bit different manufactureName: _TZE200_locansqn
I use it with iobroker.
Device: https://de.aliexpress.com/item/1005003634353180.html

[image]

Bibyte, 2022-04-06:
I found the driver with my model (_TZE200_c7emyjom) [there](https://raw.githubusercontent.com/kkossev/Hubitat/development/Drivers/Tuya%20Temperature%20Humidity%20Illuminance%20LCD%20Display%20with%20a%20Clock/Tuya_Temperature_Humidity_Illuminance_LCD_Display_with_a_Clock.groovy).
But how to install it?

Bibyte, 2022-04-06:
Hey! I have sole this problem! I use the [Nous SZ-T04](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/nous.js) converter and all works and time become shows on screen!
_TZE200_lve3dvpy - I use from this and replace to it my _TZE200_c7emyjom

closed 2022-05-16 by github-actions[bot]