Koenkk/zigbee2mqtt #12015: After upgrading to version 1.25.0-1, errors appeared. Zigbee2Mqtt does not start.

VikeDragon, 2022-04-01:
### What happened?

After upgrading to version 1.25.0-1, errors appeared. Zigbee2Mqtt does not start.

### What did you expect to happen?

Worked as before the update

### How to reproduce it (minimal and precise)

Working configuration on v 1.24.x and earlier :

{
  "external_converters": [
    "diyruz.js",
    "useelink.js",
    "DIYRuZ_Flower_WS.js",
    "tuya.js",
    "xiaomi.js"
  ],
  "devices": [
    "devices.yaml"
  ],
  "groups": [
    "groups.yaml"
  ],
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto",
    "user": "myuser",
    "password": "mypassword",
    "version": 5
  },
  "serial": {
    "port": "/dev/ttyUSB0"
  },
  "advanced": {
    "log_level": "warn",
    "pan_id": 6754,
    "channel": 15,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blocklist": [],
    "availability_passlist": [],
    "last_seen": "ISO_8601_local"
  },
  "device_options": {
    "temperature_precision": 2
  },
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": {
    "port": 8099
  },
  "experimental": {
    "transmit_power": 19
  },
  "availability": false
}

Transfered to configuration in UI addon after updated :

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto
  user: myuser
  password: mypassword
  version: 5
serial:
  port: /dev/ttyUSB0
external_converters:
  - diyruz.js
  - useelink.js
  - DIYRuZ_Flower_WS.js
  - tuya.js
  - xiaomi.js
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
advanced:
  log_level: warn
  pan_id: 6754
  channel: 15
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
  last_seen: ISO_8601_local
device_options:
  temperature_precision: 2
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  transmit_power: 19
availability: false

### Zigbee2MQTT version

1.25.0-1

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee3.0 USB Dongle

### Debug log

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[14:07:41] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[14:07:42] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.0 start
> node index.js
(node:371) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'options' of undefined
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:83:58)
    at start (/app/index.js:100:18)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:371) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:371) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

dielessen, 2022-04-01:
1 april …

VikeDragon, 2022-04-01:
Found a partial solution :

In GUI addon set only :

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt: {}
serial: {}

If you leave a mixed configuration, it does not start.

in file configuration.yaml
place everything else, but without converters

And .... He 's alive ! He lives! But apparently the converters will have to be rewritten :(
And why, at the first start, addon does not bring everything back to normal in the GUI, but leaves mixed

closed 2022-04-01 by VikeDragon