Koenkk/zigbee2mqtt #12016: Devices vanish from coordinator_backup.json

okastl, 2022-04-01:
### What happened?

Today I have shut down my z2m to update the coordinator firmware and to update z2m to new version.
coordinator_backup.json was written, but one device is missing (a frient smoke detector).
I have written a little python program to check for problems in coordinator_backup.json, otherwise such errors will be unnoticed.

### What did you expect to happen?

coordinator_backup.json should contain all devices with their keys after shutdown.

### How to reproduce it (minimal and precise)

If I only knew how to reproduce this. It just happens sometimes. Last time a Hue ceiling lamp went missing from coordinator_backup.json for no obvious reason.
The Hue ceiling lamp continued to work just fine, even with its entry missing. Starting & stopping z2m multiple times never brought the missing entry back. Only re-pairing the lamp did.

### Zigbee2MQTT version

1.22.4, 1.22.5

### Adapter firmware version

20220219 (and previous, too)

### Adapter

Slaesh CC2652RB

### Debug log

Nothing interesting in any log. Unfortunately.

Koenkk, 2022-04-02:
@castorw how can we debug this?

castorw, 2022-04-16:
Hi @okastl,
this might be cased by inconsistencies in the backup. Please provide zh/adapter snapshots in the desired and problematic state. Let the snapshots include the following:
* IIEEE address of the problematic device,
* Coordinator Backup,
* Adapter NV Dump (you can use nvram_read.py from https://github.com/zigpy/zigpy-znp/tree/dev/zigpy_znp/tools to create the dump).

I would like to also note that we have encountered situations where some devices were not populated in adapter NV (especially when they aren't APS-enabled and were paired via router). Was this device present in `coordinator_backup.json` any time before or you just noticed it's missing when the problems started?

okastl, 2022-04-16:
> Please provide zh/adapter snapshots in the desired and problematic state.

I will try.

 
> I would like to also note that we have encountered situations where some devices were not populated in adapter NV (especially when they aren't APS-enabled and were paired via router). Was this [...]

Yes, it was present, and it is APS enabled. I have re-paired it, and it is currently present in the backup:

```
...
    {
      "nwk_address": "6da3",
      "ieee_address": "0017880109877b88",
      "is_child": false,
      "link_key": {
        "key": "f3022409379da07711e0fefbb2b1eec0",
        "rx_counter": 0,
        "tx_counter": 0
      }
    },
...

```
"is_child": false is strange, as are  "rx_counter" and  "tx_counter" both 0.
It is this Philips Hue Flourish:

`{"id":136,"type":"Router","ieeeAddr":"0x0017880109877b88","nwkAddr":28067,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"4090531P9","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0,"zclVersion":2}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20211021","swBuildId":"1.93.3","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1650141927341,"defaultSendRequestWhen":"immediate"}`

castorw, 2022-04-16:
@okastl The `is_child` only indicates whether the device is direct child of coordinator. If it is paired via router it should be `false`. APS frame counters may be zero if the device does not transmit APS-encrypted payloads. Z2M/ZH does not send APS-encrypted frames at the moment.

okastl, 2022-05-17:
I made an interesting observation. I had re-added a "vanished" Develco Smoke Detector. After shutting down z2m, `coordinator_backup.json` looks like this:

```
...
    {
      "nwk_address": "5f86",
      "ieee_address": "a4c138f8ccd75fa5",
      "is_child": true,
      "link_key": {
        "key": "b0e8a055634133179ca617b7819bc192",
        "rx_counter": 0,
        "tx_counter": 11
      }
    },
    {
      "nwk_address": "384a",
      "ieee_address": "0015bc00310097c3",
      "is_child": false,
      "link_key": {
        "key": "70aef9f17bc5b600f582b74699276cf2",
        "rx_counter": 0,
        "tx_counter": 11
      }
    }
  ]
}

```
ieee_address 0x15bc00310097c3 is the smoke detector.

I have added and removed some other devices. After a z2m shutdown coordinator_backup.json looks like this:

```
...
    {
      "nwk_address": "5f86",
      "ieee_address": "a4c138f8ccd75fa5",
      "is_child": true
    },
    {
      "nwk_address": "384a",
      "ieee_address": "0015bc00310097c3",
      "is_child": false
    },
    {
      "nwk_address": "3677",
      "ieee_address": "804b50fffe028e3f",
      "is_child": false
    },
    {
      "nwk_address": "79b7",
      "ieee_address": "5c0272fffedab6cb",
      "is_child": false,
      "link_key": {
        "key": "3276a1875c7234497cc143658680b165",
        "rx_counter": 0,
        "tx_counter": 0
      }
    }
  ]
}
```

I believe "a4c138f8ccd75fa5" was removed from the network, but more important the link key of  "0015bc00310097c3" is gone!

If you now erase nvm of the stick and start / shutdown z2m, the entry for "0015bc00310097c3" is completely gone and my python script will issue an alert.

So, the "vanishing" of the device is a follow up error to missing link_key entry in coordinator_backup.json.

Maybe this points a little bit in the right direction?

I assume restoring the link key from an older backup is useless because of the unknown values of rx/tx counters?

okastl, 2022-10-02:
I have added some files, which can probably help you to reproduce the issue.
Before updating to the latest zstack firmware, I have created a nvm backup and a network backup using zigpy_znp tools.
I have added these to the archive as network_backup.json and nvm_backup.json.
I have erased nvm of a spare adapter and did a network restore.
After doing this, I created a new nvm & network backup called network_backup1.json and nvm_backup1.json.
As you can see, 2 devices have now vanished.
If you have a spare adapter you can reproduce this problem.

[network_backup.zip](https://github.com/Koenkk/zigbee2mqtt/files/9692568/network_backup.zip)

okastl, 2022-10-02:
@Koenkk can you please reopen this issue? Thank you.

okastl, 2022-10-02:
A related issue: It looks like there is a limitation of the link_keys (TCLK_TABLE?)
I noticed, that a couple of my devices have lost their link keys. If I had to guess, something limits the number of link keys to be <100, and if a new device is paired, some other link key is overridden.

Koenkk, 2022-10-04:
I expect the limit to be `200` (because of https://github.com/Koenkk/Z-Stack-firmware/blob/262b9d2cf2e1cc8db591d784176b3223b589d706/coordinator/Z-Stack_3.x.0/firmware.patch#L347). Can you check if with this fw the limit is `120`?

[znp_LP_CC2652RB_tirtos_ticlang.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/9709070/znp_LP_CC2652RB_tirtos_ticlang.hex.zip)

okastl, 2022-10-05:
Sorry, I don't understand.
If I look at previous versions, they have ZDSECMGR_TC_DEVICE_MAX already set to 200. Both the release (master) version, as the current development version.
Or did you change something else? If yes, I am happy to try it.

> I expect the limit to be `200` (because of https://github.com/Koenkk/Z-Stack-firmware/blob/262b9d2cf2e1cc8db591d784176b3223b589d706/coordinator/Z-Stack_3.x.0/firmware.patch#L347). Can you check if with this fw the limit is `120`?
>

`120`, is this a typo and you mean `200`?

okastl, 2022-10-05:
> i increased the 'NWK_MAX_DEVICE_LIST' to 120 (it was 100 before which matches your limit). This value should not be related to the max link keys entries but lets try anyway.

Ah, now I understand. I will try it in with my production stick.

What I already tried, is the "lost" devices behavior with a spare stick on network restore.
I don't know if the problem is related, but I repeated the test here https://github.com/Koenkk/zigbee2mqtt/issues/12016#issuecomment-1264641503 with the new firmware, and when restoring the network backup, two devices were lost.

okastl, 2022-11-06:
> [znp_LP_CC2652RB_tirtos_ticlang.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/9709070/znp_LP_CC2652RB_tirtos_ticlang.hex.zip)

I tried this firmware, added some Tuya plugs to increase the number of devices and all seems to work well.
I kept z2m running until the 1.28.1 release. I shut down z2m before the update and z2m created a coordinator_backup.json.
I noticed that a hue lamp was missing from the backup. In this state I used zigpy_znp tools to create a nvm backup and a network backup.
When creating the network backup, zigpy_znp.tools.network_backup showed the following warning:

`2022-11-03 17:58:34.938 alarm zigpy_znp.znp.security WARNING Skipping hashed link key 2b:7b:b1:1c:b6:ff:8e:9b:f1:89:34:30:f8:48:6c:79 (tx: 33, rx: 0) for unknown device 00:17:88:01:09:87:7b:88
`
and the same device was missing like in the coordinator_backup.json.

A device still vanishes from the NVM and the backup coordinator_backup.json.
I assume this problem is happening inside the TI stack?

What else can I do, which firmware should I use? The current beta?
Should I just ignore this problem?

I attach the mentioned files in case you want to take a look.

[missing.zip](https://github.com/Koenkk/zigbee2mqtt/files/9944956/missing.zip)

Koenkk, 2022-11-06:
> A device still vanishes from the NVM and the backup coordinator_backup.json.
I assume this problem is happening inside the TI stack?

Yes, but you can try the latest fw from https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin

okastl, 2022-11-07:
Thank you, I am running the latest development firmware now.
But something weird happened. To flash the new firmware, I had to shutdown z2m and it created a new coordinator_backup.json.
To my surprise, the previously missing device 00:17:88:01:09:87:7b:88 re-appeared in the backup, including its link key!
The TI stack must have added it again, after it was used by z2m.

Before flashing the new firmware, I created a nvm backup with zigpy_znp tools, then I flashed the new firmware.

With the new firmware flashed, I did some tests.
1.) Restore the created nvm backup with zigpy_znp tools.
2.) Create a network backup with zigpy_znp tools. All devices are present in the backup.
3.) Erase the nvm
4.) Restore the network backup.
5.) Create a fresh nvm backup and network backup. In both these backups 2 devices were missing.

The conclusion would be (apart from "TI stack is weird"), that the network restore to an empty device doesn't work reliably.
The workaround is now to create an nvm backup before flashing and restore it afterwards. This seems to work across similar devices (based on the same architecture) and between firmware versions.
Is this correct and I can use nvm backups instead of the network backups?

Koenkk, 2022-11-08:
I would expect that the NVM backup contains everything, so restoring it on a device and creating a NVM backup again should give the same amount of devices unless some filtering is happening inside the TI firmware.

okastl, 2022-12-09:
Still testing latest beta firmware. Make github bot happy.

okastl, 2023-01-09:
With firmware 20221102 a Philips Hue 4090531P9 (Hue Flourish) was kicked out by the firmware. The link key of one Develco SMSZB-120 smoke detector was removed for no apparent reason.
Now testing 20221226...

closed 2023-02-18 by github-actions[bot]